SwiCago/HeatPump #10: Settings callback not triggered after update()

jarrod180, 2017-02-15:
if update() is called and some new wantedSettings are transmitted, the callback is not triggered because currentSettings are assigned to wantedSettings if the lastUpdateSuccessful is true.

currentSettings should probably only be updated by receiving a settings message from the HP.

I remove this code in update()

`if(lastUpdateSuccessful) {`
   `currentSettings = wantedSettings;`
`}`

if someone confirms it's a problem in here and not something I did in my fork I'll submit a pull request

SwiCago, 2017-02-15:
jarrod, I agree. The library should not just assume because the update was successful, that that is what the library believes is the settings of the heatpump. There is also no guarantee that the next set of info data is 0x61, especially if the HP is flushing a previous 0x62 info packet. So a false failed update could be triggered. That was why I never bothered looking at 0x61 in my original library, as it should confirm those settings through sync.
I used real data from the HPs current settings to check if they matched my wanted settings at sketch level. If they didn't match I just resent the command to update. In theory sync could do this automatically after a given time frame to make sure the settings info set has been grabbed and if they differ auto resend.
Kayno, looking at the codes between the original and the big merge, I see a lot has changed. I will have to go through it more closely and fix a few things.  As suggested above, I think I will want to modify sync to do a comp of wanted and current settings after x seconds update was sent and do a resend if they don't match. A safe guard needs to be in place for manual changes via IR. I suggest using the success var. If success already true(do to previous check), then do not resend wanted if differ from current.

kayno, 2017-02-16:
I brought over the code from unreality to check the 0x61 library, as I liked the way it allowed for a return value on the `update()` function. It meant calling scripts could do things like `if(!update()) { // update didn't work, do something }`. If we relied on the `sync()` method to do this, calling scripts would have to call `update()`, then `sync()`, and test that the callback was called - not near as ideal as just calling `update()` and seeing what it returns.

I think we should be checking for the 0x61 - I have seen that packet come after every update I have issued and think it's reliable. We just need to make sure the library waits for it after an `update()` - perhaps a timeout?

I think the library should be the place where current and wanted settings is dealt with - we should be abstracting the workings of the protocol, syncing, etc from the user. They just want getters and setters for the settings, and notification (i.e. callback) when settings change outside of the user's sketch (e.g. IR remote). 

Looking at the debug, I can see that the callback is firing after an update(), but this must be coming from the 0x62/0x02 (settings) packet that comes through after the update though:

```
heatpump/debug {"packet":"02 00 00 00 03 05 00 07 00 00 0c b4 00 00 00 00 "} // normal incoming settings packet
heatpump/set {"mode": "AUTO"} // issued a settings change over MQTT
heatpump/debug {"packet":"03 00 00 0c 00 00 ac 00 00 00 00 00 00 00 00 00 "} // normal incoming room temp packet
heatpump/debug {"packet":"00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "} // this must be the 0x61 acknowledging the update(). We should consider adding the header to the packet that the packet received callback returns, to confirm
heatpump/debug {"packet":"02 00 00 00 08 05 00 07 00 00 0c b4 00 00 00 00 "} // another incoming settings packet, this time with the new mode (5th packet now 08, was 03)
heatpump {"power":"OFF","mode":"AUTO","temperature":26,"fan":"AUTO","vane":"SWING","wideVane":"SWING"} // callback called because 02 settings packet changed, status issued over MQTT
```

SwiCago, 2017-02-16:
@kayno, yes, 0x61 should stay, but I agree with Jarrod that 
if(lastUpdateSuccessful) {
currentSettings = wantedSettings;
} 
should go. The update should only return a bool, not update currentSettings with wantedSettings as the hp already does this when sync'ing, this way the user knows the data in currentSettings is from the hp and not what they sent. I have not removed it yet, but think it should go.
I also, still believe the library should attempt a resend if it deems the settings wanted do not match the current settings within a given time frame(say 10seconds)....that way changes from IR do not get changed.

SwiCago, 2017-02-16:
@kayno, don't forget to resync your branch. I made some updates just recently.

kayno, 2017-02-16:
@SwiCago it's making more sense now! I just want to make sure everything is thought out :)

An alternative would be to call the callback within the if statement you are proposing be removed, e.g.:

```
if(lastUpdateSuccessful) {
  currentSettings = wantedSettings;
  settingsChangedCallback();
}
```

if it was done this way, we take away the reliance on the sketch calling `sync()`. Even though I imagine `sync()` will always be in a sketch's loop, we can't be sure it will be. Take for example this sketch - one I made up for a scenario where you wanted a "keep alive" to be sent to the heatpump every 24hrs to just send a power on update to the heatpump, and then go into deep sleep for the 24hrs:

```
void setup() {
  hp.connect(&Serial);
  hp.setSettingsChangedCallback(hpSettingsChanged);
}

void hpSettingsChanged() {
  // do something - e.g. send a notification somehow to say it was powered on
}

void loop() {
  hp.setPower("ON");
  hp.update();

  // put ESP8266 into deep sleep for 24hours
}
```

I know it might not be realistic, but still if someone was writing this, they might not think `sync()` is required as they don't care what the settings are... We can't force people to put `sync()` in their `loop()`.

I guess I just feel that by removing that if statement what updates `currentSettings`, we are saying that we don't trust that the 0x61 is telling the truth, so we are ignoring it and waiting for a settings packet....

I may also be over looking a race condition here, with calling the callback from `update()`...

I need to stop thinking so much!

kayno, 2017-02-16:
I think the real question here is - should one expect that `currentSettings` is changed as part of the `update()` call? I think the answer is _yes it should_. An easy way to argue this is:

```
void loop() {
  String requested_power_setting = getRequestedPowerSetting(); // this prompts the user or processes a request from the user somehow, and get's an ON or OFF
  hp.setPowerSetting(requested_power_setting );

  if(hp.update()) {
    var current_power_setting = getPowerSetting();
    mqtt.publish("topic/name", "power is now "+current_power_setting);
  }

  hp.sync(); //keep it in sync
}
```

If `update()` doesn't update `currentSettings`, than the "power is now ON/OFF" message will never be published. Even though the user is calling `sync()` in their loop as required.

Perhaps `update()` could call `sync()` (and we pass `sync()` a param that tells it which info packet to send, e.g. ask for settings, not current room temp), if we really don't want to trust the 0x61 packet?

SwiCago, 2017-02-16:
@kayno you have a point, however i still dont like the fact that we just assume wanted settings is now current settings, when the heatpump accepts the update command. In theory instead of the library over writiing the settings, the user loop could just assume their setting worked, because of the returned true.
Example:
Loop() {
String reqPwrSetting = getReqPowerSetting
Hp.setPowerSetting(reqPwrSetting)
If(hp.update())
{
Mqtt.publish(TOPICNAME,reqPwrSetting)
}
Hp.sync()
}

comes down to same thing, except that instead of library forcefully assuming wanted is now current, the user can assume on their own, untill a callback is made when it is really known.
Excuss my sloppy typing, typing on tablet

kayno, 2017-02-16:
@SwiCago @jarrod180 - I have started to code up how I think this could work. I am hearing the desire to not rely solely on the 0x61 packet, however I still believe that update() should return the result of update. So I am going to look at doing both, by (essentially) calling sync() and as I mentioned earlier, passing an optional param to it to instruct it to do a settings sync (rather than relying on it to decide settings or room temp). Something like that!

I have yet to code that up, but in the meantime I have done some prep work. See my progress at https://github.com/kayno/HeatPump/commit/e5fffd992e012f35c699073bf4a850eef7b0d39e. Keen to hear your thoughts :)

kayno, 2017-02-17:
Those 4 commits combined should address this! 

With these commits, if update() gets a 0x61 response confirming it was successful, it then calls sync() requesting that the settings be updated. This means currentSettings is only ever changed when a 0x62/0x02 packet is received from the heatpump, which will then trigger the callback (solving the original issue!)

If anyone can give my fork a go, that would be great: https://github.com/kayno/HeatPump. It also has a fix for #9.

SwiCago, 2017-02-17:
@kayno, is this issue fully resolved or should we leave it open? 
I had a chance to review the latest changes. I like them. They make sense and look clean. I'll retest my simple web app and see if all is still good. But from the looks of it, it will be fine.

kayno, 2017-02-17:
I think it can be closed. I'm pretty sure that the changes addresses all of the issues and requirements.

closed 2017-02-17 by SwiCago