SwiCago/HeatPump #6: mqtt example: room temperature reported as 1073741826 (degrees C)

kayno, 2017-02-15:
I am running the mitsubishi_heatpump_mqtt_esp8266.ino and in the last couple of days the temperature is being reported as 1073741826 (looks like an int variable has not been set?). It's only started in the last few days - might be related to the [callback commit](https://github.com/SwiCago/HeatPump/commit/b68324baa6ee8b1ed745740554d98a54bba56112)?

I switched on debug mode over MQTT and I can see that the info packet (03) is reporting in the 4th byte (where room temp lives) "0a" (hex) which maps to 20 degrees according to ROOM_TEMP_MAP

here is the MQTT output:

heatpump/temperature {"roomTemperature":1073741826}
heatpump/debug/set on
heatpump/debug debug mode enabled
heatpump/debug {"packet":"03 00 00 0a 00 00 a8 00 00 00 00 00 00 00 00 00 "}
heatpump/debug {"packet":"02 00 00 00 08 04 00 07 00 00 0c b6 00 00 00 00 "}
heatpump/debug {"packet":"03 00 00 0a 00 00 a8 00 00 00 00 00 00 00 00 00 "}
heatpump/debug {"packet":"02 00 00 00 08 04 00 07 00 00 0c b6 00 00 00 00 "}
heatpump/debug {"packet":"03 00 00 0a 00 00 a8 00 00 00 00 00 00 00 00 00 "}
heatpump/debug {"packet":"02 00 00 00 08 04 00 07 00 00 0c b6 00 00 00 00 "}
heatpump/temperature {"roomTemperature":1080950825}
heatpump/debug {"packet":"03 00 00 0a 00 00 a8 00 00 00 00 00 00 00 00 00 "}
heatpump/debug {"packet":"02 00 00 00 08 04 00 07 00 00 0c b6 00 00 00 00 "}
heatpump/debug/set off
heatpump/debug debug mode disabled
heatpump/temperature {"roomTemperature":1073741826}
heatpump/temperature {"roomTemperature":1073741826}
heatpump/temperature {"roomTemperature":1073741826}
heatpump/temperature {"roomTemperature":1073741826}

I don't think the fact that it changes to 1080950825 during debug mode is anything to read into, as I am guessing it's reporting a random piece of memory for roomTemperature because it's not set, and that random piece of memory changes when debug mode is switched on.

I haven't had a chance to debug further yet, but wanted to post it here first :)

SwiCago, 2017-02-15:
It is set in constructor with a default setting. 
heatpumpSettings defaultSettings = {POWER_MAP[0],MODE_MAP[3],TEMP_MAP[9],FAN_MAP[2],VANE_MAP[1],WIDEVANE_MAP[2],ROOM_TEMP_MAP[12]};
currentSettings = defaultSettings;

kayno, 2017-02-15:
Maybe getRoomTemperature() is not picking it up then? The commit I referred to saw me change the way roomTemperature was populated in the JSON - I switched to the getRoomTempperature() method (previously it was extracted from getCurrentSettings().

I wont be able to debug/test until tonight, and can't see anything glaringly obvious in the code.

jarrod180, 2017-02-15:
Also saw this yesterday when testing mine out

in HeatPump::getData() looks like could be a race condition between receiving the 0x02 settings and the 0x03 room temp. When receiving the 0x02 settings, it creates a new receivedSettings and never sets roomTemperature, then assigns it to currentSettings. Thus roomTemperature might be lost there, alas, I can't test until later either.

SwiCago, 2017-02-15:
Never had that with the version before the merge with your code. It is probably something really stupid

SwiCago, 2017-02-15:
Kayno, jarrod180 is on to something. Before the merge with your vesion, I store received data differently. It looks like receivedSettings is messing up the room temp. Maybe copy current.roomtemp to received.roomtemp and then replace currentsetting

SwiCago, 2017-02-15:
Here is how it was before we merged
if(data[21] == chkSum) {          
 +          if(data[5] == INFOMODE[0]) { //Set packet
 +            currentSettings[0] = POWER_MAP[findValueByByte(POWER, 2, data[8])];
 +            currentSettings[1] = MODE_MAP[findValueByByte(MODE, 5, data[9])];
 +            currentSettings[2] = TEMP_MAP[findValueByByte(TEMP, 16, data[10])];
 +            currentSettings[3] = FAN_MAP[findValueByByte(FAN, 6, data[11])];
 +            currentSettings[4] = VANE_MAP[findValueByByte(VANE, 7, data[12])];
 +            currentSettings[5] = DIR_MAP[findValueByByte(DIR, 7, data[15])];
 +          }
 +          else if(data[5] == INFOMODE[1]) { //Temp packet
 +            currentSettings[6] = ROOM_TEMP_MAP[findValueByByte(ROOM_TEMP, 32, data[8])];
 +          }

SwiCago, 2017-02-15:
Kayno, jarrod180..i updated the code to copy roomTemp from current to received, so that when received is cooied back to current, that it isnt lost...see if that fixes the issue

kayno, 2017-02-15:
great pick up @jarrod180 (again!). I feel a bit guilty opening an issue that I caused and others fixed, so thanks!

I will test tonight and see how it goes.

Also, this highlights [one of my comments in the code](https://github.com/SwiCago/HeatPump/blob/master/src/HeatPump.h#L53), which says that roomTemperature "isn't a "setting" as such, shouldn't be in struct", and this is further highlighted by the fact that it's not compared in comparison operators for the struct. I will create an issue to move roomTemperature out of the struct, and get that implemented.

Thanks for the fix too @SwiCago - will test tonight :)

SwiCago, 2017-02-15:
Hopefully the fix works...i didnt get a chance to test yet

jarrod180, 2017-02-15:
@kayno, don't worry .... it's more interesting than the work I'm supposed to be doing!

Agreed, a different mechanism for handling room temp outside the settings struct and methods would be more appropriate IMO.

kayno, 2017-02-15:
@SwiCago the fix works - thanks again to both yourself and @jarrod180!

```
heatpump/temperature {"roomTemperature":22}
heatpump {"power":"OFF","mode":"AUTO","temperature":27,"fan":"AUTO","vane":"SWING","wideVane":"SWING"}
heatpump/temperature {"roomTemperature":24}
heatpump/temperature {"roomTemperature":24}
heatpump/temperature {"roomTemperature":24}
```

closed 2017-02-15 by kayno