SwiCago/HeatPump #7: WideVane

SwiCago, 2017-02-15:
WideVane setting seems to be ignored by heatpump. Changing via IR remote, HP reports the change. change via esp and it is ignored.
Maybe we nee to look at the IR protocol and see if that works with the custom packet sender
https://github.com/r45635/HVAC-IR-Control/blob/master/Protocol/Mitsubishi_IR_Packet_Data_v1.1-FULL.pdf
Also, I hooked up my kumo cloud again and it doesn't even have wideVane option LOL...I got mine for free, but mitsu sells these for $180 a pop, what a ripoff

kayno, 2017-02-18:
not related to windVane, but in the name of trying to decode the settings packet more which might help with windVane, I created a web page that listens to the debug mqtt topic and nicely formats and displays the packet. It shows the current packet from the heatpump, as well as the previous packet that was different to the current packet. In other words, current packet is what the heatpump is spewing out each second or so, and previous is the last packet before the current settings were set!

I also grouped by packet types, e.g. settings and current room temp. it highlights the bytes that differ between previous and last. This means as I press buttons on the IR remote, the webpage instantly updates and I can see exactly what bytes change for each setting.

It's a work in progress, but one interesting thing i have discovered so far is that when you change the temperature setting, two bytes are updated, not one. The library uses the 11th byte (i.e.` packet[10]`) but the 17th byte (i.e. `packet[16]`) also changes. seems to go up by a value of two for each temperature increment.

Here is a screenshot of what I am seeing: http://i.imgur.com/fyQwCa4.png

Interesting!

uronito, 2017-02-18:
@kayno,

The pattern is always the same?

Only happens to you with the temperature?

I say it because we could implement these changes in the code too, maybe those bytes are important.

regards

kayno, 2017-02-18:
i'd need to do more testing to confirm it's always the same, e.g. always comes through with the same values for each mode, etc.

SwiCago, 2017-02-18:
@kayno, I have also seen vane come back on temp packet and settings packet, when I use IR. I thought it was wideVane, but it is the regular vane. My HP has two vanes, a left and a right. I can only control them together, however IR can control them both. Hence why I think we may have to look at IR protocol and try sending a valid I packet to serial(cn105) via custom packet. See if it changes anything. I highly doubt mitsubishi used to different decoder functions for IR and serial, I am willing to bet all packets received by IR or serial end up at the same decoder. This decoder understands all the different codes sent and does what it needs to according, like when we get 0x03 or 0x02 from heatpump. We know what the next bytes represent

lekobob, 2017-02-20:
@kayno   The 17th byte in the settings info packet is a .5 deg resolution of the temperature using this formula:  (17th byte in dec - 127) / 2.  
Example (0xAE=174) 
(174-128)/2 = 23

On my heatpump that type of value is returned in the temp info packet as well, I just can't remember which byte.  I cant take credit for this, Jacob from the nicegear comments found it I am just connecting the dots.

SwiCago, 2017-02-20:
@kayno, 17th byte is not wideVane as lekobob mentioned
@lekobob please see #11 Temperature I noticed that too

PR #26 (2017-02-22): wide vane (horizontal vane) is now working!
- the wide vane setting is sent on byte 18 (assuming first byte is 0)
- the bitmap control byte value for this packet is set in byte 7 (not 6 as for the other settings) and has a value of 0x01

This resolves #7 and is being discussed in #25 as well.

I also did some work on the custom packet sender to make it work. Currently (for example) I send a payload over mqtt of:

```
{"custom": "41 01 30 10 01 08 01 00 00 07 02 00 00 00 00 00 00 08 00 00 "}
```

note that the space at the end is important, or else the last byte isn't sent
closed 2017-02-22 by SwiCago