SwiCago/HeatPump #8: Room temperature

jarrod180, 2017-02-15:
I thought it's a good idea to raise a new issue for improvements or changes to the room temp.

Is there a reason it sends every 60 seconds? I thought the MQTT philosophy is to send when it changes, same way the other settings work.
I already modified and tested the following changes:
-create callback for room temp modified
-separate room temp from the settings struct
-trigger the callback when room temp changed with mqtt retain set to true

What do you guys think??

kayno, 2017-02-15:
@jarrod180 60s was just a number I chose, and I've seen a lot of MQTT examples where sensor values are sent periodically, rather than only when they change. I know for current settings we don't send them periodically, however I more see these as a "status" of the heatpump, and not a value of a sensor. 

I'm not sure on what's the best approach. Reading http://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages, it says:

> **A retained message makes sense, when newly connected subscribers should receive messages immediately and shouldn’t have to wait until a publishing client sends the next message.** This is extremely helpful when for status updates of components or devices on individual topics.

This seems to match nicely with the heatpump being the device, and it has it's individual topic, which just has a retained message showing the last known status of all the different settings. 

The article then goes on to say:

> The same is true for clients, which send data in intervals, temperature, GPS coordinates and other data. **Without retained messages new subscribers are kept in the dark between publish intervals.** So using retained messages helps to provide the last good value to a connecting client immediately.

It talks about "clients" sending data in intervals, and mentions temperature. Not sure if that is justification for periodic updates or not, but I do agree that it would be good to have the message retained on the heatpump/temperature topic, for newly connected clients so they don't have to wait.

I'm not sure either way if it should be a callback for temp or not. I struggle to think of a reason about why it should be different to the settings coming from the heatpump though. That said I also worry about the readability/maintainability/usability of the callbacks, and having many of them might be messy?

One reason I can think to keep it on the 60s (or whatever number, that really should be changed to a constant in the code that the user can set, much like their SSID or topic names) is that it's kind of like a heartbeat for the heatpump - you know it's still working if you have regular updates. Even in the application that I have consuming the temperature topic (home assistant), when I click on the temperature I get a nice little graph and a timestamp telling me "updated 30s ago", so I know it's still communicating with the heatpump.

If you have retained messages and the device dies for whatever reason, there is no way for newly connected clients to know this - they will assume it's ok because the retained message says so, but it would never update. I guess that's a reason to scrap the callback for currentSettings too, and have it start polling to ensure it's still alive! 

Or perhaps both settings and temp should poll and have callbacks - periodic 60s polling to ensure it's still alive and make sure a settings change wasn't missed, and also callbacks to catch the settings calls as they happen, e.g. when someone changes the heatpump with it's IR remote control instead. In fact I kind of like this idea - imagine if your wifi dropped out and you missed an update from the heatpump because someone changed the settings with the remote - when the wifi came back up, you'd still be relying on that retained message that was now stale...

Thoughts!?
 
Does [last will and testament](http://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament)? I haven't looked into this yet, so I might have a read :)

jarrod180, 2017-02-15:
I guess I was failing to see the benefit in sending the temperature constantly, since the retained MQTT message should theoretically always be up to date?? But i can see the merit in getting regular updates.
The benefit of a callback for me is also, if the application needs to trigger something immediately on change, it can do that without storing states and looping on the get method.

BUT there seems to be a few scenarios here

- ESP publishes heat pump settings, or room temp and then dies Last messages sent with "retain" will be received by any subscriber. Subscriber gets potentially stale data. It looks like the retained message won't be deleted. Stale data might be better than no data at all, I guess ESP should get current settings and publish when it starts. 

This seems like a pretty good idea, we should think about adding it:

> Best Practices – When should you use LWT?
LWT is ideal for notifying other interested clients about the connection loss. In real world scenarios LWT is often used together with retained messages, in order to store the state of a client on a specific topic. For example after a client has connected to a broker, it will send a retained message to the topic client1/status with the payload “online“. When connecting to the broker, the client sets the LWT message on the same topic to the payload “offline” and marks this LWT message as a retained message. If the client now disconnects ungracefully, the broker will publish the retained message with the content “offline“. This pattern allows for other clients to observe the status of the client on a single topic and due to the retained message even newly connected client now immediately the current status.

- imagine if your wifi dropped out and you missed an update from the heatpump because someone changed the settings with the remote - when the wifi came back up, you'd still be relying on that retained message that was now stale...
Whatever happens the ESP should know the current settings and transmit them, if the ESP wifi drops out, does the MQTT library queue the messages until connected again? If the subscriber wifi drops out and you miss it .... connect back up again and the retained message should be published to you immediately shouldn't it?

SwiCago, 2017-02-15:
Sending temperature at intervals is important, in order to data log and setup indoor/outdoor temperature maps. With the data, you could change the behavoir of your controlling software to more closely meet the needs of the occupants.
Ex: I have hydronic heat, these radiators take time to eat a room. With temperature logging I can see how long it takes to get the room to reach desired temperature for a given cold day. If I want 70F at 7AM in a room, I have to kick on the demand for heat earlier, when outdoor temps are lower. To know when I need to do this, requires data logging.

SwiCago, 2017-02-15:
Also, I'd like to note that the library has nothing to do with the interval. That is in Kayno's MQTT demo. The library should not have any constraints like that, as you never know how it will be used by others. Someone else will use the library with a completely different application in mind. I am closing this, as example sketches should not be a discussion point, since it is up to each user to do as they please with their surrounding code using the library. Let us concentrate on library issues only, unless an example sketch has a serious issue.

kayno, 2017-02-16:
Can we keep this open @SwiCago, to discuss any changes that need to be made to the example script (which even though I wrote it, I'd still like it to be part of the library, as an example of it's use!)??

There might be merit in adding a callback for temperature, especially if you want to know sooner than 60s (if that's the interval).

I'd also like to investigate LWT too. And probably retain the roomTemperature message.

SwiCago, 2017-02-16:
Kayno, it stays as part of the library, but I don't see it as an issue with the library. It is an example sketch. 
I'll re-open it for a while, for discussion points. But it will be closed sooner or later.

SwiCago, 2017-02-16:
BTW: I updated the library to add call back for temperature. Please sync , if you have not done so. I grabbed it out of Jarrods fork and fixed a few things, such as calling back only if it is defined.

kayno, 2017-02-16:
I will merge your changes into my fork. Also, somehow in githum you can link commits and issues and pull requests, so that we can see which commits/pull requests closed an issue or were related to an issue. More that just referencing them within the text of these comments too. I need to work out how to do that - perhaps we need to add issue URLs into commit messages or something...

SwiCago, 2017-02-16:
@kayno Don't over think the project. It is not like we work for some corperation and need to be thorough with every update. Document the crap out of it, etc.... I also think the average user, who uses this library know what they are doing, otherwise they wouldn't risk their investment. Like myself, my setup is 19k$ for 6 zones, ducted and none ducted. I knew more about the physics and optimization, than my installer lol.
Again, do what you can, you bring a lot to the table and have great ideas, very much appreciated...But do take time to step back and relax a bit.

SwiCago, 2017-02-17:
According to @Kayno, latest merge fixes this issue

SwiCago, 2017-02-17:
Opps closed wrong issue

PR #14 (2017-02-18): mqtt example: roomTempChangedCallback and settings header file
- made use of the roomTempChangedCallback callback to send current room temp as it changes. 60s periodical send is still there too
- also added retain flag to room temperature message
- these changes should close https://github.com/SwiCago/HeatPump/issues/8
- added SEND_ROOM_TEMP_INTERVAL_MS constant (set to 60000ms - 60s) for setting temperature update interval
- added wifi/mqtt/etc settings to a header file, to make maintaining the example easier when pushing to github!
- few other general formatting and removed unused lastSettin [... cut]
closed 2017-02-18 by SwiCago