SwiCago/HeatPump #9: wantedSettings need to default to currentSettings once known

kayno, 2017-02-15:
Currently when the code starts, `currentSettings` and `wantedSettings` is set to:

- POWER: OFF
- MODE: FAN
- TEMP: 22
- FAN: 1
- VANE: 1
- WIDEVANE: SWING

Once connect is done and we are syncing, only `currentSettings` is updated, not `wantedSettings`. This means if you call `setPowerSetting("ON")` and call `update()` to make it happen, your heatpump is set to ON with the other settings as the defaults, and not what the `currentSettings` are based on the last time the unit was running (in my case this is usually my "default" setting: AUTO/another temp/SWING/SWING/SWING). 

This becomes a problem when the likes of home assistant has consumed the retained status message. You can see this in the messages that occur (annotated to show what's happening:

```
$ mosquitto_sub -h localhost -p 1883 -v -t 'heatpump/#'
heatpump {"power":"OFF","mode":"AUTO","temperature":27,"fan":"AUTO","vane":"SWING","wideVane":"SWING"} // last known settings of heatpump, matching what it was last set to

// ESP8266 boots up

heatpump/temperature {"roomTemperature":24} // reports temp every 60s
heatpump/temperature {"roomTemperature":24} // reports temp every 60s
heatpump/set {"power": "ON"} // in home assistant, I press the switch to switch it on (leaving all other settings as they are, which currently replicates retained status message above). This sends the power command which calls setPowerSetting() which just changes wantedSettings.power, and then update() is called
heatpump {"power":"ON","mode":"FAN","temperature":22,"fan":"1","vane":"1","wideVane":"SWING"} // heatpump now on, but all the other settings have changed because wantedSettings was still on the defaults for the others, and hadn't been synced

// I then go and set the other settings how i want them
heatpump/set {"temperature": 22}
heatpump {"power":"ON","mode":"FAN","temperature":22,"fan":"1","vane":"1","wideVane":"SWING"}
heatpump/set {"mode": "AUTO"}
heatpump/set {"fan": "AUTO"}
heatpump {"power":"ON","mode":"AUTO","temperature":22,"fan":"AUTO","vane":"1","wideVane":"SWING"}
heatpump/set {"vane": "SWING"}
heatpump/set {"temperature": 25}
heatpump/set {"temperature": 26}

// eventually it's back to my "default" settings
heatpump {"power":"ON","mode":"AUTO","temperature":26,"fan":"AUTO","vane":"SWING","wideVane":"SWING"}
heatpump/set {"power": "OFF"} // and I power down the heatpump
heatpump {"power":"OFF","mode":"AUTO","temperature":26,"fan":"AUTO","vane":"SWING","wideVane":"SWING"} // and get confirmation with a new retained message showing the status
heatpump/temperature {"roomTemperature":24}

//  next time I power it up, it's all good, because wanted settings has been updated with each change above.
heatpump/set {"power": "ON"}
heatpump {"power":"ON","mode":"AUTO","temperature":26,"fan":"AUTO","vane":"SWING","wideVane":"SWING"}
```

I think `wantedSettings` should be left as unset and not set by the constructor, and then in `getData()` when a settings packet comes in we check to see if it's unset (indicating script has just started and this is the first settings packet we have recived), and set it to `currentSettings`.

When I have some time I will code this up - just wanted to get it logged as an issue while it's on my mind!

SwiCago, 2017-02-15:
Sounds like a good idea. I used the default wanted for my webapp, however it is the webapp that should decide what the default settings are.

kayno, 2017-02-17:
The commit that fixes this also fixes and relies on other commits I have made, so if anyone wants to test before I submit them all in a pull request to @Swicago's repo, please do so :)

SwiCago, 2017-02-17:
closing as per latest merge @kayno

closed 2017-02-17 by SwiCago