SwiCago/HeatPump #13: Please list you unit numbers that work

SwiCago, 2017-02-17:
Hi guys, I would like to use this area to list unit types that are know to work with this library.
Please state Region and models that you have personally tested with. If you can provide weblink of unit that would help. I'll start
www.mitsubishicomfort.com/
North America - [MSZ-FH](http://meus1.mylinkdrive.com/item/MSZ-FH06NA.html) (Wall unit)
North America - [SEZ-KD](http://meus1.mylinkdrive.com/item/SEZ-KD09NA4R1.TH.html)  (Horizontal Ducted unit)

uronito, 2017-02-17:
Europe (Spain) - PEAD - RP100 - JA(L)  -----  Works perfect with last version.

kayno, 2017-02-18:
Australia - MSZ-GE80VA2 ("Split system" - Wall unit)
Australia - MUZ-GE80VA2 ("Split system" - Outside unit)

This is one system - unit on the wall, connected to the bigger unit outside.

lekobob, 2017-02-18:
US - MSZ-GL [http://meus1.mylinkdrive.com/item/MSZ-GL06NA-U1.html](url)

schotime, 2017-02-20:
Australia - MSZ-GE35VAD

SwiCago, 2017-02-20:
@kayno @uronito @lekobob @schotime , thank you all for helping contribute to this library. Making it be a works for everyone, is always a group effort. What works just me, may not be ideal for others, so I was quite surprised and humbled to see that this library got traction! And that we have a good group going to make it work for everyone. So again thanks for making this library what it is so far and thanks for posting up supported units.

kmdm, 2017-07-09:
Works on MSZ-SF23VE (UK) So I'd venture MSZ-SF*(VE*) at least. given 23 reflects the power of the indoor unit.

kmdm, 2017-07-17:
I guess the models listed here are probably compatible too:-

http://innovations.mitsubishi-les.com/en/controls/wifi-adapter

SwiCago, 2017-07-18:
@kmdm so you are in the UK? Good to know that the library works for your units as well. I will update the list soon.

kmdm, 2017-07-18:
@SwiCago Yep - UK.

rtwfroody, 2017-10-03:
Sorry if this is not the right place to ask a question.

How would this work for the indoor/outdoor units (eg. PVA + PUZ)? It doesn't make sense to pass a desired temperature since both units will be far away from the area that you're heating.

AlbertHakvoort, 2017-10-03:
You can use the thermostat of the unit 
[image]

or any other "Home Automation" temperature sensor (Zwave / esp8266 etc)

rtwfroody, 2017-10-04:
Thank you, @AlbertHakvoort. So digging a bit more, I think I can call setTemperature() to set the preferred temperature, and then periodically call setRemoteTemperature() to let the heat pump know what the temperature in my living room is. Is that right?

SwiCago, 2017-10-17:
@rtwfroody that is correct. I have 2 ducted(living, kitchen/dining) units in my attic, where internal temp sensor is way off obviously! I use two esp8266 coupled with dht22 sensors to send remote temperature to the esp8266 controlling each unit. I hid the esp8266 and dht22 combo in a faux candle. I also use the temperature readings from all 6 zones in my home to control my hydronic heating system. All automation, no thermostats. Works very well. I have pics of my entire installation, just need to find to blog it.

rtwfroody, 2017-10-18:
Returning this issue more to its intended subject, I can confirm the library works with the unit below:
North America - PVA-A36AA7

I'm still far from finished, but I'm building a stand-alone thermostat and it's at least able to heat to a minimum temperature with this unit.

Prinsessen, 2017-11-08:
Mitsubishi MSZ-FD25VA - European/Danish Model

Works perfect :)

cams-Nexion, 2017-12-14:
Australia MSZ-GL80VGD

Running on a Wemos D1 Mini (probably a clone). No pullups on rx/tx.

Working well

mydewji, 2017-12-19:
@SwiCago -> would love to see a blog post of how you did it. Very interested in setting up the remote temp sensors and updating via homeassistant as my internal temp sensors are not the greatest (and that's putting it nicely)

scottdalton, 2018-03-02:
North America - MSZ-GL18US

dzungpv, 2018-05-30:
Anyone try with Asia version start with MSY-JP* ?

SwiCago, 2018-06-02:
@dzungpv  If it has a cn105, which according to  http://www.mitsubishicomfort.com/kumocloud/compatibility it does, it should work fine

vlad36N, 2018-07-04:
North America - MSZ-GL18NA and MSZ-GL09NA

F43RY, 2018-08-01:
Italy - MSZ-SF25VE3 - Working

ajobbins, 2018-08-09:
Australia - MSZ-GL50VGD (indoor) and MUZ-GL50VGD (outdoor) working.

Using a Adafruit Feather HUZZAH with no voltage regulator or pullups.

miguelangel-nubla, 2018-09-13:
Spain - MSZ-FH25VE - Working
Using a ESP-WROOM-02(esp8266 chip) and a voltage regulator. Nothing else.

sklarsky, 2018-10-22:
Glad to see that the FH units are working since I've got 3 of those going in this week. I plan to test out on my one KJ unit and will report back (all indications are that it should work as well). I'm putting mine together with ESP-01 and [this 12v-3.3v regulator](https://www.amazon.com/Waterproof-Converter-Step-down-Regulator-Transformer/dp/B00C64ZJYU?crid=2M4JVWLF989N2&keywords=3.3v+regulator&qid=1540233452&s=Electronics&sprefix=3.3v+re%2Celectronics%2C134&sr=1-9&ref=sr_1_9) instead of using the 5v, just to make it easier for myself.

nantostars, 2019-01-24:
Italy, working with these units:
msz-ge35va
msz-ef35vew

I used WEMOS D1 MINI chips

unixko, 2019-01-27:
Asia - MSZ-EF09VAS / MSZ-EF13VAS

I also use Wemos D1 mini board. It is not only cheaper than using ESP-01 + 5V adapter but also supported OTA after tucked into wall unit thanks to 4M flash.

For new builder, don't be scared by demo circuit.  You can simplified project by using only 2 parts:
- Wemos D1 mini - already supported 5V input and USB-to-serial port
- Revolectrix Cellpro-JST-PA-Battery-Pigtail-10-5-Position
and then direct soldering cables into board.

[image]

SwiCago, 2019-01-27:
@unixko nice adapter.

@EVERYONE, thank you all for giving details to your setups. I need to review them all and add them to the list. But this is a great spot for folks to check confirmed locations and units.

PiR43, 2019-01-28:
Europe (France) - MSZ-GC works but need to change the bitrate from 2400 to 9600 in the library. I have found the bitrate fix here : https://www.esp8266.com/viewtopic.php?f=29&t=13207&start=24.
@SwiCago I think try change bitrate on a connect faild could be helpfull for other people who have a unit not working at 2400. (I will try do a PR in few days).

logon84, 2019-01-31:
Europe (Spain) - MPEZS-60VJA working perfectly here

YellowSub, 2019-01-31:
Europe (France) - MFZ-KJ25VE works well in 2400 Bd. 
I use a Wemos D1 mini (Lolin, V3.1.0), and I cut the link between ESP pin RX and CH340C Pin TXD. To Flash the Wemos, I reconnect the link, of course!
 I have two pull up resistors, the one of the RX line is on 3.3V, the other one (Tx line) is on 5V.

Nebukadneza, 2019-02-16:
Europe (Germany) — have MFZ-KJ25 up and working out of the box. Thanks!

jerryhjones, 2019-02-18:
I just successfully wired up my MLZ series unit (MLZ-KP09NA Flush mount).

I used an Adafruit HUZZAH32 Feather, with pull-ups to 5v.

My initial efforts without pull-ups resulted in being able to control the unit, but no data was being received. This was different from my MSZ (wall-mount) units, which are using the same board, without pull-ups.

Marc--, 2019-02-24:
Hello,

Europe (France), working on a MFZ-KA50VA. Wemos d1 mini, no pull up resistors.

Now i need to understand how to use it with Jeedom.

Marc--, 2019-03-05:
Hello, Works great too on a MSZ-GE42VA. Now i've understand how to publish with MQTT, it's awesome ! Thanks !

fer8sp, 2019-04-02:
Europe - Spain
Indoor units MSZ-AP20 & MSZ-AP35.
Working perfect with esp-01S(1M), 5V adapter & MQTT & OTA

hawkefly, 2019-04-03:
Works on a MSZ-BF09VC (China). But , there is no connector on CN105. I  had to soldering a connector to the board.The connector is B05B-PASK-1.

[image]
[image]

hawkefly, 2019-04-05:
Works on a MSZ-ZF12VA (China). You need solder a connector to CN105 as I mentioned above.

wellsy57, 2019-04-20:
Works fine on MUZ-GE60VAD (Australia) CN105 found existing on board.

[image]

Cheers to all who have contributed!

luiscgalo, 2019-05-02:
Hello all,
Thanks for the valuable information provided in this repository. ;)

I confirm that this protocol works in latest generation PEAD-M140JA ducted units (european version) with R32.
The only small deviation is related with compressor frequency since referred value is placed in a "data[5]" instead of "data[3]" in packet 0x06.

I confirm also that this protocol works perfectly with MSZ-SF25VE and MSZ-SF35VE wall mounted units.

Regards,
Luís Galo

jameshilliard, 2019-05-15:
@SwiCago I'm looking at getting some SEZ-KD ducted units installed, how do the mitsubishi wired thermostats connect to the control board on those units?

I'm trying to confirm the wired thermostats won't block the CN105 ports so that I can use them in combination with this and so that the SEZ-KD ducted units can still be controlled manually via the wired thermostats if the home automation controller and/or Arduino is off.

From what I can tell the SEZ-KD units would use a wired thermostat which appears to connect to the TB5/TB15 lines(although the service manuals I found are super confusing and don't seem to actually tell you how they are wired).

luiscgalo, 2019-05-16:
@jameshilliard, I think that I can help clarifying your question.

I've a similar setup in operation with PEAD-M ducted units where I've a wired remote controller (PAR-31MAA) and a custom control circuit via CN105 port connected at the same time.

This setup works perfectly and both control systems can operate at the same time.
Basically, the last action performed on the controllers ("wired" or CN105) defines what the AC unit should do.

If, for instance, you change something via CN105 port, the wired controller display will update its contents after a few seconds.
On the other hand, if you change something on the wired controller you are also able to check current status of the AC unit reading it's current status via CN105 packets.

In your setup it should work exactly the same. :)

ajobbins, 2019-05-16:
@luiscgalo but doesn't the PAR-31MAA also need to connect to a CN105 port?

luiscgalo, 2019-05-16:
@ajobbins No. Wired remote controllers such as the PAR-31MAA connect to a different port /bus in the indoor AC unit. It's a two pins, non-polarity, connection...

The CN105 port is used for connection with other type of controllers or interface modules such as MHK1 or MAC-567IF-E.
The gold rule here is that you can only have one device connected to the CN105 port.

ajobbins, 2019-05-16:
@luiscgalo ah. Any idea if they work with non-ducted (e.g. high wall) units as well? I really wanted wired controllers rather than IR remotes, but understood the wired remotes used CN105. But maybe your controller only works on ducted units?

luiscgalo, 2019-05-16:
@ajobbins Yes, it works exactly the same 😊
I've also a similar setup working with MSZ-SF25VE wall units.

You can operate at the same time via UR command or CN105 port.

The only small different detail here is that the operation state presented on the IR command display could be incorrect if you force a change via CN105.
I.e. if you change something via CN105, the IR command will not be informed about that  since it is a unidirectional device (with wired controller this does not happen because there is bidirectional communication)

Anyway, like I said, you can use IR command and CN105 port for controlling the AC unit.

jerryhjones, 2019-05-16:
> @jameshilliard, I think that I can help clarifying your question.
>
> I've a similar setup in operation with PEAD-M ducted units where I've a wired remote controller (PAR-31MAA) and a custom control [...]

@luiscgalo I have been thinking about doing _exactly_ this. Can you share anymore about your approach?

ajobbins, 2019-05-17:
> Anyway, like I said, you can use IR command and CN105 port for controlling the AC unit.

Yep, aware of that (and the limitations of an IR push) - that's the set up I have today: IR remote + ESP controller with this software. What I would like (but seems not possible with non-ducted units as far as I can tell) is a WIRED wall controller, plus still the ability to plug an ESP8266 in via the CN105 port.

SwiCago, 2019-05-20:
This thread is used only to list working units, not have discussions about hardware. Use the gitter chat for that, unless it is an issue where the library does not work. In which case open an issue.

SittingDuc, 2019-05-31:
New Zealand. MSZ-GE25VAD2 works fine. 
I think MSZ-GE42VAD2 works also, but I kinda blew up the onboard 0603 12V 3A fuse shorting cn105 by mistake, so work is ongoing.

(ESP01 with vreg and level shifters, JST-PA and pins from DigiKey)
Big thanks to everyone.

Tobias007, 2019-06-08:
Have here an Asian SEZ-KD35VAL. with CN105 - unfortunately seems to be no signal on the CN105. Can't receive anything on the CN105 TX. Just for understanding - does the PCB send regularly info on the TX or is it only a feedback after I have transmitted something on RX?

Tobias007, 2019-06-08:
Correction: When I send commands. The AC switches ON :)
However don't receive anything back from the AC. I've used a 10K Pullup between AC/TX and 5V.

SwiCago, 2019-06-10:
@Tobias007 the makes of esp8266 years ago confirmed all inputs are 5V tolerant. Just not core voltage!

jascdk, 2019-06-15:
Hi all :)

I want to share my setup for Home Assistant:

- My unit is an European Mitsubishi wall mounted heatpump model number: MSZ-SF25VE2.

Controlled with:

- ESP-1 module (based on ESP8285 - think it is a newer version than the ESP-01??)
- ESP-01 Breakoutboard with integrated level shifters and voltage regulator.

CN105 Connector/cable setup:

- Wire made of standard CAT5 ethernet cable with the PAP-05 connector. At the end to the ESP i 
  crimped on a male RJ-45 plug  (used pin 1 +2 and 7+8 ).

- Then i soldered an female RJ-45 connector to an RJ-45 breakoutboard - soldered some pin headers 
  onto that.

The cable and ESP are connected with dupont connectors.

I first wanted to make a 3d printed case for the module and place it outside the heatpump, but i found out it could be tucked away of sight behind the cover, so I choose that option.

I run the latest code, and it runs awesome! No problems or whatsover for now - very happy! Big thanks to @SwiCago 

Here are some pictures of my components and actual project:

[image]

[image]

[image]

[image]

[image]

[image]

[image]

cpsskipper, 2019-06-23:
Guys, I ask for help.
Air conditionner MSZ-HJ35VA er2 + WEMOS D1 mini. Not work.
Connected according to the instructions. I tried with 10K on TX and RX and without. 
hp.connect(&Serial) trying to connect but nothing happens. Only {"packetSent":"fc 5a 01 30 02 ca 01 a8 "}
What can i do to connect to the air conditioner? 
[image]
[image]
[image]

PabloAriza, 2019-07-24:
Morning 

My units are MSZ-HJ.  Anyone knows before hand if this project works? I’ve seen the P.C. board has a cn105 connector but I’ve also seen that Mitsubishi doesn’t support this model in their WiFi Device or cloud service.

Any clue?

Thanks for your time.

SwiCago, 2019-07-25:
@PabloAriza can't say for sure, but about $10 worth of parts will tell you for sure. I say give it a try, doesn't cost much, so not much to lose

PabloAriza, 2019-07-29:
So both MSZ-HJ and MSZ-DM series work!

Thanks!

SwiCago, 2019-08-02:
@PabloAriza glad to hear they worked.

aignashev, 2019-08-22:
Does it work with SEZ-M25DA?

[285 more replies]