SwiCago/HeatPump #18: Adding auto update

SwiCago, 2017-02-20:
A feature, to allow end user to set library into auto update mode.
Auto update should occur witha slight delay to ensure that end user finished setting individual settings. Default should be false, just like any library that has automatic features, such as db auto commit. 
We can modify constructor to set auto update on at start, if that makes sense

SwiCago, 2017-02-20:
@kayno post some suggestupions here if you want them.
I say again 
If auto == true && wantedSettings != currentSettings && updateTimedelay > 10ms
Update

kayno, 2017-02-20:
reposting here (tnaks for creating new issue!)

i hear you about commits in db land, but i think this is different. Different in the sense that if you are calling a function heatpump.setPowerSetting("ON"), I think anyone would expect that to switch on the heatpump.

the library never calls a set function itself (apart from setSettings() which calls all the individual sets, but setSettings() is never called by the library), so there is no risk of doing an update against a user's will - they choose when to call update()/set functions/sync()/etc...

thinking about this from another way, what is the advantage for the end user to be able to call any of the set functions, and then update()? what would they want to do in between? this might help us decide whether we even expose update() to the end user...

uronito, 2017-02-20:
Hi @kayno ,

My heatpump max temp  in heat mode is 28ºnot 31º . 28-----> (0x00 in my heatpump), for that reason i had said that wantedsettings when first time connect to heatpump is empty. For that reason i have made this modification,

Now when I send a change to the heat pump by mqtt (either mode, fan, or temperature) the pump makes the change and memorizes it if it is off and when I send the command on by mqtt, it is when the pump launches the Changes it had modified)

Regards

SwiCago, 2017-02-20:
@kayno, I thought about it and delays would not be necessary, as sync happens only during loop, so if a user sketch uses setSettings or individually sets more than one setting, it won't update/sync till they are done.
Now however, if they send another command later(just after a loop), then 1 second wait worst case.

kayno, 2017-02-20:
Sorry @SwiCago, I don't understand what you are getting at here, sorry. Are you saying the delays will/wont allow for auto update? 

Wouldn't adding autoupdate remove any issues with calling multiple set functions and sync() getting in the way of each other, as sync() could never happen between a set function and update()?

I can see that if a user wanted to set all the settings to something new and used the individual setter functions, and each of those called update() automatically, then it might be slow (there is a delay(), maybe 2 delay()s when doing an update()) but that wouldn't break anything, would it? It would just be education through documentation if that were the case - "if you want to update just one setting, you can use an individual settter function. or if you want to update multiple settings, best to use setSettings() and pass a heatpump struct".

SwiCago, 2017-02-20:
@kayno can you try the following;
in .h file 
`
    bool autoUpdate;
`
in .c file change the following 
`// Constructor /////////////////////////////////////////////////////////////////

HeatPump::HeatPump() {
  lastSend = 0;
  infoMode = false;
  autoUpdate = true;
}
`
and 
`void HeatPump::sync(byte packetType) {
  if(autoUpdate && !wantedSettings && wantedSettings != currentSettings && !packetType) {
     update(); }
  else if(canSend()) {
    byte packet[PACKET_LEN] = {};
    createInfoPacket(packet, packetType);
    writePacket(packet, PACKET_LEN);
  }

  readPacket(); 
}`

I think this will do autoupdate only if allowed and wantedSettings has settings and they differ from current and we are not asking sync to force get a specific info type

SwiCago, 2017-02-20:
@kayno let me know if the above works for you in MQTT and I'll add a setter function that allows a user to turn off autoUpdate and commit.

kayno, 2017-02-20:
So update() is done on sync() rather than the set functions?

I am interested to hear your thoughts on this idea?? :)

SwiCago, 2017-02-20:
@kayno did you try the above code snippet. Otherwise I can commit it and then revert if it doesnt work

schotime, 2017-02-20:
I think this is how the original python snippet was written. I don't mind it. Guarantees no send/receiving conflicts.

kayno, 2017-02-20:
I'm unable to test it right now. Looking at it though, I think it will work from a "will it compile" and "logical" point of view.

However I'm still not certain if this is the best solution. I still think setSettings() and individual setters should call update(). Maybe it's just me though, and no-one else thinks so! I am just trying to think about it from the point of view of making a really easy to use library, with functions that are self explanatory and don't rely on certain combinations of function calls.

kayno, 2017-02-20:
@schotime do you mean the original python did seperate calls or it's sync called update?

schotime, 2017-02-20:
@kayno did it in the sync if there were changes to be synced i believe.

SwiCago, 2017-02-20:
@kayno, problem is if I put in my code
setModeSetting(heat)
setTemperature(20°C)
setFanSpeed("AUTO");
Then by your logic it would send packets on each change. Each send requires the 1 second delay before the next is sent, thus forcing ESP to wait 3 seconds before it can respond to MQTT calls

That is why I suggest to do it at sync level, so it is once and done.

schotime, 2017-02-20:
I mean the consumer is still complicated. This is my current loop.

```
void loop() {
  if(sendUpdate) {
    hp.setSettings(newSettings);
    if (!hp.update()) {
      hp.sync();
    }
    sendUpdate = false;
    lastSend = millis();
  } 
  else {
    hp.sync();
    if ((millis() - 60000) > lastSend) {
      hpSettingsChanged();
      lastSend = millis();
    }
  }

  sleep(1000);
}
```

SwiCago, 2017-02-20:
@kayno, I am going to commit my latest changes, but set autoUpdate false for now, so as not to interfere with current testing everyone else is doing

kayno, 2017-02-20:
Yeah, multiple sets would cause delays.

I can't see a perfect solution here!!

Doesn't sync call update? We aren't creating an infinite loop, are we!?

kayno, 2017-02-20:
@SwiCago I suggest you create a branch to test this. Keep it out if master until we are happy with I :)

schotime, 2017-02-20:
@SwiCago i'm having a look now at it.

SwiCago, 2017-02-20:
@kayno, no it should not cause infinite loop
in sync
if(should do udate){update}
else if (canSend){do infoPacket}
exit loop

schotime, 2017-02-20:
There is a bit more to it. Mine just keeps saying the settings have changed.

SwiCago, 2017-02-20:
@schotime @kayno 
the following 
if(autoUpdate && !wantedSettings && wantedSettings != currentSettings && !packetType)
should only call if autoUpdate is true
wantedSettings has been defined (as in after first boot) 
wantedSettings != currentSettings
packetType for sync not set (this would happen after send)

@schotime you say it keeps repeating? this should not happen as currentSettings should eventually match wantedSettings

schotime, 2017-02-20:
@SwiCago doesn't !wantedSettings mean that its not set?

SwiCago, 2017-02-20:
@schotime correct, that would be the case at first boot. After first boot wantedSettings == receivedSettings == currentSettings
@kayno, I made a dev branch of my master, can you give it a go in MQTT scenario, for web app I doubt I will use auto feature. I send update command separately, but since I don't have MQTT setup yet I can not test this feature.

SwiCago, 2017-02-20:
@schotime good catch...it should be wantedSettings and not !wantedSettings, going to make that change now

schotime, 2017-02-20:
@SwiCago has to be `!!wantedSettings` for me

SwiCago, 2017-02-20:
@schotime correct !! or !wanted == false

schotime, 2017-02-20:
@SwiCago  and `&& packetType == PACKET_TYPE_DEFAULT`

SwiCago, 2017-02-20:
I just updated my dev branch

SwiCago, 2017-02-20:
@schotime another good catch, that is right, packetType changed as well....fixed in dev branch just now

SwiCago, 2017-02-20:
A little tire LOL, it is 7AM here I woke up at 4AM, couldn't sleep, so I decided to putz around with the code

schotime, 2017-02-20:
yeh. 12midnight here. time for sleep i think. still some weird stuff going on. could be me. needs more debugging.

SwiCago, 2017-02-20:
@schotime  I left it as !wantedSettings == false, seemed easier to nderstand than !!wantedSettings

SwiCago, 2017-02-20:
Ok thanks fot the test. I think this feature is more for MQTT than anything else
@kayno, let me know when you have given this branch a try

schotime, 2017-02-20:
@SwiCago well that was a journey. Still awake. Submitted a PR. Have a look see and we can discuss there. Working nice for me though. Yay.

SwiCago, 2017-02-22:
@kayno when you get a chance, test auto update feature in that control_packets branch. I'd like to close this if it works

closed 2017-02-23 by SwiCago