SwiCago/HeatPump #21: Warnings on Arduino Nano

schotime, 2017-02-20:
1. 
```
src\HeatPump.h:136:33: warning: converting to non-pointer type 'byte {aka unsigned char}' from NULL [-Wconversion-null]

  void sync(byte packetType = NULL);
```

2. 
```
src\HeatPump.cpp:324:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

  packetCallback(packet, length, "packetSent");
```

3. 
```
src\HeatPump.cpp:387:58: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

  packetCallback(packet, PACKET_LEN, "packetRecv");
```

Not awesome at cpp but suggested solutions?

1
```
void sync(byte packetType = 99);

if(packetType != 99) {
    packet[5] = INFOMODE[packetType];
} else {
    packet[5] = INFOMODE[infoMode ? 1 : 0];
    infoMode = !infoMode;
}
```

2,3 should be ok by casting?
```
packetCallback(packet, length, (char*)"packetSent");
packetCallback(packet, PACKET_LEN, (char*)"packetRecv");
```

SwiCago, 2017-02-20:
@schotime casting works in ESP, I am looking at your other suggestion and as soon as @kayno can confirm issue #18 I will commit

SwiCago, 2017-02-20:
@schotime confirm this compiles cleaner.

schotime, 2017-02-20:
Excellent. 
@SwiCago @kayno I'm not sure what you guys are finding but when an update occurs it rarely returns true the first time. 

I have this type of code at the moment.

```
if(sendUpdate) {
    hp.setSettings(newSettings);
    if(hp.update()) { 
      sendUpdate = false;
    }   
  } 
```

and it runs through this 2-3 times everytime before it returns true.

SwiCago, 2017-02-20:
@schotime me personally I never actually cared for the returns true on update LOL, as I figure with wantedSettings != currentSettings I will just send again. Comparing settings is more accurate then a confirm send. The confirm send was something @kayno said worked for his HP

By the way, your above changes have been committed. Next time, if you don't mind, do a pull request that way we can see exactly where what changes came from

closed 2017-02-20 by SwiCago