SwiCago/HeatPump #28: Over the Air update OTA

SwiCago, 2017-02-22:
@kayno 
I plan to make an example sketch that will allow OTA updates. I have been reading up on it and will probably tackle that after work. Seems pretty straight forward. It is basically two sketches mixed into one, where one part is the OTA listener and the rest is what ever you want.

SwiCago, 2017-02-22:
@kayno, hew is a nice sample mDNS that finds an MQTT server automatically, basically discovery mode
https://github.com/mrdunk/esp8266_mdns/blob/master/examples/mdns_test/mdns_test.ino
I am thinking we should have a sketch that starts as a captive portal/Access point, where user logs into a config page. They can then set WIFI user/pass MQTT ip/user/pass and it is saved in spiffs. If MQTT part is left blank, then auto discovery could be used to scan network for all MQTT servers. Then user can go into config page again, after connected to local network and select which MQTT host and set user/password (save to SPIFFS), esp restarts and connects to MQTT and connects to HP. User can still go into web interface and change settings if needed, maybe even set a admin password or offer a RESET, to go back to start

SwiCago, 2017-02-26:
@kayno tried OTA. Code seems to compile, but I couldn't for the lfe of me get my IDE to show OTA option as per wiki page
http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html
Even tried re-installing my boards.txt as per instructions and restart IDE, nothing. Bummer

lekobob, 2017-02-26:
@SwiCargo  what sketch are you using to test with?  I was able to see the board in my IDE but have never actually been able to successfully flash.  It finishes the flash and then never starts the new code.

SwiCago, 2017-02-26:
@lekobob, I think at end of flash "onEnd" you need to force reboot ESP
Can you post a sample of your OTA somewhere? I took this example
https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino
and merged it into MQTT
I don't think my problem is with the sketch, but rather with the install of IDE. Did you have to reboot your computer or not. I only restarted the IDE, but do not see OTA option in my menu, irregardless with ESP model I choose.

lekobob, 2017-02-27:
@SwiCago I was able to get the BasicOTA sample to work after a restart of just the Arduino IDE.  I used the 2.3.0 version of the library.

SwiCago, 2017-02-27:
@lekobob I will have to check what library verion I am using. I would think it is 2.3.0, since it is esp library stable. But i will double check or maybe just reinstall from scratch

closed 2017-02-28 by SwiCago