SwiCago/HeatPump #39: Other Packets

lekobob, 2017-03-04:
I have been doing some experimenting with my heat pump. I found if you send a request packet (0x42) with a 0x09 in byte 5 (0xfc is 0)  you get a packet with a couple info bytes.  Here is what I found so far.

							
Examples:
fc 62 01 30 10 09 00 00 00 01 01 00 00 00 00 00 00 00 00 00 00 52
fc 62 01 30 10 09 00 00 00 03 02 00 00 00 00 00 00 00 00 00 00 4f
fc 62 01 30 10 09 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 4c
fc 62 01 30 10 09 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 52

Byte 8:
0x04 is preheating 
0x08 is standby or waiting, In a multi head system if others are already heating and one wants to cool it            goes to this mode.

Byte 9: Heating or cooling stage.  
0x01 to 0x04, in heating 0x01 is lowest power 0x05 is highest output

Byte 10: 
Only in "Auto" mode is shows what the unit is doing 
0x01 is cool
0x02 is heat

In the examples above #1 is Auto mode calling for low  cooling
#2 is Auto mode calling for 3rd stage heating
#3 is Waiting (other units in cool while calling for heat)
 #4 is Heating mode at stage 2.  Looks the same for cooling mode at stage 2

I also purchased a MHK1 thermostat for one of my units to figure out how it works.  Because you can set it up to sense temperature at the thermostat, it sends an "external room temp" packet.  

It looks like this:
fc,41,01,30,10,07,01,00,ad,00,00,00,00,00,00,00,00,00,00,00,00,c9

Byte 8 is the room temp in the normal hi resolution formula ((n -128)/2)= deg c

the unit responds with:
fc,61,01,30,10,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,5e

For the most part the thermostat send the same commands for power, fan, vane and such.  It never uses "Auto" mode, it figures out the temp difference and sets the unit in cool or heat accordingly, and sends a setpoint for each.   

Lastly, there is a byte 5 0x04  info packet that can be requested,  I think it is a error information packet.  For me it always responds with 0x80 in byte 9.  Based on [this](https://www.intesisbox.com/intesis/product/media/intesisbox_me-ac-knx-1-v2_user_manual_en.pdf?v=2.1) 0x80 is "normal operation" .  I have no way to test this, so it is only speculation.  My thermostat requests that packet constantly.

kayno, 2017-03-04:
I've been sending custom packets too, and found similar. I have a branch with some of this that I need to push up once I merge in latest master, which I will try and get up today.

kayno, 2017-03-04:
I've found that it responds to the following "command types" or "info types" (whatever we want to call byte 5!):

0x02 - settings (we knew this)
0x03 - room temp (we knew this)
0x04 - not sure what, but it responds
0x05 - timers. i'm still determining logic
0x06 - i think this includes "operating" status in byte 9, i.e. matches the "operation indicator lamp" on the heatpump, and is 0x00 if the heatpump is "ON" but the room has reached the desired temperature and the heatpump isn't "operating"
0x09 - I've noticed it change similar to you, @lekobob, but I am yet to determine what it all means

I have not been able to find powerful, long or isave mode, however I think iSave is really just a saved set of settings so I doubt anything will be in the packets.

I've tried command bytes from 0x00 to 0x1f and only the bytes above responded on my heatpump.

I have a [commit here](https://github.com/kayno/HeatPump/commit/10b74442f93f24ca5a8065147dce63f68d016c62) that checks all 6 of the bytes, and actions the "operating" status byte (command 0x06, byte 9). This commit also changes how roomTemperature is handled - I have put it, along with the new "operating" status into a new struct called "heatpumpStatus". This will allow us to add timers and other "status" type info without having to add callbacks for each different status when it changes. I'd like to eventually deprecate the callback for roomTemperature change as it's now covered by the new status changed callback.

I will get timer info sorted next, and prepare a PR.

SwiCago, 2017-03-04:
@kayno @lekobob, great progres guys..soon we will know then the mitsu devs lol

SwiCago, 2017-03-04:
@kayno, so the timer bytes are only for reading right now? No setters yet?

kayno, 2017-03-04:
@SwiCago yes only reading. I am looking at setting now, but not sure where to start! The "command" (5th bytes) packets don't match for settings, i.e. when you read settings it's `0x02`, but when you write settings it's `0x01`. If that had been the same, then it would be logical to assume you would just set it the same to match between getting and setting.

So the dilemma I see - if I just start randomly setting the command byte and passing in timer data in the bytes where it's read, I might inadvertently set something else, and who knows what. It could have test modes, maintenance modes, all kinds of things that I may not what to activate!!

lekobob, 2017-03-04:
@kayno @SwiCago I can probably help with that.  The MHK1 can set on/off times.  That was next on my list to look at.

SwiCago, 2017-03-04:
@kayno yes, that might be the case, however powering off for 15min will clear any lock out settings/modes...i locked myself out of one of my units. Turned the power off, next day all was good again.
@lekobob does mhk1 use cn105 or one of the other ports? Different port may have different protocol, but would still be interesting to see what it sends. Kumo cloud blows, melcloud is the EU version and its app can do much more

lekobob, 2017-03-04:
@SwiCago Yes it uses the cn105 port.  From what I have seen so far is sends the same commands and uses the same responses that the library and Kumo cloud uses.

kayno, 2017-03-04:
@lekobob I will wait until you capture what the MHK1 sends to set the timers then. Removes any risk of sending a bad packet!! thanks :)

SwiCago, 2017-03-05:
@lekobob @kayno I just tried Kumo Cloud app again, it features programs for single zone and multiple zones. However I never saw it send anything to the kumo cloud device. I suspect it is stored on the cloud service and through periodically polling of the server, the device gets the command to set what ever the program was setup for. I did an event to turn on in fan mode, all it did as issue that command withing a few seconds of the set time. The MHK1 does not use internet, but it wouldn't surprise me if it just stores the events setup in its memory and not on the heatpump, like any other conventional thermostat would. But am eager to see what you see.

SwiCago, 2017-03-05:
@kayno, I think @lekobob deiscovered another Programming Setter Byte
We send the following when we want to set stuff
fc,41,01,30,10,**01**,CB,CB   (CB ==  ControlByte)
His, when getting remote temperature setting pushed
fc,41,01,30,10,**07**,CB,CB
It looks like byte 6 is used for Command Set
we know byte 2 is 41 = set, 42 = get
And bytes 7 and 8 say what byte is being sent for the command set
Byte 6 must be a command set byte.
fc,41,01,30,10,07,01,00,ad,00,00,00,00,00,00,00,00,00,00,00,00,c9
Command set 0x07 byte 1 = set remote temperature, in the sample lekebob provided it was set as 0xad
It is possible other things could be sent via that command set byte, such as timers, powerful, iSee

SwiCago, 2017-03-05:
@lekobob, you could test your 0x05 theory by removing the comms wire from indoor to outdoor unit. It should throw 6840,6841,6844 or 6845

lekobob, 2017-03-05:
@SwiCago When I have some time I was planning on doing that.   
@kayno I am afraid @SwiCago is right.  When I set an off time in the MHK1 it waits until the set time and sends the off command. 

On another note, I watched the 0x06 type return packet and found that byte 8 seems to return either the coil temperature or the difference between return air and output air.  I see if fluctuate up when the power output is higher and down when the power goes down.

kayno, 2017-03-05:
@lekobob can you post the initial connect packet that the MHK1 sends please?

SwiCago, 2017-03-05:
@lekobob, when the mhk1 sends the 0x07 + temperature, how do you get the hp to use internal temp again?
You mentioned info temp packet returns all 0x00, does it revert to hp temp sensors if removed? I wonder if it is just a matter of setting 0x07 + temp as 0x00 to get it to go back to internal sensors.
I like being able to use remote temp to control setpoints, but knowing running temp of ducted units in certain conditions can be helpful too.

lekobob, 2017-03-05:
@SwiCago  The MHK1 has a setting to change the sense location back to the indoor unit.  When I change that setting the MHK1 sends:
fc,41,01,30,10,07,00,00,80,00,00,00,00,00,00,00,00,00,00,00,00,f7
every second instead of the one with the temp.
The indoor unit always send its return air temperature in the 0x03 packet even if it is getting an external temp in the 0x07 packet.

lekobob, 2017-03-05:
@kayno 
here is the initial connect exchange:
fc,5a,01,30,02,ca,01,a8
fc,7a,01,30,01,00,54
fc,5b,01,30,01,c9,aa
fc,7b,01,30,10,c9,03,00,20,00,14,07,75,0c,05,a0,be,94,be,a0,be,a9

SwiCago, 2017-03-05:
@lekobob @kayno 
I just test the 0x07 remote temperature packets.
I discovered a few new things. If I set the temperature it is returned back via info 0x03 as lekebob discovered, however it is returned back as prior temp reading and current temp reading. Not sensor 1 and sensor 2 like we thought. I send the packet again and the prior becomes like the current. I send 0x00 and it reverts to internal sensors, where prior is what I sent manually and current is hp temp sensor, eventually both are the same again. So if we want current temp, we should be looking at second byte, not first
Actual room temp before sending remote temp
First 90 is prior reding and second 90 is current reading
fc 62 01 30 10 03 00 00 00 00 01 **90 90** fe 00 00 00 00 00 00 00 3b 

Sent
fc 41 01 30 10 **07 01** 00 **ae** 00 00 00 00 00 00 00 00 00 00 00 00 c8 
Received
fc 62 01 30 10 03 00 00 00 00 01 **90 ae** fe 00 00 00 00 00 00 00 1d 
Sent
fc 41 01 30 10 **07 01** 00 **ae** 00 00 00 00 00 00 00 00 00 00 00 00 c8 
Received
fc 62 01 30 10 03 00 00 0d 00 01 **ae ae** fe 00 00 00 00 00 00 00 f2 
Sent
fc 41 01 30 10 **07 01** 00 **00** 00 00 00 00 00 00 00 00 00 00 00 00 76 
Received
fc 62 01 30 10 03 00 00 0d 00 01 **ae 90** fe 00 00 00 00 00 00 00 10
And shortly there after
fc 62 01 30 10 03 00 00 00 00 01 **90 90** fe 00 00 00 00 00 00 00 3b 
 
So if I send a remote temp of 0x00, it reverts to what the HP reads, first temp byte is prior reading and second is current reading.

SwiCago, 2017-03-05:
@kayno @lekobob, not sure of the best way to incorporate sending Remote Temperature to the HeatPump, as it would require byte 6 to be changed from 0x01 to 0x07...And as lekebob said it returns the change on info Temp 0x03, which does not necessarily mean remote temp sensor. Also, since this is not a wanted setting, it should not be part of that struct. 
I would like to add this feature, but don't quite see a good way to do it. Any suggestions?

SwiCago, 2017-03-05:
@kayno @lekobob, I made a first version of setting remote Temperature via the MQTT sketch.
Set remote temp sensor value
mosquitto_pub -h localhost -p 1883 -t 'heatpump/set' -m '{"remoteTemp":21}'
revert back to HeatPump temp sensors
mosquitto_pub -h localhost -p 1883 -t 'heatpump/set' -m '{"remoteTemp":0}'
This is my test branch only, as again I am not sure if this is the best way to have this in the library. It was more of a "it works " concept. 
https://github.com/SwiCago/HeatPump/tree/testing

SwiCago, 2017-03-05:
@lekobob @kayno Got it, if I send
fc 41 01 30 10 **07 02** 00 00 **01** 00 00 00 00 00 00 00 00 00 00 00 74
it reverts

So to set remote temp, you send
fc 41 01 30 10 **07 01** 00 **TT** 00 00 00 00 00 00 00 00 00 00 00 00 76
and to revert
fc 41 01 30 10 **07 02** 00 00 **01** 00 00 00 00 00 00 00 00 00 00 00 74

I am updating my test branch

schotime, 2017-03-05:
What is this remote temp about?

SwiCago, 2017-03-05:
@schotime it lets you use a remote temperature, instead of HeatPump internal sensor
If you have a ducted unit, the temp could be off by 10°C or more if in attic during summer heat.
Using a remote temp sensor tells the heat pump what the room temp is, so it can decided if it should cool or not on its own.

schotime, 2017-03-05:
I see. That's pretty cool.

SwiCago, 2017-03-05:
@lekobob figured out how to turn it on, and I just figured out how to turn it back off again.
Could be useful when you want to use a remote temp sensor and then it's battery goes out or it stops sending updates. This can allow for the controller to revert back to internal sensors without power cycle or removing of controller.

kayno, 2017-03-05:
@lekobob thanks for the connect packets. I had a theory that maybe 0x07 would appear in there, but not to be. Interesting that it sends two different short connect packets, and that it also gets a short 0x7a reply (I am assuming that the 4 packets are sent, received, sent, received?)

SwiCago, 2017-03-06:
@lekobob i didnt see your edit where you mention 0x80 being sent to use internal sensor. I saw you mention it was sent every second. What happens if it stops  sending 0x80, does it stay reverted to internal sensors? I'll have to try that out as well and send 0x80 just once and see if it stays reverted. If not, I know the signal i found reverts it.

kayno, 2017-03-06:
given my unit is (what we call here) a "split system" (outdoor compressor unit, indoor wall mounted head unit) I haven't given much thought to using an external source for room temperature. However with just a little thought, I am surprised that the ducted units would actually use the temperature of wherever the compressor unit is installed (e.g. an attic) to moderate their operation! Surely they would ALWAYS use a sensor that was in a room being heated/cooled? What am I overlooking here!?

schotime, 2017-03-06:
Sometimes where that sensor is placed can be cooled quicker than other parts of the house so you may want to set the temp to somewhere else.

SwiCago, 2017-03-06:
@kayno  i think you have misunderstood. I have a 6nzone setup. 4 wall units and two in attic ducted units. 1 outdoor compressor. All the units have their own temperature sensors, but ducted units wil be obviously off due to what ever temperature the attic has. Using a external (not exterior) temperature sensor, allows the unit to know what the actual room temperature is. 
Also a high wall unit may not be accurate either, depending on how tall the room is. The difference could be as much as 2 degrees from floor to ceiling. Using an external sensor would place the temperature sensor closer to where people would actually be and not where the unit is

lekobob, 2017-03-06:
@SwiCago Because the manual mentions that if the MHK1 ever looses signal it will revert to the internal temp, I assume if you stop sending external temp packets it will automatically revert to internal temp.  Mine just starts sending the packet with 80 in it when I change the setting in the MHK1 to use internal temp.  Notice though that byte 6 (or control byte) is 00 so it is saying "I am setting nothing".
 
I made a screen cap of what the stream between the MHK1 and the heat pump looks like.  https://youtu.be/cRFqmeP7KaQ
The only other packet I have seen is a Setting Set packet when I change a setting like Mode or the setpoint temp.

SwiCago, 2017-03-06:
@lekobob, i tried that. I let it not send for 20min, it still sent back the last sent temp. It only went back when i removed the esp8266! I know this because the first few temp pakets were from hp, but shortly after it started reporting back the last sent temp of external. Power cycle or the special packet i sent made it revert

lekobob, 2017-03-06:
@SwiCago when I have a chance I will pull the batteries from the thermostat and see if the receiver sends anything new to the unit.

lekobob, 2017-03-06:
@SwiCago If I just pull the batteries from the thermostat the receiver sends that same 80 packet to the unit.
  fc,41,01,30,10,07,00,00,80,00,00,00,00,00,00,00,00,00,00,00,00,f7

lekobob, 2017-03-06:
@SwiCago  I am thinking about using multiple sensors in a room (all esp8266-01 in a little wall wart package) and averaging them to send back to the unit.

SwiCago, 2017-03-06:
@lekebob, when i get a chance tonight, i'll see if sending 0x80 0°C reverts and stays reverted... I think it is temporary, until it gets real data again. I wouldnt want the library to have to keep sending 0x80, i rather it sends a signle once to revert.
And yes multiple esp-01 with humidity temp sensor is the best way to go. Small and indescreet sensors, instead of huge ugly thermostats

SwiCago, 2017-03-06:
Off subject @lekobob, I saw in supported units you mention North America. Where are you in exactly? I'm in Chicago. Would be funny if you were close by.

SwiCago, 2017-03-06:
@lekobob Just tried sending
fc,41,01,30,10,07,00,00,**80**,00,00,00,00,00,00,00,00,00,00,00,00,f7
and this is the response
fc 62 01 30 10 03 00 00 0f 00 01 b2 **80** fe 00 00 00 00 00 00 00 1a 
sent it again and the response is 
fc 62 01 30 10 03 00 00 00 00 01 **80 80** fe 00 00 00 00 00 00 00 5b
it stays this way until I send it the magic reset
fc 41 01 30 10 07 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 75

lekobob, 2017-03-06:
@SwiCago I am in Las Vegas.  You should say hi if you ever com this way.  I hear it is a fun place to visit.

SwiCago, 2017-03-06:
@lekobob LOL I was out there 2 xmas' ago..My brother lives in North LasVegas
I used to live in So.Cal so I am familiar with Vegas ;)

SwiCago, 2017-03-06:
@lekobob do you know of a wall wart project box that would be a good fit? I have not found any online yet. Still looking.

lekobob, 2017-03-07:
@SwiCago https://www.itead.cc/smart-home/smart-socket.html might me be a good place to start.

SwiCago, 2017-03-07:
@lekobob, so you are planning to gut/reuse an existing wall wart device. Hmmm maybe use one that has esp8266 built in and just add temp sensor

lekobob, 2017-03-07:
@SwiCago Found this https://www.polycase.com/gs-2415 and https://www.itead.cc/sonoff-sc.html too.

SwiCago, 2017-03-07:
@lekobob that first looks usable, but not when outlets are not high enough from the ground.
The second "sonoff" one may work, but they are using the ESP8266 with a ATMEGA. I found the source code page for the ATMEGA and it is using AT commands, so it looks like the ESP is only used as WIFI dongle and all sensors are connected to ATMEGA. The ATMEGA code is a .ino file, so it could be modified and there is instructions on how to re-flash with custom code. Also uses DHT11 and not DHT22 and a bunch of other sensors I wouldn't need making the packaging bulky. A small disc would have looked better. But then again, seems to have a hacker following and some peeps already did MQTT and OTA for it and replaced dht11 with dht22

SwiCago, 2017-03-07:
@lekobob, here is an idea for a sleek hidden box. A faux candle. Modify the base to fit ESP8266 and DHT22.
The battery compartment could easily house electronics. Some dremmel trimming in back, so DHT22 can breath and a little trimming for micro USB port(for power)...

lekobob, 2017-03-07:
@SwiCago @kayno  I disconnected the s3 (communication) wire on one of my units today,  it took probably 5 min of trying to operate this way but finally shut down and blinked the light.  The 0x04 packet returned:
fc,62,01,30,10,04,00,00,01,68,40,46,00,00,00,00,00,00,00,00,00,6a
Based on [this](https://www.intesisbox.com/intesis/product/media/intesisbox_me-ac-knx-1-v2_user_manual_en.pdf?v=2.1) That is a indoor/outdoor reception error.  I think we can safely say it is an error packet.  Just don't know what byte 11 is exactly.

kayno, 2017-03-08:
@SwiCago nice test! Would be good to create a .h file (I think that would be best way, but open to ideas) with all of the error code mappings in there, so that the library can report them if they ever happen. Then you could setup the likes of home-assistant.io to send a notification to your phone if they were to ever occur.

Also, nice find on that manual for the IntesisBox - could be quite useful!

SwiCago, 2017-03-08:
@lekobob, great job. @kayno, yes we should have a .h with all possible codes, based on the file lekebob linked. Kayno, I have looked at the document many times over, there is not much else in it that we don't already know and it also does not say which bytes or command sets are responsible for items listed. I just says how long each item is. 1bit, 1 byte, 2 bytes etc

SwiCago, 2017-03-08:
@lekobob maybe the 11 byte is the page number in the repair manual ;) -JK

SwiCago, 2017-03-08:
here is another pdf file that lists more codes and also cross references between unit types
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=4&ved=0ahUKEwjcp67r2MXSAhVK1oMKHdycA0MQFggtMAM&url=http%3A%2F%2Fwww.orionair.co.uk%2FPDF%2FMitsubishi_elec_Fault_codes.pdf&usg=AFQjCNG8xKAyIxPFq_L2sVmPHfekNd_FqQ&sig2=fSGukVvWYQJ-fpD48qRQ0g&cad=rja

kayno, 2017-03-08:
@lekobob @SwiCago clearly i need to get my eyes tested! Sorry @lekobob, nice test!!

@SwiCago can you create two wiki pages on this repo please: "Related information" and "Supported models". That was we can add all the doco we have found that is related/helpful/etc to the first, and the models that have been submitted in #13 to the second. If you gave me access to edit them, I'd be happy to add what the stuff I have stumbled across and help maintain them...

SwiCago, 2017-03-08:
@kayno I believe you got an evite to collaborate and the initial wiki is made

kayno, 2017-03-08:
@SwiCago I did get the evite and I accepted - thanks!

SwiCago, 2017-03-08:
We should create a wiki table with the byte layout of the settings and various return data sets. I am not very good with wiki, but when I get a chance I will update. Already updated supported devices and updated readme to link it

kayno, 2017-03-08:
i can sort out the markup for tables. would be good to come up with some standard names for "packet type", "control byte", "command type", etc

SwiCago, 2017-03-08:
@kayno look at the wiki, I made a table to start us off...See if this makes sense

kayno, 2017-03-08:
I did the same here: https://github.com/SwiCago/HeatPump/wiki/Mitsubishi-protocol

kayno, 2017-03-08:
I used markup, you used HTML. I find markup tends to be easier - see what you think.

SwiCago, 2017-03-08:
LOL, leave your page, makes sense to document it separately...but see if you can add a header line like I did, below your 0-21

[65 more replies]