SwiCago/HeatPump #46: MQTT over TLS

SwiCago, 2017-03-07:
@kayno @lekobob @schotime , have any of you tried to use MQTT over TLS , so that communication between ESP and Server are secure.
I read it is just a matter of using "WiFiClientSecure" instead of "WiFiClient" and then setting port to 8883 instead of 1883

kayno, 2017-03-07:
TLS/SSL is not available on the pubsubclient library, as far as I know...

kayno, 2017-03-08:
Although in late breaking news (well, 4 days old anyway!), maybe it's coming!? https://github.com/knolleary/pubsubclient/pull/251

SwiCago, 2017-03-08:
@kayno I used  Imroy/pubsubclient and it is supposed to be supported. It is the MQTT client listed on arduino ESP page too. Adafruit also has a variation that supports it.

kayno, 2017-03-08:
@SwiCago interesting! I see that the Imroy version also supports larger messages - this is something that has caught me up several times with using our library. Initially I wanted the mqtt example (which is essentially the script I run for my heatpump to talk to home assistant) to have the timer info in a single status packet with room temperature and operating mode as well. But it went way past the 128 byte message limit, especially too because I was trying to structure the json nicely, e.g. have objects for on and off timers.

closed 2017-11-12 by SwiCago