SwiCago/HeatPump #50: Memorize mode at turn off

uronito, 2017-05-17:
Hello @kayno , @SwiCago, I've discovered a little bug that I do not know is due. If I start the heat pump in ventilation mode from mosquitto and then change the mode to cold from the control or from the thermostat, and then turn off the pump from mosquitto the pump is turned off in ventilation mode and not in cold mode as it should be, it seems Which does not store the changes made in the operating mode (fan, cool, etc ...) from the thermostat or the control when turning off from mosquitto.

Thanks

SwiCago, 2017-05-24:
@uronito you have to enable external update to true! Otherwise what ever you do with the remote is not stored in your wanted settings. What happens is your current settings as known by hp are changed by remote or thermostat, however your wanted settings are still what ever you set. When you send the full command set, then your last mode that is stored in wanted is sent as well. This is good when you want automation to trump remote. However, in your case you want it in sync with the remote or thermostat, so you must allow external update. This will over write wanted settings with current settings each time they change.

uronito, 2017-05-24:
Thanks @SwiCago , 
I had not noticed that option.

Regards

closed 2017-05-24 by uronito