SwiCago/HeatPump #51: Example to convert Celsius to Fahrenheit

mattridrm, 2017-05-25:
Are there any examples to convert between temperature units? I see a helper  but am unsure how to use it. Thanks

SwiCago, 2017-05-26:
Hi, 
It is an untested method and may need to be changed to take float instead of unsigned int. It was written before we found out that temps can be read in 0.5C increments from HP and that is why it is still int.
When you check out the library, you may need to change method HeatPump::CelsiusToFahrenheit to take a float and change HeatPump::FahrenheitToCelsius to return a float
But you would use it like so
unsigned int temp_F = hp.CelsiusToFahrenheit(hp.getRoomTemperature());
hp.setTemperature( hp.FahrenheitToCelsius(temp_F) );

Let me know and I will gladly update the repo

mattridrm, 2017-06-01:
Thanks,

I have changed HeatPump.h to:

// helpers    
    float FahrenheitToCelsius(float tempF);
    float CelsiusToFahrenheit(float tempC);

and HeatPump.cpp to:

float HeatPump::FahrenheitToCelsius(float tempF) {
  double temp = (tempF - 32) / 1.8;                //round up if heat, down if cool or any other mode
  return currentSettings.mode == MODE_MAP[0] ? ceil(temp) : floor(temp);
}

float HeatPump::CelsiusToFahrenheit(float tempC) {
  double temp = (tempC * 1.8) + 32;                //round up if heat, down if cool or any other mode
  return currentSettings.mode == MODE_MAP[0] ? ceil(temp) : floor(temp);
}

and conversion works correctly.

mattridrm, 2017-06-03:
Had some issues with the round up and down during conversions. Unit operates in 1 degree F increments. Changed to the following in Heatpump.cpp

float HeatPump::FahrenheitToCelsius(float tempF) {
  double temp = (tempF - 32) / 1.8;              
  return round(temp * 2)/2;  // see https://arduino.stackexchange.com/questions/28465/arduino-round-to-nearest-0-5}

float HeatPump::CelsiusToFahrenheit(float tempC) {
  double temp = (tempC * 1.8) + 32;               
  return  int(temp + .5);    //cast float to nearest whole number
}

SwiCago, 2017-06-04:
@mattridrm  actually all our units operate in increments of 0.5 Celsius, however the IR remotes in the US are setup if Fahrenheit by the installers.  You can change the remote to Celsius by pressing both up/down temp buttons and then press reset. Keep both up/down buttons pressed until the remote shows the clock setting, then release the two buttons. The remote should be in Celsius now. Do the same steps to switch back to Fahrenheit. The remote itself already does the math for you when in Fahrenheit mode.
When I get a chance I will update the library with the conversion. Even tough I live in the US, I don't care if it is in F or C, since I don't use any remotes. I use automation only. No thermostats, no remotes. All automatic based on some predefined variables I have on a central control software. I also use the same software to operate my hydronic heating

Moxified, 2017-06-04:
I implemented this code today and am having odd results.  It could be me but I copied the code out of your most recent change and although it "works", I find there are several temps I cannot actually use.  76 for example.  If you send 76, it sends back that it is set to 75.  There are others too.

From sub heatpump/# :

Client mosqsub/29007-Test received PUBLISH (d0, q0, r0, m0, 'heatpump/set', ... (18 bytes))
{"temperature":76}
Client mosqsub/29007-Test received PUBLISH (d0, q0, r0, m0, 'heatpump', ... (104 bytes))
{"power":"OFF","mode":"COOL","temperature":75.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}

mattridrm, 2017-06-04:
SwiCago. Yeah I know about the remote settings, unfortunately my brain has a hard time converting C to F. Celsius just does not relate in my brain. The library still forces you to set the temperature in C for the unit. That is difficult due to the rounding errors presented by floats and the floor ceil in the code.  

Moxified I found some errors with my code. Im used to python and this is my first time with ardunio. Please try the following as I have found it working well.

Heatpump.cpp

float HeatPump::FahrenheitToCelsius(float tempF) {
  double temp = (tempF - 32) / 1.8;  // round fahrenheit float number to celsius float number in .5 degree increments 
  temp = temp * 2;
  temp = round(temp);
  temp = temp /2; 
  return temp;
}

float HeatPump::CelsiusToFahrenheit(float tempC) {
  double temp = (tempC * 1.8) + 32;   // return a fahrenheit integer to the nearest whole number from a celsius float value  
  temp = temp + .5;           
  return  int(temp); 
}

Moxified, 2017-06-04:
No problem, I'm not used to much of it.  I have a basic understanding of this stuff and I hack the rest together from google.

Unfortunately I seem to have similar results.  Am I doing something incorrectly?  I substituted your changes in heatpump.cpp file in libraries\HeatPump-master\src.  I then verified and uploaded to my huzzah.

Sent various {"temperature": } and got back some good and some different setpoints:

{"temperature":74}
{"power":"OFF","mode":"COOL","temperature":73.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":72}
{"power":"OFF","mode":"COOL","temperature":72.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":73}
heatpump: update() failed
{"power":"OFF","mode":"COOL","temperature":73.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"roomTemperature":70.00,"operating":false}
{"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
{"temperature":74}
{"power":"OFF","mode":"COOL","temperature":73.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":75}
{"power":"OFF","mode":"COOL","temperature":75.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":76}
{"power":"OFF","mode":"COOL","temperature":75.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":77}
{"power":"OFF","mode":"COOL","temperature":77.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":78}
{"power":"OFF","mode":"COOL","temperature":77.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
{"temperature":79}
{"power":"OFF","mode":"COOL","temperature":79.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}

SwiCago, 2017-06-04:
Guys, don't read into the F/C too much. When you send 76F with IR, the IR remote sends a converted C tempt to the unit. The IR never gets anything back from the unit, it is a one way communication. The library however can return current sensed temperature and converting back and forth from C/F can cause rounding issues. It is never perfect unless you strictly use float and do no rounding.
If you set the unit to COOL and temperature of 76, this might translate to 75 when read back again. It doesn't matter, you sent 76 and the unit sends 24.5C. The unit knows you want 24.5C and when it reaches 24.5C it will throttle back. 
Remember there will always be conversion issues between F/C, that is why in COOL I rounded up and HEAT I round DOWN as that would save energy and in the end, you won't realize it unless you are dead set on looking at the Fahrenheit temperature. When I get a chance I will re-evaluate that part and make it round up/down to nearest 0.5C depending on Mode. 
Just note, you should use this library to constantly update the HP on what to do. You should have certain settings you want to send a certain times and let the heat pump do it's thing. The only time you should constantly(1min intervals) send something, is if you decided to use remote temperature sending. The software that sends commands is not a thermostat, the heat pumps themselves are the thermostats and do their thing. External control like a real thermostat would cause the units to constantly turn on/off which is not what these things were made to do.

SwiCago, 2017-06-04:
BTW, depending on MODE, the unit also returns a higher or lower temp. Example, if you have iSee and run in FAN mode, iSee is not used and you will get back a higher temp as the unit is high on the wall, but when you use COOL mode and iSee, iSee is active and does a temp average of the room and is generally 2F cooler than the internal temp sensor. Just some food for thought. 
I am running 4 hiWall units and 2 ducted units. The ducted use external temperature senders and the hiWalls iSee/internal sensor. All my automation does is tell the unit what setting to be in at a given time of day, I let these units decide what is best fan speed, etc...

Moxified, 2017-06-05:
SwiCago, I hear you on not sweating (pun) c/f.  BUT unfortunately C just doesn't compute for me, my wife, our guests.  It bites but F has been jammed into our heads since we were 3 in every educational establishment we attended.  I can roughly translate c to f or back in my head but it takes time, is very inaccurate and not a reasonable expectation of my wife.  Wife approval factor and all that, I digress.

I honestly don't "care" if the AC unit is .5 high, .5 low or even 1.5 high or 1.5 low compared to the F set point we are asking of it.  Not like the roomtemp is all that accurate right?  What I really don't want is to set it to a temp and have it change the set point a few seconds later to a different temp.  Sure we can learn to live with it but there isn't a reason to learn to.  In my limited way of thinking, if a suitable formula cannot be established, we are only talking about 30 c to f relations.  That could literally be programmed into a select or array or something.  Not a clean method for sure but would be super reliable.

I'm really just trying to eliminate the IR remotes and create a reasonable mimic of Mitsubishi's Kumo cloud.  The Kumo cloud offering would have worked but was SUPER expensive, relies on their cloud services and apps and is closed. Not to mention the wifi units required WPS... I don't use trash AP's so WPS isn't an option.  With this, I control everything from anywhere on my own hardware, it costs far less and I CAN automate it with my existing automation controller if I want to .

I'm 1000% with you on not trying to use the library as a thermostat.  I see too many people try to replace their thermostat's with relays, sensors and automation rules.  Never ends well.  Anyway, so you can send remote temperature to the unit for it to work with?  I need to dig into that as I have wall thermostats I would love to use as a more accurate room temp.  Feel free to point me to an issue or part of library :)

Thank you (and others) for your hard work.  I wish I could contribute but I made it through 1 semester of VB 6 and went, nope, I'm not a programmer.

mattridrm, 2017-06-05:
Moxified have you tried my 3rd post of code? Works for me as follows:

set:

"heatpump/set" -m {"temperature":74}
"heatpump/set" -m {"temperature":73}
"heatpump/set" -m {"temperature":72}
"heatpump/set" -m {"temperature":71}
"heatpump/set" -m {"temperature":70}
"heatpump/set" -m {"temperature":69}
"heatpump/set" -m {"temperature":68}
"heatpump/set" -m {"temperature":67}
"heatpump/set" -m {"temperature":66}
"heatpump/set" -m {"temperature":65}

result:
{"power":"ON","mode":"COOL","temperature":74.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":73.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":72.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":71.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":70.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":69.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":68.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":67.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":66.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}
{"power":"ON","mode":"COOL","temperature":65.00,"fan":"AUTO","vane":"AUTO","wideVane":"SWING","iSee":false}

SwiCago and everybody else thank you for the hard work on this library. It is truly appreciated.

Moxified, 2017-06-05:
Yes, post number 7

All I have to do is add the changes to the .cpp file in my arduino libraries folder and verify in IDE right?  It should pull the new changes in and then feed them up to the ESP when I flash right?

SwiCago, 2017-06-06:
@Moxified the remote temp command is 
 hp.setRemoteTemperature(float tempInCelcuis); //you must update in intervals, otherwise the HP will be stuck on last temp you sent! I recommend at least once per minute! You can revert to factory with below call.

hp.setRemoteTemperature(0); //switches back to internal temperature sensors.

Moxified, 2017-06-07:
Did a bunch more testing tonight.  I ran mattridrm's code through an UNO stepping through 50 - 90F to serial monitor and the conversions worked perfectly.

I dug into the library a little trying to understand why the HP seemed to ignore the half increments I noticed in further testing.  If I understand the library correctly, it is purposefully rounding off to a whole number degrees Celsius?

If that's the case, I would expect to see the behavior I'm getting as the .5c would be needed to keep the conversion accuracy to within 1F.  I'm not sure why mattridrm is getting better results.

I recall mention was made that support for half increments on the HP was a "recent" discovery.  Is this integer map/rounding something that could be augmented to support the half increments.

Sorry if I'm all off on this.  I'm trying hard to understand it.

mattridrm, 2017-06-07:
Im not sure what board you originally were running it on. I was running the code on a esp8226. The first board you ran the conversion on was not a UNO?...... I think that may be an error with the conversion of the DOUBLE in the original library.

From the Ardunio documentation:

double

Description

Double precision floating point number. On the Uno and other ATMEGA based boards, this occupies 4 bytes. That is, the double implementation is exactly the same as the float, with no gain in precision.

On the Arduino Due, doubles have 8-byte (64 bit) precision.

Tip

Users who borrow code from other sources that includes double variables may wish to examine the code to see if the implied precision is different from that actually achieved on ATMEGA based Arduinos.

I did not modify the DOUBLE in the original library referenced my the .cpp file. Maybe certain boards do not have enough precision to calculate correctly.

Are there any testers for the above modification? I think that this can be closed with the modification of the .cpp file to:

Heatpump.cpp

float HeatPump::FahrenheitToCelsius(float tempF) {
double temp = (tempF - 32) / 1.8; // round fahrenheit float number to celsius float number in .5 degree increments
temp = temp * 2;
temp = round(temp);
temp = temp /2;
return temp;
}

float HeatPump::CelsiusToFahrenheit(float tempC) {
double temp = (tempC * 1.8) + 32; // return a fahrenheit integer to the nearest whole number from a celsius float value
temp = temp + .5;
return int(temp);
}

It seems that any time a float is converted to an integer there will be rounding errors in C or C++. I have tried my best to limit those errors within a .5 degree radius with my changes. I am running the above conversion code with the latest .5 C changes from master Latest commit dea8188  on Mar 20.

Moxified, 2017-06-07:
I'm sorry, I muddied the waters there.  I'm using an Adafruit  ESP8266 Huzzah to talk to my HP.  It is the one having trouble that all of my previous posts and test were based on.

I was using an UNO last night to test your function thinking it would be easier for quick flashing of test changes... I didn't think of the differences in chipset so please disregard the Uno stuff.

BUT, my esp based huzzah does not work correctly with the code you just listed.

closed 2017-06-16 by SwiCago