SwiCago/HeatPump #52: Demo circuit serial voltage

Moxified, 2017-05-27:
I'm very amateur when it comes to circuits but it seems to me you are pulling the ESP serial UP to 5v?  I would assume this would blow or slowly blow the 3.3v ESP?

I'm looking to build this circuit but don't want to get into replacing ESP's often as I want to install inside of HP enclosure.

kayno, 2017-05-27:
I used the Adafruit Huzzah ESP8266, which takes 5v (and has an onboard regulator) and also has a 5v tolerant RX pin (by means of a 1n4148 diode, see schematic: https://learn.adafruit.com/assets/24745). Maybe a diode is all you need? My Huzzah has been going non stop on the heatpump for a few months now.

There is a write up on hackaday about the ESP8266 and its ability to take 5v: http://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/ - jury was out on whether it was 5v tolerant.

Moxified, 2017-05-27:
Interesting.  I had been "avoiding" the huzzah to keep things small but in thinking about it, you wouldn't even need a breadboard, you could just solder the wires headed to HP directly to the Huzzah.

Isn't the TX of the huzzah only 3v though?  Did you pull up the TX or does the HP work with 3v RX.

kayno, 2017-05-28:
I have a cable with the 5 pin JST connector on one end that goes to the heatpump port, and on the other end I have single female dupont type headers ([like this](http://www.hobbyrc.co.uk/content/images/thumbs/0000546_4pcs-200mm-dupont-single-pin-to-pin-1p-connector-wire.jpeg)) for connecting directly to the male header pins on the huzzah. I have also soldered in a 10k resistor from both the RX and TX pins to 5v, as per the circuit diagram. It's then wrapped up in heatshrink.

The TX of the huzzah is only 3.3v, but it has no problems sending commands to the heatpump.

Moxified, 2017-05-29:
Did you try running without the resistors?  I shouldn't think one is needed on the RX as it should already be 5v.  The HP might be happy enough receiving in 3.3v?

I'm going to order Huzzahs tomorrow (I have 5 HP heads) and will try without the resistors.  Just curious if you already had.

I had already read the article about 5v stable.  To me it seems iffy at best.  3 months could be long enough to prove long term functionality but it could also be dieing a very slow death...  I'm no EE so I just don't know.

kayno, 2017-05-29:
I've never tried without the resistors, as the [library.properties file](https://github.com/SwiCago/HeatPump/blob/master/library.properties) states:

> Pull TX and RX to 5V with 10k resistors, otherwise the heat pump will not accept the connection.

I'm not sure where this originated from, and I just checked the original work that this library is based on (https://nicegear.co.nz/blog/hacking-a-mitsubishi-heat-pump-air-conditioner/) and whilst there is no mention in the write up of the 10k resistors, there is a recent comment (#107) from Dexter that says:

> By the way, my one is working fine without pulup resistors to +5V. Just a direct connection to the TX/RX pins. 

So based on that one comment, it's not needed! 

What are your thoughts, @SwiCago?

Moxified, 2017-05-30:
Excellent.  I suspected they might be unneeded.  I will try with my Huzzah when I get them and report back.

SwiCago, 2017-05-31:
@kayno @Moxified , The makers of esp8266 chimmed in well over a year ago and confirmed that all gpio pins are 5V tolerant. I myself have had the esp-01 connected to my heat pumps since day 1 of this project. All are still 100% operational

Moxified, 2017-06-02:
OK, so I received my Huzzahs today.  I flashed one with the mitsubishi_heatpump_mqtt_esp8266 example after changing my wifi and mqtt server details.

It appears to talk to the HP even without the 10k pullups BUT, I'm unable to control the HP.  I even tried adding the pullups and still cannot seem to get the HP to respond to any commands.

I can see at startup in the debug topic that they appear to connect (7a changes to 62 as referenced in other issues) and then they talk back and forth constantly.  If I disconnect an RX or TX line, the chatter stops.  But any attempt to publish a power on command or fan command doesn't work.  Also, I tried turning the unit on with the remote and the status updates said it was still off.

I have tried many similar to: '
mosquitto_pub -d -t heatpump/set/power -m ON
mosquitto_pub -d -t heatpump/set/power -m OFF
mosquitto_pub -d -t heatpump/set/fan -m 1
mosquitto_pub -d -t heatpump/set/mode -m FAN

mosquitto_pub -d -t heatpump/debug/set -m off (or on) does control debug mode.

Any ideas?  I'm working with an MSZ-GL09NA which is listed by somebody as working.

I'm a hack programmer at best so I may just not be understanding the code.

Example debug topic during a Huzzah reset:

Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (41 bytes))
{"packetSent":"fc 5a 01 30 02 ca 01 a8 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetRecv":"fc 7a 01 30 01 00 54 3f e8 6c 20 40 40 ee fe 3f ae 10 20 40 88 13 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (41 bytes))
{"packetSent":"fc 5a 01 30 02 ca 01 a8 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetRecv":"fc 7a 01 30 01 00 54 00 64 13 ff 3f 60 00 00 00 00 00 00 00 00 00 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (41 bytes))
{"packetSent":"fc 5a 01 30 02 ca 01 a8 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetRecv":"fc 7a 01 30 01 00 54 00 00 00 00 00 e6 64 20 40 00 00 00 00 00 00 "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetRecv":"fc 62 01 30 10 02 00 00 00 03 07 00 00 00 00 03 b0 00 00 00 00 9e "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump', ... (104 bytes))
{"power":"OFF","mode":"COOL","temperature":24.00,"fan":"AUTO","vane":"AUTO","wideVane":"|","iSee":false}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetSent":"fc 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7a "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/debug', ... (83 bytes))
{"packetRecv":"fc 62 01 30 10 03 00 00 0d 00 00 ae 00 00 00 00 00 00 00 00 00 9f "}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/status', ... (43 bytes))
{"roomTemperature":23.00,"operating":false}
Client mosqsub/2291-OpenHAB received PUBLISH (d0, q0, r0, m0, 'heatpump/timers', ... (73 bytes))
{"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}

kayno, 2017-06-02:
It's been a couple of months since I have played with this, but I think the correct format for sending commands is:

`mosquitto_pub -d -t heatpump/set -m '{"power": "ON"}'`

All of the different commands in one go: 

`mosquitto_pub -d -t heatpump/set -m '{"power": "ON", mode: "COOL", fan: "QUIET", temperature: 28, vane: "SWING", wideVane: "SWING"}'`

Moxified, 2017-06-02:
That was the ticket.  Thanks!

I knew I was doing something wrong.  I didn't know it was communicating in json because I knew little to nothing about json.  I saw mentions of json in the esp code but didn't really comprehend what it was doing.  I know documentation is tough on projects like these but I can't find clear examples of sending commands anywhere other than yours here.  

Is there a better place for general help questions than an issue?  I may have some more trouble throughout my project and don't want to fill the issues box with my ignorance.

After reading about json and how it works with openHAB (my automation software I will be interfacing with)... I need to decide if I change the esp code to send each item on different topics or if I try to figure out how to get OH to parse json.

BUT, back to the point of this issue:  **5v pullups are NOT needed** for a huzzah at least on my HP model.

SwiCago, 2017-06-02:
@Moxified  the huzzah is already pulled up, because of the signal voltage converters for tx and rx. 
For more examples of the mqtt json, look at issue #39

closed 2017-06-03 by SwiCago

SwiCago, 2018-07-19:
@Moxified  I saw you posted configs that work for openhab. Do you think you could share a nice clean setup under integrations? That would be awesome..thanks

Moxified, 2018-07-19:
I'll cook something up :)

SwiCago, 2018-07-19:
@Moxified thanks