SwiCago/HeatPump #54: Demo Circuit

jakebriggs, 2017-06-22:
Hi there

Sorry if this is the wrong place to ask but on your demo circuit you have VCC, RST and CH_PD all connected together - whats the purpose of these connections?

Jake

Moxified, 2017-06-22:
VCC is your supply voltage, RST is the reset pin that should be pulled up to 3v and CH_PD is the enable pin that also should be pulled up to 3v.

From: http://www.esp8266.com/wiki/doku.php?id=getting-started-with-the-esp8266#basic_wiring

Looking at the (8 pin) board from the top (chips and antenna side), with the antenna to the right, the pins from the top down in the edge row are:

TXD (goes to the 3.3V RxI of the UART USB adapter to the PC)
**CH_PD (enable/power down, must be pulled to 3.3v directly or via resistor)
REST (reset, must be pulled to 3.3v)
VCC (3.3v power supply)**
The inner row (top and bottom have square pads)

GND (connect to power ground)
GPIO 2
GPIO 0 (leave open or pull up for normal, pull down to upload new firmware)
RXD (goes to the 3.3V TxO of the UART USB adapter to the PC)

jakerabid, 2017-06-22:
Thanks heaps, that is really helpful. I am using an esp8266 witty like this one: http://www.schatenseite.de/en/2016/04/22/esp8266-witty-cloud-module/

This board seems to work fine so far using only gnd, and 5v - its programmable, it runs this code and also a bunch of other code I've thrown at it - I haven't touched the REST and CH_PD pins. I wonder if they are internally pulled up or something. I am unable to get this board with this code talking to the heatpump though, on the mqtt server all I get is messages saying it sent "fc 5a 01 30 02 ca 01 a8" to the heatpump which is the connection string. I am trying to diagnose what I've done wrong, probably the wiring, and I am wondering if the txd and rxd are somehow not able to work if REST and CH_PD are left floating.... I am pulling the txd and rxd high though, although I haven't taken a multimeter to double check. 

This board takes 5v across VCC and GND rather than 3.3, and has a ams1117 3.3 underneath to provide 3.3v to the chip. To pull up REST and CH_PD, I guess I have to find 3.3v somewhere

Moxified, 2017-06-22:
I don't know that board specifically but I used an Adafruit Huzzah myself and it is a similar development breakout of the ESP.  It doesn't require pullups on reset or CH_PD either.  It also has a built in 5v to 3v regulator.  

I simply connected VCC to HP +5, gnd to HP gnd, rx to tx and tx to rx.  Mine works fine without serial pullups.

One thing that pops into my head.  That board has a built in USB to Serial adaptor (micro usb port).  Are you sure you are using the RX and TX pins and not the built in USB?  I'm not sure how those are addressed on that board.  I looked briefly for more details but didn't get much.  If your code is running, your CH_PD and RST pins are fine.

jakerabid, 2017-06-22:
You are not using the 10K resistors across the tx and gnd and rx and gnd to pull them up? I have a confession, I couldn't find 10K resistors so I am using 11K resistors....

I will double check that the txd and rxd on the board are in fact what I think they are. I see you can define which GPIO pins you want to use on the connect method so maybe its simply a matter of chooing two others and setting it to that :)

jakerabid, 2017-06-22:
OMG I did it again, I logged in on my work lappy with my work account - just in case anyone reading this wonders why 2 different accounts are talking as if they were the same person.

Moxified, 2017-06-22:
Nope, the Huzzah has level shifting built in.  Also, to ground is a pull DOWN.  You would want to pull UP to 5v.  11k would probably work just fine.

If documentation is limited and I were in your shoes, I would probably start by using a simple serial sketch and connecting the tx and rx pins to a computer via usb to serial (not the one on the Witty).  Then use serial monitor to confirm serial is actually coming and going where you think it is.  This will help you learn your board and eliminate some assumptions/unknowns.

jakerabid, 2017-06-22:
Documentation does seem a tad limited, so, I think you are right. Unfortunately, my usb->serial adaptor is currently in use telling my computer the temperature of my brewing beer :P I am sure my beer won't mind if I borrow it....

I might just remove the resistors and see what happens first, maybe put them back in properly to 5v, or something.

jakerabid, 2017-06-23:
Just an update in case someone else out there is looking for a similar answer, I removed the resistors and plugged in only 5v vcc and gnd into the witty, and the voltage across\ both txd /gnd and rxd/gnd was 3.3 so I guess that means rxd and txd are pulled up internally in the witty. Still it doesn't work but we can probably close this now since its not really an issue with this project....

jakerabid, 2017-06-25:
Seems to be working now, I re-crimped the pins on the wire that goes into the cn105 connector because I hadn't done it correctly - I don't have a crimping tool, so I used needle nosed pliers and man was it fiddly.

closed 2017-06-27 by SwiCago