SwiCago/HeatPump #62: Ecodan Air/Water units

AlbertHakvoort, 2017-09-12:
Hello,

I've got a Mitsubishi Ecodan Air/Water heatpump. It uses the same WiFi unit as the air/air units.

But I don't get any response from my unit.  In the mqtt debug I see that the fc 5a 01 30 02 ca 01 a packet is sent. Could it be that I need a different connect packet?

I'm using a nodemcu with 10k pull-ups on the TX/RX

[image]

AlbertHakvoort, 2017-09-12:
I've connected my WiFi module to my ftdi cable and tried to monitor the traffic 

(first part is the output of the esp8266, the red part is output from the WiFi module)

[image]

AlbertHakvoort, 2017-09-13:
I've tried to log the WiFi unit again :

* * * * * * * * * * COM Port Opened * * * * * * * * * * 
Data received: 000 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 063 063 118 063 063 002 
Data received: 063 063 126 063 018 
Data received: 063 119 063 063 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 
Data received: 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 
Data received: 000 000 000 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 126 063 018 
Data received: 063 063 063 063 126 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 
Data received: 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 063 063 118 063 063 002 
Data received: 063 063 119 063 063 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 000 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 063 063 118 063 063 002 
Data received: 063 063 126 063 018 
Data received: 063 119 063 063 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 
Data received: 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 
Data received: 000 000 000 002 
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 126 063 018 
Data received: 063 063 063 063 126 016 
[... 81 lines cut ...]
Data received: 063 063 119 063 063 002 
Data received: 063 063 126 063 018 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 118 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 063 063 126 063 063 063 063 016 
Data received: 063 063 126 016 
Data received: 063 063 126 063 063 063 063 016 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 
Data received: 002 255 255 000 000 000 000 002 

* * * * * * * * * * COM Port Closed * * * * * * * * * * 

So I've tried to send 

const byte CONNECT[CONNECT_LEN] = {0x3f, 0x3f, 0x3f, 0x3f, 0x76, 0x3f, 0x3f, 0x02};

but no response..

lekobob, 2017-09-14:
Try connecting the wireless module to the heatpump and tapping into the signals with the FTDI cable.  You will have to capture rx and tx seperatly to get the full picture.

Belinda-Bird, 2017-09-15:
Try spying on the mitsubihi Wifi unit when connected to the ecodan.
To see both sides (TX and RX) in a single capture, connect the inputs of an AND gate to RX and TX at the ecodan side, and the output of the AND gate to RX on your FTDI - usb.

Hopefully the serial data is only half duplex. other wise you'll get garbage.

SwiCago, 2017-09-19:
Is the wemos the one that has mini usb port built in? I bought a esp8266 with mini usb on it that was supposed to have 5V safe tx/rx (before I did esp01) and it never worked. Might be some issue with level conversion. I suggest you use chips setups that for sure work! Maybe some others can chime in on which chip setups they use.

AlbertHakvoort, 2017-09-19:
@SwiCago I've also tried a NodeMCU. I'll order a ESP01 or Arduino Micro to make sure the 5v isn't the cause..

Belinda-Bird, 2017-09-19:
I don't have access to our ecodan any more, we moved :/

Did you get a chance to capture the TX and RX data while the wifi module was connected to the ecodan to determine the baud rate, and ideally capture the data to compare with non ecodan.

The AND gate can be made with 2 diodes and a resistor, the cathodes in parallel with RX and TX, the anodes  connect together and connect to RX of FTDI, also RX of FTDI needs to be pulled with resistor to 5V.

[image]

AlbertHakvoort, 2017-09-19:
@Belinda-Bird 
I've ordered the AND parts. Today I'm gonna do a test with 2 ttl usb adapters. I will link both with socat, I should then also be able to log the communication between the Wifi and the Ecodan.

Belinda-Bird, 2017-09-19:
Using 2 adaptors should avoid corruption if both the Ecodan and the Wifi send at the same time.

I think I see, RX of each ttl USB adapter goes to RX or TX of Wifi <-> ecodan link, so as nothing interferes with the normal comms between wifi and ecodan, and TX of USB adaptor remains not connected.

Wish I'd of captured the data when I could, but didn't understand it at the time.

CudaNet, 2017-09-21:
I'm using an Adafruit Huzzah Feather ESP8266, no resistors on multiple HP's with zero issues. Adafruit 2821 Feather HUZZAH with ESP8266 Wi-Fi,

AlbertHakvoort, 2017-09-21:
@CudaNet I've also tried the Python script with a TTL USB adapter, also no luck.. 

You didn't use any resistors, so only rx/tx/gnd connected?

CudaNet, 2017-09-21:
@AlbertHakvoort

Yes, no resistors whatsoever.. Attaching photo. Hoping this helps.

[image]

CudaNet, 2017-09-26:
I'm starting to doubt that the Huzzah Feather was a good choice. My first one faulted in 2 weeks and now I have another which is acting up (same unit). Odd though since my other units are working perfect. I'm going to switch over to a Adafruit Huzzah Breakout and report back. MSZ-GE09's are the working units and the FE09's seem to be having issues with the Feather.

SwiCago, 2017-09-27:
@CudaNet my cheap esp01 have been running non stop all summer long. No issues

AlbertHakvoort, 2017-09-27:
I'm now using 2 RPI's with 2 USB TTL adapter (and a usb relay switch to power on/off the wifi module remotely (A))

While the Wifi module is "booting up" I get the following data :

8000 f800 7878 0000 8000 8000 f800 7878
0000 7800 0000 7800 78f8 0080 f880 80f8
8000 7878 7800 0078 f878 8078 8000 f8f8
7878 7880 0078 0078 8080 8000 f8f8 0000
f800 0080 0078 0000 0000 0000 7800 0000
0000 0078 0078 8080 8000 f8f8 0000 0078
0000 0000 0000 0000 0000 0000 0000 8078
7800 8000 0080 f880 00f8 8000 0000 0000
0000 0000 0000 0000 0000 0000 0000 7800
8000 0080 f880 00f8 f800 0000 0000 0000
0000 0000 0000 0000 0000 78f8 7800 8000
0080 f880 00f8 0000 0000 0000 0000 0000
0000 0000 0000 0000 7880 7800 8000 0080
f880 00f8 7800 0000 0000 0000 0000 0000
0000 0000 0000 7878 7800 8000 0080 f880
00f8 8000 0000 0000 0000 0000 0000 0000
0000 0000 7800 7800 8000 0080 f880 00f8
f800 0000 0000 0000 0000 0000 0000 0000
0000 78f8 0000 0078 8080 8000 f8f8 0000
0000 0000 0000 0000 0000 0000 0000 0000
8000 8078 7800 8000 0080 f880 00f8 0000
0000 0000 0000 0000 0000 0000 0000 0000
7880 0080 0078 8080 8000 f8f8 0000 0000
0000 0000 0000 0000 0000 0000 0000 8000
8078 7800 8000 0080 f880 00f8 7800 0000
0000 0000 0000 0000 0000 0000 0000 7878
7800 8000 0080 f880 00f8 8000 0078 0000
0000 0000 0000 0000 0000 0000 7800 7800

Belinda-Bird, 2017-09-27:
have you tried 2400, SERIAL_8E1 or 9600, SERIAL_8E1

You had much better capture 15 days ago when only connected to the wifi adaptor.

AlbertHakvoort, 2017-09-27:
Oops... old test file.. With 2400 baud I get these 

0200 ffff 0000 0000 0202 ffff 0001 0100
0000 5afc 7a02 ca02 5d01 5bfc 7a02 c901
fc5f 0241 107a 0034 0001 0000 0000 0000
0100 0000 0000 fcfd 0242 107a 0001 0000
0000 0000 0000 0000 0000 0000 fc31 0242
107a 0002 0000 0000 0000 0000 0000 0000
0000 fc30 0242 107a 0003 0000 0000 0000
0000 0000 0000 0000 fc2f 0242 107a 0004
0000 0000 0000 0000 0000 0000 0000 fc2e
0242 107a 0005 0000 0000 0000 0000 0000
0000 0000 fc2d 0242 107a 0006 0000 0000
0000 0000 0000 0000 0000 fc2c 0242 107a
0007 0000 0000 0000 0000 0000 0000 0000
fc2b 0242 107a 0008 0000 0000 0000 0000
0000 0000 0000 fc2a 0242 107a 0008 0000
0000 0000 0000 0000 0000 0000 fc2a 0242
107a 0008 0000 0000 0000 0000 0000 0000
0000 fc2a 0242 107a 0009 0000 0000 0000
0000 0000 0000 0000 fc29 0242 107a 000a
0000 0000 0000 0000 0000 0000 0000 fc28
0242 107a 000a 0000 0000 0000 0000 0000
0000 0000 fc28 0242 107a 000a 0000 0000
0000 0000 0000 0000 0000 fc28 0242 107a
000b 0000 0000 0000 0000 0000 0000 0000
fc27 0242 107a 000c 0000 0000 0000 0000
0000 0000 0000 fc26 0242 107a 000d 0000
0000 0000 0000 0000 0000 0000 fc25 0242
107a 000e 0000 0000 0000 0000 0000 0000
0000 fc24 0242 107a 000f 0000 0000 0000
0000 0000 0000 0000 fc23 0242 107a 000f
0000 0000 0000 0000 0000 0000 0000 fc23
0242 107a 000f 0000 0000 0000 0000 0000
0000 0000 fc23 0242 107a 0010 0000 0000
0000 0000 0000 0000 0000 fc22 0242 107a
0011 0000 0000 0000 0000 0000 0000 0000
fc21 0242 107a 0012 0000 0000 0000 0000
0000 0000 0000 fc20 0242 107a 0012 0000
0000 0000 0000 0000 0000 0000 fc20 0042

The Unit repsonse with 

ff02 80ff 0000 010a 4000 0000 0206 007a
b500 ff02 81ff 0000 8100 7afc 7a02 0001
fc09 027b 107a 03c9 0100 1400 0002 0000
0000 0000 0000 fc16 0261 107a 0000 0000
0000 0000 0000 0000 0000 0000 fc13 0262
107a 1101 1b09 2e09 1324 0000 0000 0000
0000 fc6e 0262 107a 0002 000f 0000 0000
0000 0000 0000 0000 fc01 0262 107a 0003
0080 0000 0000 0000 0000 0000 0000 fc8f
0262 107a 0004 0000 0000 0000 0000 0000
0000 0000 fc0e 0262 107a 0005 0000 0000
0000 0600 0000 0000 0000 fc07 0262 107a
0006 0000 0000 0000 0000 0000 0000 0000
fc0c 0262 107a 0007 0000 0000 0100 0101
0001 0000 0000 fc07 0262 107a 0809 07fc
08d0 0098 1900 3c64 5a78 0000 fc03 0262
107a 090b f060 f0c4 0bc4 9808 6a76 0000
0000 fcab 0262 107a 080c 7698 ca08 0977
0bc4 0000 0000 0000 fccf 0262 107a 090d
0bc4 c409 090b 0bc4 c409 000b 0000 fca5
0262 107a 090e 02c4 c409 0002 0000 0000
0000 0000 fc66 0262 107a 0010 0000 0000
0000 0000 0000 0000 0000 fc02 0262 107a
3211 8d00 1000 0000 3e00 0000 0000 0000
fcf4 0262 107a 0013 2800 0a00 0000 0000
0000 0000 0000 fccd 0262 107a 0014 0000
0000 0000 0000 0000 0000 0000 fcfe 0262
107a 0015 ff64 0000 0000 0000 0400 0000
0000 fc96 0262 107a 0016 0000 0000 0000
0000 0000 0000 0000 fcfc 0262 107a 0017
0000 0000 0000 0000 0000 0000 0000 fcfb
0262 107a 0018 0000 0000 0000 0000 0000
0000 0000 fcfa 0262 107a 0019 0000 0000
0000 0000 0000 0000 0000 fcf9 0262 107a
001a 0000 0000 0000 0000 0000 0000 0000
fcf8 0262 107a 001c 0000 0000 0000 0000
0000 0000 0000 fcf6 0262 107a 001d 0000
0000 0000 0000 0000 0000 0000 fcf5 0262
107a 001e 0000 0000 0000 0000 0000 0000
0000 fcf4 0262 107a 001f 0000 0000 0000
0000 0000 0000 0000 fcf3 0262 107a 0020
0000 0000 0000 0000 0000 0000 0000 fcf2
0262 107a 0026 0100 0000 0100 8813 fc08
0000 0000 fc4b 0262 107a 0027 0000 0000
0000 0000 0000 0000 0000 fceb 0262 107a
0028 0000 0100 0000 0000 0100 0000 0000
fce8 0262 107a 0029 0000 fc08 d007 0000

AlbertHakvoort, 2017-09-27:
The last output was captured with socat. But if I compare the results with the ouput of the Windows tool, the data looks inverted (right to left)

AlbertHakvoort, 2017-09-27:
I've made some captures with different setpoint settings : https://www.dropbox.com/sh/lxsn16s18gv8o8u/AABAi_C5jrWo1aNXhMIklXtpa?dl=0

[image]

SwiCago, 2017-09-28:
This looks like the wifi module I once dumped...it seems to cycle through various codes, until the unit has satified it with all it's questions. They are either various connect calls for different units, until it hits a match or diagnostic calls to get version number and unit type. This can go on for a few minutes, but eventually you'll start to see stuff that looks familiar.
Have you tried using Hadleys raspPi python code to connect and send codes?  If that works, then it is the esp chip you tries that did not work.

AlbertHakvoort, 2017-09-28:
I've tried Hadleys python code again, but no response...

AlbertHakvoort, 2017-09-28:
Is the serial data msb or lsb?

twangus, 2017-09-29:
@CudaNet There was a discussion in the gitter Lobby between @Moxified and @Morgon on July 19 about wiring the Huzzah Feather.  I think the conclusion was that the 5V from the HP should be wired to the 5V "USB" pin on the feather, not the "3V" pin you have connected.  That might be the issue with your feather units.

Moxified, 2017-09-29:
Definitely!  The feather is only 3v stable on vcc like you have it wired. The regular huzzah is what I’m using and all 5 have been great for more than 4 months now. 

If you want to keep the feather, you will have to use usb terminal. 

Aren’t the regular huzzah cheaper?  Why did you go with the feather?

Belinda-Bird, 2017-09-29:
@AlbertHakvoort 
I though I might merge the communications from your wifi and ecodan setpoint 15 files.

[setpoint15 w-e.txt](https://github.com/SwiCago/HeatPump/files/1343556/setpoint15.w-e.txt)

Belinda-Bird, 2017-09-29:
@AlbertHakvoort
 I though I might merge the communications from your wifi and ecodan setpoint 15 files.
File attached:
[setpoint15 we.txt](https://github.com/SwiCago/HeatPump/files/1343593/setpoint15.we.txt)

Belinda-Bird, 2017-11-08:
@AlbertHakvoort 
using the base64encoded capture from https://github.com/ncaunt/meldec/tree/master/examples/sample.cap your data is nearly the same, except pair of bytes are swapped around, and the chksum that would normally be at the end is for some reason just after the leading FC of the next transmission.
groupcode 6 is the best example as the data portion is all zeros

from sample.cap decoded base64
<CODE><GROUPCODE>06</GROUPCODE><RECORDNUM>01</RECORDNUM><DATA><DATDATE>2016/03/18 19:39:44</DATDATE><VALUE>fc62027a1006000000000000000000000000000000**0c**</VALUE></DATA></CODE>

However your data has the following; see the "0C" has moved to the next line
fc 07 0262 107a 00 06 0000 0000 0000 0000 0000 0000 0000
fc **0c** 0262 107a 00 07 0000 0000 0100 0101 0001 0000 0000

Belinda-Bird, 2017-11-10:
After swapping the byte pairs around, I came up with the attached file (with a few hints from meldec).
Lines now have a familiar feel, the initial lines starting with 02's seem like a way to identify the attached unit. There is an 027a in there.

[we-setpoint15-swapped.txt](https://github.com/SwiCago/HeatPump/files/1460857/we-setpoint15-swapped.txt)

SwiCago, 2018-01-17:
Since these are not the same type of units and you don't seem to be actively helping with debug packets, I am closing as we can't make it work with nothing

closed 2018-01-17 by SwiCago

SwiCago, 2019-02-02:
@AlbertHakvoort , were you able to figure out the correct packets for you air/water unit? We recently found out that some units prefer 9600baud, maybe that is the case for yours. If you did figure it out, maybe we can integrate it to this library.

AlbertHakvoort, 2019-02-03:
This "project" was on hold for a while (A) but I'll give it a try this week, thanks for the tip.

AlbertHakvoort, 2019-08-15:
At last, I did some new testing..

When I hook up the WiFi module to my laptop, I keep getting Line/Break Errors etc.

But when I "spy" on the serial port (on the Wifi module side),  I get the some good looking data (with 2400/8/E/1) 

[image]

AlbertHakvoort, 2019-08-15:
And when I use the serial port of a Pi I now get also data :

[image]

m000c400, 2020-05-02:
Did anyone make any further progress on this?
I've been decoding the various messages by changing settings on the ASHP and looking what changes in the output,  Examples at the bottom of the post,
I have messages responses to commands 0x01, 0x09, 0x0B, 0x0C, 0x0D and 0x26 done so far, which covers most data you might want to know.

I'd like to move on to sending commands to change stuff next, modify set points etc, But I've not found a hint on how to format the request payload

Any hints most welcome!

_______________________________________________________________________________
Change Detect in Response 0x26
Was 0xFC 0x62 0x02 0x7A 0x10 0x26 0x00 0x00 0x01 0x01 0x01 0x02 0x02 0x17 0x70 0x0B 0x54 0x0D 0xAC 0x00 0x00 0x46
Now 0xFC 0x62 0x02 0x7A 0x10 0x26 0x00 0x00 0x01 0x00 0x01 0x02 0x02 0x17 0x70 0x0B 0x54 0x0D 0xAC 0x00 0x00 0x47
Message 0x26 Decode
HWSetPoint 60.0 Flow Setpoint 29.0, FlowTemp 35.0 Compensation Mode

Change Detect in Response 0x0B
Was 0xFC 0x62 0x02 0x7A 0x10 0x0B 0x07 0x9E 0xF0 0xC4 0xF0 0xC4 0x0B 0x09 0xC4 0x0B 0x6A 0x00 0x00 0x00 0x00 0xAD
Now 0xFC 0x62 0x02 0x7A 0x10 0x0B 0x07 0x9E 0xF0 0xC4 0xF0 0xC4 0x0B 0x09 0xC4 0x0B 0x6C 0x00 0x00 0x00 0x00 0xAB
Zone1 Temp 19.5, OutSide Temp 15.0

Change Detect in Response 0x0C
Was 0xFC 0x62 0x02 0x7A 0x10 0x0C 0x17 0x0C 0xC2 0x16 0xA8 0xC1 0x17 0x0C 0xC2 0x00 0x00 0x00 0x00 0x00 0x00 0xBD
Now 0xFC 0x62 0x02 0x7A 0x10 0x0C 0x16 0x76 0xC0 0x16 0x76 0xC0 0x17 0x0C 0xC2 0x00 0x00 0x00 0x00 0x00 0x00 0x89

Message 0x0C  Decode
Water Heating Feed 57.5,  Water Return 57.5, Hot Water Temp 59.0

Change Detect in Response 0x0B
Was 0xFC 0x62 0x02 0x7A 0x10 0x0B 0x07 0x9E 0xF0 0xC4 0xF0 0xC4 0x0B 0x09 0xC4 0x0B 0x6C 0x00 0x00 0x00 0x00 0xAB
Now 0xFC 0x62 0x02 0x7A 0x10 0x0B 0x07 0xD0 0xF0 0xC4 0xF0 0xC4 0x0B 0x09 0xC4 0x0B 0x6C 0x00 0x00 0x00 0x00 0x79
Zone1 Temp 20.0, OutSide Temp 15.0