SwiCago/HeatPump #63: European model with 10 degree iSave

flic, 2017-10-10:
Hi.

I have the European model with the special iSave mode that allows me to set the temperature to 10 degrees celsius. The library reports that as 16 degrees since that's the minimum, and it's not possible to programmatically set the target temp to 10 degrees either.

I tried looking through the code to see if it would be easy to make the necessary changes, but I really suck in C++ so I couldn't make heads or tails of it.

Would this be an easy fix? I would really appreciate it!

Jakob A posted this on nicegear some time back:

> However, this model has an "i-save" mode, which allows you to set the temperature down to 10 C. I want to be able to set this mode, but looking at the code it doesn't look like it supports that. 
Of course I tried to set it from the remote and see what happens, this is what I saw in the debug print outs: 
isave 10 degrees, heat, fan auto, vane swing: 
HP Packet: 0x62 : 02,00,00,01,01,0f,00,07,00,00,03,94,00,00,00,00 : 0xac 

> isave 16 degrees, heat, fan auto, vane swing: 
HP Packet: 0x62 : 02,00,00,01,01,0f,00,07,00,00,03,a0,00,00,00,00 : 0xa0 

> heat 16 degrees, heat, fan auto, vane swing: 
HP Packet: 0x62 : 02,00,00,01,01,0f,00,07,00,00,03,a0,00,00,00,00 : 0xa0 

> The only thing that changes, as far as I can see, is the 12th byte. 
"94" for 10 degrees, "a0" for 16 degrees. 
I tried setting the temperature to 31 C, and it changes to "be" 

Thanks for all of the hard work, it's a great library.

SwiCago, 2017-10-16:
@flic yours is an older model that does not support 0.5C increments I presume. We only have the hard mapping on those models. You can add more temp mappings to TEMP and TEMP_MAP in the h file and adjust the array length to all calls that use it in the cpp file.
I would use the remote and see what is returned, so that you know what the mappins are

closed 2017-11-12 by SwiCago

jascdk, 2019-06-15:
@flic hi:) Did you figure more out on this I-save feature?

flic, 2019-06-15:
Nope, never did I’m afraid.