SwiCago/HeatPump #66: MQTT

Prinsessen, 2017-11-06:
Hi All

I have my ESP8266 connected to my MQTT broker, I can see all Packet sent, but when I try the heat pump/set I get : invalid J SON on heat pump_set_topic.

Anyone have the correct syntax of the payload, so I can narrow out the payload as error?

Prinsessen, 2017-11-07:
The MQTT test were done without the serial connection to the heatpump - so I decided to do all connections to the pump.

I Have the NodeMCU Lua WIFI Internet Development Board Based on ESP8266 ESP-12E CP2102 with the mitsibishi MQTT example loaded successful, but it seems there is no seriel communications. Also tried with 10K pull-up resistors with no luck.

Anyone can get me further?

Prinsessen, 2017-11-07:
What Json library are you using?

SwiCago, 2017-11-09:
I used Imroy/pubsubclient
Sample syntax
mosquitto_pub -h localhost -p 1883 -t 'heatpump/set' -m '{"power":"ON"}'
If you have a long topic/device name you may have to mod the mqtt library and up the number of chars pubsubclient library defaults to 128 max packet size. I increased MQTT_MAX_PACKET_SIZE to 256 in pubsubclient.h

Prinsessen, 2017-11-09:
Thanks alot:)

I were missing the } stupid me :(

Great GREAT work, and it works like a charm with MSZ-FD25VA European units :)

SwiCago, 2017-11-10:
Glad too hear..I believe we have the MSZ already listed under supported units, but thank you for confirming your model.

closed 2017-11-10 by SwiCago