SwiCago/HeatPump #69: RemoteTemp

Prinsessen, 2017-11-11:
I have the MQTT example and running 95%

I do have an issue with the remoteTemp

When I publish as example - {remoteTemp:24} to the heatpump/set topic the HP is returning 10
I have tried with different temperature settings with the same result.

When I publish{remoteTemp:0} it returns to use the internal temperature sensor.

Anyone have an idea what is wrong?

SwiCago, 2017-11-12:
Have you tried a floating point number ;)
remoteTemp:24.0

Prinsessen, 2017-11-12:
Yes I also tried that, it returns:

{
  "roomTemperature" : 10,
  "operating" : false
}

When I Publish remoteTemp:0 it returns to use the internal sensor

SwiCago, 2017-11-12:
@Prinsessen I am using the "mitsubishi_heatpump_mqtt_esp8266.ino" example as is on 6 units. Only thing I changed on each unit is the .h file for topic names.
I am running 2 attic units, that use remoteTemp and this is how my remote sensors send the data
{"remoteTemp":22.0}
here is an example using mosquitto pub
 mosquitto_pub -h localhost -p 1883 -t heatpump/kitchen/set -m '{"remoteTemp":19}'
and here is subscribe debug output
 mosquitto_sub -h localhost -p 1883 -t heatpump/kitchen/#

{"remoteTemp":19.00}
{"packetSent":"fc 41 01 30 10 07 01 00 **a6** 00 00 00 00 00 00 00 00 00 00 00 00 d0 "}
{"packetRecv":"fc 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5e "}
{"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
{"packetRecv":"fc 62 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 "}
{"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
{"packetRecv":"fc 62 01 30 10 02 00 00 00 03 06 00 01 00 00 00 b2 00 00 00 00 9f "}
{"packetRecv":"fc 62 01 30 10 03 00 00 09 00 00 **a6** 00 00 00 00 00 00 00 00 00 ab "}

A6 = 166 = t 
Temp = (t - 128) / 2 = (166 - 128) / 2 = 19
Temp = 19°C 

Use mosquitto for debugging. And start with a clean ino, with just modded .h, until you get what it is expecting.
Don't forget "quotes" around Strings, even if it is a key.

Here is a cheesy sample ino for using a DHT22 as remote temp sender, 

`#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <PubSubClient.h>
#include "DHT.h"

// wifi settings
const char* ssid     = "warts";
const char* password = "drp3pp3r";

// mqtt server settings
const char* mqtt_server   = "192.168.1.1";
const int mqtt_port       = 1883;
const char* mqtt_username = "";
const char* mqtt_password = "";

// mqtt client settings
const char* client_id                   = "tempsens-living"; // Must be unique on the MQTT network
const char* tempsens_topic              = "tempsens/living";
const char* heatpump_set_topic          = "heatpump/living/set";
int fails = 0;

// sketch settings
const unsigned int SEND_ROOM_TEMP_INTERVAL_MS = 6000;
unsigned long lastTempSend;

#define DHTPIN 2       // what digital pin we're connected to

//#define DHTTYPE DHT11   // DHT 11
#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321
//#define DHTTYPE DHT21   // DHT 21 (AM2301)

DHT dht(DHTPIN, DHTTYPE);

// wifi, mqtt and heatpump client instances
WiFiClient espClient;
PubSubClient mqtt_client(espClient);

void setup() {
  delay(2000);
  Serial.begin(9600);
  Serial.println("DHTxx test!");

  dht.begin();

[... 115 lines cut ...]
    Serial.println("Failed to read from DHT sensor!");
    return -99;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(c, h, false);

  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(c);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t");
  Serial.print("Heat index: ");
  Serial.print(hic);
  Serial.print(" *C ");
  Serial.print(hif);
  Serial.println(" *F");
  sendData(c, f, h, hic, hif);
  return c;
}`

Prinsessen, 2017-11-12:
I have turned on debug mode and set the remoteTemp to 19 as in your example - this is my debug:
I see the a6 but it's not beeing returned - even 94 as 10 is equal to is not there.

Do you have any idea

```remoteTemp:19.0}
{"packetSent":"fc 41 01 30 10 07 01 00 **a6** 00 00 00 00 00 00 00 00 00 00 00 00 d0 "}
{"packetRecv":"fc 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5e "}
{"packetSent":"fc 42 01 30 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 "}
{"packetRecv":"fc 62 01 30 10 04 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 d9 "}
{"packetSent":"fc 42 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 78 "}
{"packetRecv":"fc 62 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 "}
{"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}
{"packetRecv":"fc 62 01 30 10 06 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 53 "}
{"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
{"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
{"packetRecv":"fc 62 01 30 10 02 00 00 01 03 0b 06 03 00 00 05 00 00 00 00 00 3e "}
{"packetSent":"fc 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7a "}
{"packetRecv":"fc 62 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5a "}
{"packetSent":"fc 42 01 30 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 "}
{"packetRecv":"fc 62 01 30 10 04 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 d9 "}
{"packetSent":"fc 42 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 78 "}
{"packetRecv":"fc 62 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 "}
{"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}
{"packetRecv":"fc 62 01 30 10 06 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 53 "}
{"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
{"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
{"packetRecv":"fc 62 01 30 10 02 00 00 01 03 0b 06 03 00 00 05 00 00 00 00 00 3e "}
{"packetSent":"fc 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7a "}
{"packetRecv":"fc 62 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5a "}
{"packetSent":"fc 42 01 30 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 "}
{"packetRecv":"fc 62 01 30 10 04 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 d9 "}
{"packetSent":"fc 42 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 78 "}
{"packetRecv":"fc 62 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 "}
{"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}
{"packetRecv":"fc 62 01 30 10 06 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 53 "}
{"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
{"roomTemperature":10,"operating":false}
```

SwiCago, 2017-11-12:
Does your unit support 0.5°C increments?

SwiCago, 2017-11-12:
Oops hit wrong button...again, does it support 0.5 increments? When you use the IR remote, can you set 22.5 or only 22 and 23 for example?

Prinsessen, 2017-11-12:
I have just checked - My unit doesn't support 0.5 increments.
(We are talking about the setpoint right?) setTemp

SwiCago, 2017-11-12:
Ok, then that is probably the problem. 
The function setRemoteTemperature was created with 0.5 increments in mind. 
setTemperature was created first to support older 1 increment and then later updated to 0.5 increments. Remote temp was discover after and wasn't coded to support older units.

I bet if you modify setRemoteTemperature to set packet[8] with
lookupByteMapIndex(TEMP_MAP, 16, (int)(setting + 0.5)) > -1 ? setting : TEMP_MAP[0];

Try below modded function. It may have 1 or 2 syntax errors. Typed it up from my tablet.
I bet this will work, if it doesn't then chances are the remoteTemp byte is not on 8 but rather another. You'll have to experiment. sendCustomPacket can be used for that.
```
void HeatPump::setRemoteTemperature(float setting) {
  byte packet[PACKET_LEN] = {};
  for (int i = 0; i < 21; i++) {
    packet[i] = 0x00;
  } 
  for (int i = 0; i < HEADER_LEN; i++) {
    packet[i] = HEADER[i];
  }
  packet[5] = 0x07;
   if((setting > 0) && (!tempMode){ //units that don't support 0.5 increment
    packet[8] = lookupByteMapIndex(TEMP_MAP, 16, (int)(setting + 0.5)) > -1 ? setting : TEMP_MAP[0];
   }
  else if(setting > 0) { //units that do support 0.5 increment
    packet[6] = 0x01;
    setting = setting * 2;
    setting = round(setting);
    setting = setting / 2;
    float temp = (setting * 2) + 128;
    packet[8] = (int)temp;
  }
  else { //reset to use internal sensors
    packet[6] = 0x00;
    packet[8] = 0x80; //MHK1 send 80, even though it could be 00, since ControlByte is 00
  } 
  // add the checksum
  byte chkSum = checkSum(packet, 21);
  packet[21] = chkSum;
  while(!canSend(false)) { delay(10); }
  writePacket(packet, PACKET_LEN);
}
```

Prinsessen, 2017-11-13:
Hi @SwiCago 

Thanks for all your effort, helping me out with this.

I have changed the HeatPump.cpp file according to your example.

Now when I publish a remoteTemp to the HP, it's not changing anything like to 10 before.

You are suggesting that the remoteTemp could be on something else that 8, can you somehow guide me how to proceed, i'm not familiar with the custompacket

Prinsessen, 2017-11-13:
When I publish 19 to remoteTemp the package that is sent is 👍 

{remoteTemp:19}
{"packetSent":"fc 41 01 30 10 07 00 00 13 00 00 00 00 00 00 00 00 00 00 00 00 64 "}
{"packetRecv":"fc 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5e "}
{"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}
{"packetRecv":"fc 62 01 30 10 06 00 00 08 01 00 00 00 00 00 00 00 00 00 00 00 4e "}
{"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
{"roomTemperature":22,"operating":true}

13h = 19decimal 

It somehow doesn't use temp=(t-128)/2

SwiCago, 2017-11-13:
@Prinsessen  I forget to set packet[6] to 0x01, as said I typed it with my tablet. Did not and could not test.
Anyhow packet 6 turns the mode on and packet 8 has the temp setting.
I updated my comment above with the additiin, try it again. 

Yes, you are correct it is not using the [t-128)/2 because your unit doesn't support 0.5 increments, instead it uses the TEMP_MAP, where 31C = 0x00 and 16C = 0x0f

custom packet is last resort, but basically you send via mqtt a hex string, minus the header FC amd the chksum(last byte)

But first try the update. It seems your unit does understand remoteTemp, since it auto sets bad values to 10C
To think of it, you may have to use ROOM_TEMP_MAP instead!
The above is using the set map(TEMP_MAP), not the read map(ROOM_TEMP_MAP)
First try with the edition of packet 6 and if it still reports incorrectly, try changing the following 
   packet[8] = lookupByteMapIndex(TEMP_MAP, 16, (int)(setting + 0.5)) > -1 ? setting : TEMP_MAP[0];
To
   packet[8] = lookupByteMapIndex(ROOM_TEMP_MAP, 32, (int)(setting + 0.5)) > -1 ? setting : ROOM_TEMP_MAP[0];

Prinsessen, 2017-11-13:
Hi Again :)

I tried both your suggestions in the order you mentioned -It keeps setting the remoteTemp to 10 and turns off operation.

Could it be that this unit doesn't support this function?

SwiCago, 2017-11-13:
It is very well possible that it does not support it or it uses a different byte for remote temperature.
See if your units supports MHK1 acessory, if it does then you may need to experiment first with cycling all possible hex values on packet 8, and if they all yield 10, then try a different byte. Maybe one before it or after it. Best thing you could do, of it supports the accessory, is to borrow one and see what it sends to your unit. Packet sniffing

Prinsessen, 2017-11-13:
Ill give that a try - and thanks for all your effort :)

The good thing is that when I publish a remoteTemp it turns operation to false - means I can just let my automation system do the math. I already have a temperature sensor in the room.

When the Room sensor is below the HP setpoint I can turn off operation and back on when it is greater.

SwiCago, 2017-11-13:
According to this pdf, it looks your unit is not support the mhk1
http://meus1.mylinkdrive.com/files/Controller_Compatability_Chart_M-Series.pdf
However, that does not mean it won't support maybe an older product that does something simular. You'd have ro do some research and see what you can find. Maybe experiment with custom packet.
Or as you mention, do automation yourself. I dod something similar with my 4 wall units. I set them to set point temp and then auto turn them off if below or above a specific threshold to keep cycling lol

closed 2017-11-13 by SwiCago