SwiCago/HeatPump #70: Strange behavior when remote used

robsee, 2018-01-07:
Hi,

 I'm seeing the following behavior. I don't know that it is related to the HeatPump software, but I just wanted to see if anyone else has noticed it.
 When I use the remote, every other press of the temp up or down results in the temp jumping up to the max. I noticed this while testing converting the temp from C to F.
Here is the MQTT output. c__temp is the original temp
lrheatpump {"power":"ON","mode":"HEAT","c_temp":18,"temperature":64,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":31,"temperature":88,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/status {"roomTemperature":64,"operating":true}
lrheatpump/timers {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
lrheatpump/status {"roomTemperature":64,"operating":false}
lrheatpump/timers {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":17,"temperature":63,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":31,"temperature":88,"fan":"AUTO","vane":"AUTO","wideVane":"|"}

Anyone else run into this ?

robsee, 2018-01-07:
Additional info. When I switch to C mode on the remote, I do not see this behavior. The remote does not support half steps. I've tried with 2 different remotes with the same effect. I'm testing with a MSZ-GE18NA

SwiCago, 2018-01-07:
Since it is defaulting to 31 each time, makes me believe your unit is older and does not support 0.5C increments and is using temp_map instead. You could have mqtt periodicaly make it return tempMode to check. If false, it does not support 0.5C increments.
If that is the case, you could turn on debugging and see what the HP os returning to the software, when in F mode on remote. I think the 13th byte is the temperature byte on older units. You could see what it returns when changing temp in F. The value should map the temp_map, if it doesn't it defaults to 0, which maps to 31C in the map. In thoery if you can get the value, then you could add it to the map or you could check if another if this byte already returns as zero from HP, but set another for the incremental values.

robsee, 2018-01-08:
So it looks like the half steps have 0x10 added to them.

Can I just add the values to TEMP and TEMP_MAP (and increase their length) in HeatPump.h, or is there more to it ?

Here are some examples of debug packets:
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 04 00 00 00 00 03 00 00 00 00 00 52 "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":27,"temperature":81,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 15 00 00 00 00 03 00 00 00 00 00 41 "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":31,"temperature":88,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 06 00 00 00 00 03 00 00 00 00 00 50 "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":25,"temperature":77,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 18 00 00 00 00 03 00 00 00 00 00 3e "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":31,"temperature":88,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 09 00 00 00 00 03 00 00 00 00 00 4d "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":22,"temperature":72,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
lrheatpump/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 1b 00 00 00 00 03 00 00 00 00 00 3b "}
lrheatpump {"power":"ON","mode":"HEAT","c_temp":31,"temperature":88,"fan":"AUTO","vane":"AUTO","wideVane":"|"}

SwiCago, 2018-01-08:
You can add values to the map and then adjust the length on each call that uses the map
There are two maps, one is the byte and the other is the temp it translates too TEMP and TEMP_MAP
You may also need to see what is returned on room temp status 
That is fc 62 01 30 10 03 response

SwiCago, 2018-01-17:
Closing, assume you got it worked out

closed 2018-01-17 by SwiCago