SwiCago/HeatPump #71: Home Assistant integration breaks in HA 0.60.1

alexholman, 2018-01-08:
Upgraded my version of HA to 0.60.1, and all of a sudden the mitsubishi_mqtt platform has stopped working with a completely non-informative error message in the log. Removing the mitsubishi_mqtt from the configuration and putting in a generic_thermostat object works just fine.

`Sun Jan 07 2018 21:27:09 GMT-0500 (EST)

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity.py", line 231, in async_update_ha_state
    attr)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity.py", line 333, in _attr_setter
    value = getattr(self, name)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/climate/__init__.py", line 736, in supported_features
    raise NotImplementedError()
NotImplementedError`

robsee, 2018-01-08:
Checkout https://github.com/robsee/mitsubishi_mqtt/ . I just went through the same issues this weekend. I have my setup configured to work in fahrenheit, so if you are working in celsius, you would need to replace all instances of TEMP_FAHRENHEIT with TEMP_CELSIUS

lekobob, 2018-01-08:
Fixed with #72  merged in last night.

closed 2018-01-10 by SwiCago