SwiCago/HeatPump #73: Incorrect temperature reporting when heat pump isn't actively "operating"

twangus, 2018-03-05:
I'm loving this library and set up controllers for my four MSZ-GL06NA heatpumps, each running the standard mqtt example (modified only with my config details).  It's really awesome!  

However, I'm seeing some odd behavior around reporting room temperature if the unit is not actively "operating" to heat or cool the room and I can't figure out whether it is an issue with my heat pumps, the library, or the sketch.  Basically, while the unit is heating or cooling to get to the target temperature, it reports the room temperature with relative accuracy.  However, once it has gotten to that target temperature (or a few degrees past it), it seems to stop sending accurate temperature readings (essentially ignoring natural temperature changes that would take the room well beyond the target).

Here's a specific example -- last night we had the units set to "HEAT" with a target temp of 20 C / 68 F.  It accurately reported the current room temperature when I first set it (18 C / 64.5 F) and while it was operating to heat up to the target.  In the morning, though, the room started to heat up on its own from the sun coming in the windows and got well above the target temperature.  But no matter how hot it got in the room, the heat pump never reported that it was more than 22.5 C / 72.5 F.  When I switched the units to cool, the unit immediately started reporting the accurate room temperature (24.5 C / 76 F).  My units all do this and they do it consistently, even with much larger discrepancies in the temperature, where the reported temperature will be off by 8-10 degrees F and then immediately correct themselves when the mode is switched.

Here's a look at the MTQQ messages that are being generated from that specific example:

hp/3 - {“power":"ON","mode":"HEAT","temperature":20,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
… [several hours]
hp/3/status - {“roomTemperature":22.5,"operating":false}
hp/3/status - {“roomTemperature":22.5,"operating":false}
hp/3/status - {“roomTemperature":22.5,"operating":false}
hp/3/status - {“roomTemperature":22.5,"operating":false}
hp/3 - {“power":"ON","mode":"COOL","temperature":20,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
hp/3/status - {“roomTemperature":24.5,"operating":false}
hp/3/status - {“roomTemperature":24.5,"operating":true}
hp/3/status - {“roomTemperature":24.5,"operating":true}
hp/3/status - {“roomTemperature":24.5,"operating":true}

As you can see the reported roomTemperature immediately jumps from 22.5 to 24.5 when the mode is changed from HEAT to COOL.

I've looked through the code for the library and the sketch, but I don't understand it well enough to tell whether the issue is with the heat pump itself reporting the wrong temperature when requested or if there's something about the library that isn't requesting or correctly updating the status under those circumstances.

Has anyone else come across this?  Any ideas?

SwiCago, 2018-03-06:
Do your units have iSee?

twangus, 2018-03-06:
No, they don't have iSee.

SwiCago, 2018-03-07:
Your units may need air movement to accurately report temps. And yes the units do report different base temps depending on mode. You can change to FAN and also see temp change. That is a mitsubishi thing. 
I saw this while using official kumo cloud. 
For really accurate temps, use a remote sensor and send that to the HP.  Using temp from unit is really very accurate unless air is moving, because they sit up high.

closed 2018-03-11 by SwiCago

melyux, 2020-08-18:
I know it's closed, but I second this "problem", as far as it goes. Here's a graph. The blue line is the Mitsubishi's internal thermometer, and the red line is a calibrated external thermometer placed next to the unit's temperature air holes.
[image]
Before the cooling is turned on in the middle of the graph, the unit is idle and reports temperatures higher than the real temperature, and the discrepancy grows wider as the ambient temperature rises. But once the cooling is turned on, the internal thermometer suddenly starts tracks very closely with the real temperature.

I suspect it may be the added heat from the Arduino placed into the unit, since the internal thermistor connection (CN111) is pretty close to the CN105 port and everything. But I'd have thought that would also add heat when the unit is cooling, instead of suddenly making it hyperaccurate.

superflyxxi, 2023-09-24:
Thanks @SwiCago. Your comment helped me narrow down the issue (seems some built in 6 or 7 degrees Fahrenheit offset).

If anyone knows a way to turn this off or customize the offset, that'd be great. MLZ-KP12NA