SwiCago/HeatPump #74: calling sync() when millis() < (PACKET_SENT_INTERVAL_MS * 10) doesn't work

llemtt, 2018-04-11:
If I call sync() in the loop while millis() is less than (PACKET_SENT_INTERVAL_MS * 10) it tries to connect again and again forever.

I think that:

(millis() - (PACKET_SENT_INTERVAL_MS * 10) > lastRecv))

is not a correct time comparison, it should be done something like this

(millis() > lastRecv ? (millis() - lastRecv) : -(lastRecv - millis())) > (PACKET_SENT_INTERVAL_MS * 10) 

other similar time comparison have to be reviewed similarly

SwiCago, 2018-04-12:
I have 6 esp01 units running the MQTT sample code without issue for 9 months straight. No one else has reported an issue either. In theory both statements above are identical, other than the size of the number being compared. What chip type are you using? If it is an esp you shouldn't be having problems.

llemtt, 2018-04-12:
I'm currently running on ESP32.

Even in theory those statement are not identical because of UNSIGNED integer arithmetic as explained also on the above link:
------------------------
Most C programmers would write the above loops in a terser form, like

while (millis() < start + ms) ;  // BUGGY version

and

while (millis() - start < ms) ;  // CORRECT version

Although they look deceptively similar, the timestamp/duration distinction should make clear which one is buggy and which one is correct.
-------------------------

If "millis() is less than (PACKET_SENT_INTERVAL_MS * 10)" then 

(millis() - (PACKET_SENT_INTERVAL_MS * 10)) should be less than 0 but, being millis() an UNSIGNED long int it actually becomes a big positive unsigned integer quantity greater than lastRecv and then connect(NULL) gets called again inside sync().

I hope this clarifies why connect(NULL) is called again the first time in the loop, then I don't know why calling it a second time while already connected fails! Maybe additional checking has to be made inside connect() to handle such a case.

SwiCago, 2018-04-13:
I see, seems more like a compiler issue and not a program issue. Any compiler worth anything should be able to handle either way, same for the processor. In  any case, have you flipped it around in your copy and checked to see if that even solves the problem you are having? If it does, I can put this change into the master. Let me know

SwiCago, 2018-04-17:
3 days passed since I flipped the time check. I assume this either fixed your issue or your issue waa something else all together. Repen of this is still an issue

closed 2018-04-17 by SwiCago

llemtt, 2018-04-17:
Sorry I was away and not connected...

I agree that different compilers may end up optimizing differently, anyway there's just one correct C++ semantic.

I fixed my problem using the swapped version:

(millis() - lastRecv > (PACKET_SENT_INTERVAL_MS * 10))

If it works also in your environment you can change the master!

Many thanks