SwiCago/HeatPump #77: NodeMcu

dsacristan, 2018-05-23:
First of all, thank you very much for your project. 

Has somebody tried a NodeMcu(with the ESP8266 and working with 5v)? I wonder if we can use it directly to the CN105, without any tension regulator

thanks

SwiCago, 2018-05-23:
Only inputs are 5V safe on esp8266, not VIN. Which is 3.3V max... so unless it has onboard regulator, you will kill it. 
The adadfruit huzzah does have a regulator with a pin that can have 5V on it.

closed 2018-05-23 by SwiCago

llemtt, 2018-05-24:
I run mine without issues, only an additional mosfet is required at output because esp8266 doesn't support open drain directly (esp32 does).

I think it's not a matter of CN105 but what kind of circuit is connected behind it... I've studied my schematic before doing that way!