SwiCago/HeatPump #84: No signal from MSZ-SF25VE3

reibuehl, 2018-07-31:
I am trying to control a MSZ-SF25VE3 indoor unit. I have hooked up an FTDI USB to serial adapter to RX, TX and GND of the CN105 and tried to receive with 2400 8E1 but don't get anything. The 12V, 5V and GND signals are there but there is no serial data.
Are there other serial speeds or parameters used by the ME units? Or does one need to enable the serial functionality somehow first?

SwiCago, 2018-08-01:
I do not know of any other speed, other than 2400_8E1. Are you sure you got the pins right? Does your USB adapter pull RX/TX up?

reibuehl, 2018-08-02:
The adapter is still working - I use it to flash ESPs. I tried reversing TX and RX just in case I might have mixed up these but that also didn't change anything. 
Is there a way to check (easily) if the adapter pulls up TX/RX?
I have another FTDI adapter on order and will check with that one again once I receive it.

F43RY, 2018-08-02:
Hi,
I use a [ESP-01 adapter](http://img.dxcdn.com/productimages/sku_403053_1.jpg) from Amazon and had no problem with my MSZ-SF25VE3.

reibuehl, 2018-08-02:
Did you use any additional external level shifting with that adapter or just connected it to the AC?

F43RY, 2018-08-02:
just connected to AC. No additional hardware, just adapter and esp-01. 
I plugged only 4 wires (TX, RX, GND, 5V) from CN105, no 12 V.

llemtt, 2018-08-02:
I don’t understand how is your CN105 wired to esp serial port, can you explain?

F43RY, 2018-08-02:
A = ESP-01 adapter (shown in a previous post)
B = CN105

A(TX) <--> B(RX)
A(RX) <--> B(TX)
A(VCC) <--> B(5V)
A(GND) <--> B(GND)
  NC      --> B(12V)

I apologize for my concise post but I'm writing from my mobile phone.

llemtt, 2018-08-03:
Which pin numbers on CN105?

But if I understood correctly it works with your esp-01 but not with your FTDI usb-serial adaper, correct?

What kind of serial is your FTDI adapter? RS232?

SwiCago, 2018-08-03:
@reibuehl  I suggest you check which pin is the 12V on the cn105, by using a meter. Then connect your serial to usb adapter accordingly. Again, you may need a usb serial with pull ups, as the heatpump cn105 does not have pull ups, but is floating! The esp adapter @F43RY suggested is an adapter for a esp module, not PC. It has level shifts/pull ups.

reibuehl, 2018-08-03:
@SwiCago the pin numbers are clear, I already checked where the 12V and the 5V is. I have now ordered some ESP-01s and the same adapter that @F43RY already had success with and will use these. I just tried with the FTDI USB adapter directly as part of the troubleshooting why my ESP-201 board did not return anything.

SwiCago, 2018-08-03:
@reibuehl let us know when you get your order, so we can close this issue

F43RY, 2018-08-04:
No changes. I flashed esp-01 with an external FTDI, plugged in the adapter and connected to the AC. I flashed the mqtt code for openhab.

reibuehl, 2018-08-04:
Using the ESP-01 and the board recommended by @F43RY I now get data from my AC. Thanks all for the help!

closed 2018-08-04 by reibuehl

SwiCago, 2018-08-06:
Great, glad it's working for you

reibuehl, 2018-08-06:
One little question from an Arduino beginner, @SwiCago : I am using the mitsubishi_heatpump_mqtt_esp8266.ino from the examples but since I have 2 indoor units in different rooms, I need to use different values for heatpump_topic in the .h file. As there are other constants like heatpump_set_topic  and heatpump_status_topic that "build up" on heatpump_topic, I was wondering what would be the best way to define heatpump_topic and the others based on client_id to set the variable part only once and use it in the others?

How would I for example define heatpump_status_topic as "/SmartHome/<client_id>/status" and  heatpump_topic to "/SmartHome/<client_id>"?

SwiCago, 2018-08-07:
@reibuehl , I have 6 units. Named east-bd, nw-bd, ne-bd, middle-bd, living, kitchen so my topics are as follows
`const char* client_id                   = "heatpump-east-bd"; // Must be unique on the MQTT network
const char* heatpump_topic              = "heatpump/east-bd";
const char* heatpump_set_topic          = "heatpump/east-bd/set";
const char* heatpump_status_topic       = "heatpump/east-bd/status";
const char* heatpump_timers_topic       = "heatpump/east-bd/timers";

const char* heatpump_debug_topic        = "heatpump/east-bd/debug";
const char* heatpump_debug_set_topic    = "heatpump/east-bd/debug/set`

I did the above for each one. Hope that helps

SwiCago, 2018-08-07:
Do remember, you may need to raise the number of chars you can use in the mqtt library see #53