SwiCago/HeatPump #88: Proposal to merge openHab integration into MQTT example

F43RY, 2018-08-09:
Hi,
is it possible to "merge"

`integrations/OpenHAB/mitsubishi_heatpump_mqtt_esp8266_template/mitsubishi_heatpump_mqtt_esp8266_template.ino`
 into 
`examples/mitsubishi_heatpump_mqtt_esp8266/mitsubishi_heatpump_mqtt_esp8266.ino`

and

`integrations/OpenHAB/mitsubishi_heatpump_mqtt_esp8266_template/mitsubishi_heatpump_mqtt_esp8266.h`
 into 
`examples/mitsubishi_heatpump_mqtt_esp8266/mitsubishi_heatpump_mqtt_esp8266.h`
?

Files in `integrations` folder are more updated and the codebase is the same.

SwiCago, 2018-08-10:
@F43RY, I am not sure about that. @Moxified said it was based on an older version and had hacks to make it work for openhab, hence why he placed it as it's own copy in the integrations. Since I do not work with either of the integrations, I cannot say if it will or will not break anything. Can you two work together to verify that it won't break anything. Otherwise, let's keep it separate for now. 
Tha ks

F43RY, 2018-08-10:
I'm using openhab version in my openhab system. In brief openhab comunicates with esp through an external Mqtt broker (mosquitto in my case). The only code dedicated to openhab are the rest of files in integrations/Openhab folder (such as items, sitemap and transform etc) that have to be separated. Files above diffs only for mqtt topics in header and some optimized function (such as  wifi connect function external from setup function). Let's wait a reply from @Moxified about my proposed merge.

Thx

Moxified, 2018-08-10:
Hi @F43RY,

You are likely correct that the meat of the differences are my enhancements and topics.  I don't know that it will or won't break anything as I didn't/don't have the time at the moment to go back through the code and verify it.  My intention is to do so at some point but life has been too busy.

If you have time and would like to test using the example code on your OH and HP's, please do and let us know!!

There is also a [thread ](https://community.openhab.org/t/mqtt-json-mitsubishi-heatpump/29345/77?u=moxified) on OH forums where a user suggested replacing the maps with java transforms.  It requires another transform binding so I never looked further into it cause my method works fine.

On a side note.  I'm no developer but many of my "enhancements" to wifi and milli based timers are more robust from what I have found.  We may want to vet and merge some of them into the example.  The biggest reason I haven't tracked with the example is because I don't want to have to add my changes every time.  As I said, I'm not a developer so I don't tend to make suggestions cause I'm far from confident in anything I'm doing with this stuff.

F43RY, 2018-08-10:
Hi @Moxified , thanks for your reply. I'm using your code and no MQTT example because I found your code (as you write) more robust from MQTT example, especially in Wifi connection. In latest days I did pull requests to add some missing "map value" and file in transform for openhab system and added a "flag" in header file to choose among fahrenheit and Celsius. As you did, I don't know if your code with o without my changes will or will not break anything. I test only in my home HP's and it works fine, but I don't know if this stuff is good for everyone in every model or country, I can't test it on any further HP. **So NO guarantees from me  on my code, only my personal experience**.

About the [thread](https://community.openhab.org/t/mqtt-json-mitsubishi-heatpump/29345/76) you suggested I'm having a look asap, using Jscript could be the next challenge.

Thx

Moxified, 2018-08-10:
Thanks for adding the HPVane.map.  it's in the folder on my laptop that I uploaded but I must have missed it when doing so.

Also, love the F/C parameter.  I wasn't writing the code for sharing so thanks for that!

Perhaps the first step would be to try to merge the "enhancements" for wifi, F/C and such into the library mqtt example, then I will make an effort to evaluate it against my HP's as well.

As for the jscript, I don't think it's really a challenge so much as it required another binding that I don't use for anything else.  I try to minimize how many bindings I use.  I thought about it before writing up the 'tutorial' and concluded it was simpler to ask people to dump the .map files than it was to ask them to add yet another binding.

I stick to the full OH release instead of snapshot.  As such, bindings often will have a bug or two that end up requiring me to manually install a snapshot binding.  I dislike this immensely and thus use as few bindings as possible.

F43RY, 2018-08-10:
I'd like to understand why the code includes 3 temperature vars (Temp, roomTemp and remoteTemp). I thought that 2 of them were enough. Maybe I've to learn better the code .

Another part I'd like to improve/add (if I had time... ) is "timer status" because, at the moment, it is not included in OH files `HP.items` and `HP.sitemap`.

SwiCago, 2018-08-10:
Thanks @F43RY  @Moxified  for collaborating together. I also did not write the original mqtt example. I wrote the library and the 2 simple examples. I believe @kayno wrote the mqtt example. I mainly just made the library and figured everyone would do their own thing, but it looks like the mqtt example and the integrations is easiest for everyone lol. 
I am on holiday right now, but when I get back I can help review
Cheers

Moxified, 2018-08-10:
@F43RY  Um, I think... off the top of my head:

temp: HP target setpoint
roomtemp: HP's internal temperature sensor value
remotetemp: a construct to feed an external thermometer's value back to the HP for the HP to use in lieu of it's internal temp (roomtemp).

As for timers... definitely can put them back.  I have 0 use for them cause I want all my automation to be done by OH, not the HP's... I just removed the code from mine to reduce sketch size and mqtt load on my broker... not that either really matters.

@SwiCago  enjoy your vacation!  don't focus on our banter :)

F43RY, 2018-08-10:
Thx for the explanation of temp vars, infact  I didn't understand the existance of remoteTemp. In my case I don't need it.

For timers I'd like just display status, I don't care set them (sometimes I forget if I set them by remote, and sometime I forget where I left the remote......).