SwiCago/HeatPump #89: Can transmit but not receive (Nodemcu)

svzguy, 2018-08-24:
I have my SVZ heatpump connected to a Nodemcu. This one has an onboard regulator, so I have the 5V from the Mitsubishi unit connected to the 5V of the Nodemcu. I can transmit data and was able to set the desired setting (e.g. cool, low fan) to the unit. However, no data is being transmitted back, neither the temperature.

The RX/TX are ok though, I tested this.

I do use the pullup resistors on RX and TX.

svzguy, 2018-08-24:
I measured the voltage between RX/GND and TX/GND, and while TX/GND is at 3.3V, RX/GND is 5.5V. I don't think my NodeMCU can handle 5V. So I would need a logic level shifter to step that down, and while I am at it, I should probably also shift up the 3.3V from TX to 5V, even though the heatpump seems to interpret 3.3V as high...

Can anyone confirm? Surprised I didn't find anything like this on the blog post comments...
(will cross post over there as well)

SwiCago, 2018-08-25:
#61  #77

closed 2018-08-25 by SwiCago