SwiCago/HeatPump #90: Resistors in the diagram

mdoary, 2018-11-02:
What are the Resistors purpose in this diagram?

SwiCago, 2018-11-02:
They are pull ups. The heat pump TX and RX float. If you use level shifters you should be fine. If you use a hazzah you should be fine as well.

closed 2018-11-02 by SwiCago

jascdk, 2019-05-11:
So if I use the ESP01 breakout board I can connect it directly to the heatpump without the resistors as mentioned in the diagram ?

Stendec1, 2019-05-12:
Indeed:

Met vriendelijke groeten,
With kind regards,

Rudi Schoors.
(Sent from my mobile)

jascdk, 2019-05-12:
@Stendec1  Great thanks for the answer:)

jascdk, 2019-06-06:
@Stendec1 Hi again:) OK - I am soon having all the parts for the project:) Well - if I use the ESP-01 breakout board together with an small esp 01 module - will it then be identical to the schematic this projects has in the readme. Is there pulp resistors in the breakout board? I read it has an level shifter - is that the same as using the 2 resistors in the diagram? 

If I choose to use an demos d1. mini. then I have no pull-ups on the rx and tx pins right? so the best option is to use the small 01 module together with the breakout board?

Sorry for asking, but im pretty stupid about this :)

aqualx, 2019-06-07:
You can use regular ESP-01S board + this [ESP8266 ESP-01 ESP01 Serial WiFi Wireless Adapter Module](https://www.aliexpress.com/item/ESP8266-ESP-01-ESP01-Serial-WiFi-Wireless-Adapter-Module-3-3V-5V-Compatible-Serial-Board-For/32831958321.html) and directly connect to CN105 port.

jascdk, 2019-06-07:
@aqualx Hi:) Thanks ! I ask a lot - sorry - it is because my low knowledge of electronics:/ I have the parts you mention - so using these parts would be better than choosing a D1 mini and connect that only using the cable connected to the corresponding pins?