SwiCago/HeatPump #93: onConnectallback() called every 2 seconds

salas81, 2018-11-15:
After calling heatpump.connect(&Serial) the onConnectCallback() gets fired every 2 seconds (approximately). Looks like esp is not able to connect

SwiCago, 2018-11-15:
If it can't connect, then you have a wiring problem. Some esp modules don't work right with the HP. Do you have pull ups? Is your HP a supported model?

salas81, 2018-11-15:
I have pull-ups, but how do I know if my model is supported?

SwiCago, 2018-11-15:
https://github.com/SwiCago/HeatPump/wiki/Supported-models

SwiCago, 2018-11-15:
Hmmm, only thing I can think of is that your wiring may not be good. RX/TX flipped or wires not crimped right.

SwiCago, 2018-11-18:
@salas81 , have you figured out your problem? It isn't the library. It is doing what it is supposed too. Attempt connect every 2 seconds until successful and send a signal to callback if defined, so the end user can further pass that and be notified of issues. Some guys use that call back to let them know that their HP lost connection.

SwiCago, 2018-11-19:
Closing. Hit us up in the gitter chat, if you are still having issues with connection

closed 2018-11-19 by SwiCago