SwiCago/HeatPump #96: Related project

dgoodlad, 2019-01-03:
HI @SwiCago 

I've just finished building [dgoodlad/esp8266-mitsubishi-aircon](https://github.com/dgoodlad/esp8266-mitsubishi-aircon) based on your HeatPump library, thanks for so much hard work! I wonder if you've any interest in pulling this project in as an example, or to link to it? The big changes from your README and home assistant example are:

* Uses a TXB0104 logic level shifter rather than a pullup resistor to talk 5V
* Switches modes in software based on whether 5V power is coming from the heatpump or from another source (e.g. FTDI/USB power) for development purposes
* Uses the alternative UART pins (`Serial.swap`) to keep the ESP's bootup messages off the heatpump's serial lines, which I'd found sometimes caused it to panic and reset
* Easy support in Home Assistant via the [MQTT HVAC Component](https://www.home-assistant.io/components/climate.mqtt/)
* Online/offline availability state in HA via MQTT's "last will" feature
* [WiFiManager](https://github.com/tzapu/WiFiManager/tree/development) for configuration

I have a multihead compressor with five heads in my home, and they're working great with this hardware+software now :smile:

SwiCago, 2019-01-03:
Hi @dgoodlad , nice custom board. 
I think we could link the projects, however I believe it be then best if you link to my library instead of copying it in a sub folder. We can add the serial swap, so that your setup still works, even though I have not had the issue you described. This would be a smart way of approaching it, so that your custom finished variation uses latest code. 
Char* is something I would want to convert to, but that requires all examples to be updated too. Maybe you can help with that, it would be appreciated. 
I would like to keep this project mainly about the library and the examples are just that. Examples to get people started, not be finished projects...however as said, I don't mind adding your project as a complete finished example, as long as it is not hosting the library, but rather links to it. In my examples we use MQTT, but I don't host the library as it is updated and there are other variations. I just note that we use it.
Let me know if that works for you.

dgoodlad, 2019-01-03:
@SwiCago absolutely agree on not copying the library; I did so initially so that I could make the changes that I wanted/needed and iterate on those without bugging anyone with PRs :smile: I'd be happy to get my changes upstreamed here, will submit some pull requests soon.

:+1: to keeping this repo about the library, you're right that's the best approach.

SwiCago, 2019-01-03:
Thanks, a long time back someone did the char* conversion, but unfortunately that pull is too far out of date and as said, all the examples need to be compliant. They are short, so it shouldn't be too much work. So, when you do your pull, can you pull with at least the primary examples updated? We can work on the sub examples after that.
One thing I want to add in the near future is HP error code detection. I know what message has it, just need to add all the codes and meanings as a sub lib or something.

unixko, 2019-01-27:
@dgoodlad I'm very interested in your code. It is future-proof to use native HA component rather than custom component. HA always has breaking changes so I have to constantly modify custom component. Also MQTT HVAC component just now supports more command topic then it is perfect time to migrate. At first I tried to figure out how to utilize your repo for a week on my current hardware but I was not success to strip out some hardware-specific codes. Now you are kindly fix example codes also for hardware that you didn't even use. Big thanks!

SwiCago, 2019-01-27:
@dgoodlad , no problem we all have our daily lives to get to...Anyhow, I have a few questions/concerns which I mention in the pull. please review and let me know.

SwiCago, 2019-01-29:
@dgoodlad char* looked good. Hopefully nothing broke lol

unixko, 2019-01-30:
I tested char* library with new example from PR on existing hardware and it worked normally. Confirmed.