SwiCago/HeatPump #97: Pigtails HOWTO?

notlaforge, 2019-01-09:
Hold my hand here? :)

I see the links to digikey parts, but I'm not entirely sure how to make these pigtails. Do you need a crimping tool of some sort? Five contacts per housing?

notlaforge, 2019-01-09:
Ok, so I found this, seems pretty good: https://iotexpert.com/2018/01/18/jst-connector-crimping-insanity/

fbnts, 2019-01-09:
Digikey do also stock individual wires with the crimps already on.  You just then push them into the plug housing.
This is the housing I used:
https://www.digikey.co.uk/product-detail/en/molex-llc/0510670600/WM3333-ND/2405304

These are the pre-crimped wires:
https://www.digikey.co.uk/product-detail/en/jst-sales-america-inc/APAPA22K305/455-3086-ND/6009462

notlaforge, 2019-01-09:
@fbnts that housing has 6 positions. It works?

fbnts, 2019-01-09:
@notlaforge Oops, wrong link, this is it: https://www.digikey.com/product-detail/en/jst-sales-america-inc/PAP-05V-S/455-1489-ND/759977