SwiCago/HeatPump #109: Configuration Error

naldoxable, 2019-02-07:
From today, after a reboot I'm getting this message:

```
019-02-07 19:41:04 ERROR (MainThread) [homeassistant.components.hassio] INFO:homeassistant.util.package:Attempting install of colorlog==4.0.2
Testing configuration at /config
Failed config
  climate.mitsubishi_mqtt: 
    - Invalid config for [climate.mitsubishi_mqtt]: [temperature_state_topic] is an invalid option for [climate.mitsubishi_mqtt]. Check: climate.mitsubishi_mqtt->temperature_state_topic. (See ?, line ?). Please check the docs at https://home-assistant.io/components/climate.mitsubishi_mqtt/
    - platform: mitsubishi_mqtt
      command_topic: heatpump/Matteo/set
      name: Condizionatore
      state_topic: heatpump/Matteo
      temperature_state_topic: heatpump/Matteo/status
Successful config (partial)
  climate.mitsubishi_mqtt:
```

I have been using this component for months, always updated and everything worked as a charm.
I haven't changed the configuration and I do not understand what is going wrong. Now the component is not working at all.

vlad36N, 2019-02-07:
Same. Today update HA to 0.87.0, after restart error in log file and all my climate 3 AC gone from UI.
Error:
2019-02-07 14:36:06 ERROR (MainThread) [homeassistant.config] Invalid config for [climate.mitsubishi_mqtt]: [temperature_state_topic] is an invalid option for [climate.mitsubishi_mqtt]. 
Check: climate.mitsubishi_mqtt->temperature_state_topic. (See /config/include/climate.yaml, line 5). Please check the docs at https://home-assistant.io/components/climate.mitsubishi_mqtt/

SwiCago, 2019-02-08:
I do not maintain integrations. They were implemented by others and are not part of the library, but rather as examples of how you can use the library. I will not maintain them.

closed 2019-02-08 by SwiCago

david-yam, 2019-02-08:
I have just created a pull request #110  for fixing the issue.

SwiCago, 2019-02-08:
Thank you for helping HASS folks out.

SwiCago, 2019-02-12:
@david-yam  someone reports your changes don't work right. Can ypu look at #111