SwiCago/HeatPump #111: Not working with home assistant 0.87.0

kentivar, 2019-02-11:
Updated to home assistant 0.87.0, downloaded latest custom component from repo that was updated for 0.87, restarted hass, and its not working :( 

Log:
019-02-11 11:32:43 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:32:57 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:33:57 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:34:57 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:35:08 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:35:57 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:36:20 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None
2019-02-11 11:36:58 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 232, in async_update_ha_state
    state = self.state
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/climate/__init__.py", line 207, in state
    if self.current_operation:
  File "/config/custom_components/climate/mitsubishi_mqtt.py", line 169, in current_operation
    return me_to_ha[self._current_operation]
KeyError: None

SwiCago, 2019-02-11:
I specifically noted on the integrations page that they are not part of the library and issues should not be opened for them. Use gitter chat for that. Someone there will help. 
I suggest gitter chat or look up who last updated thatbintegration and chat with them directly

closed 2019-02-11 by SwiCago

SwiCago, 2019-02-12:
I have a feeling the default value of "none" on line 86 is the issue. It is not in the list of modes referenced in me_to_ha which is derived from ha_to_me on line 47/48 ... most likely either needs a valid default, such as 'OFF' or another if statement to deal with 'none', until hp updates the values. But again I am going out on a limb here. I don't use home assistant and really should have to deal with it in a library that doesn't have to do with it.