SwiCago/HeatPump #115: Question (SRK Aircon)

coyote1, 2019-03-06:
Hi Swicago,
is it possible to use this library with an SRK Aircon or is it possible to integrate "rjdekker's" library in your code? Unfortunately, he doesn't continue on his code and doesn't answer on Github. Maybe someone can embed this code or continue?
I often have problems with the setup of rjdekker and i don't know how i can debug this.
Sometimes the ESP no longer reconnects to MQTT after a connection break or sometimes strange subscriptions come in mqtt (like Room%&§, or Air§$%%) maybe someone can help me to debug this?!

SwiCago, 2019-03-06:
Hi @coyote1, his library was for heavies via SPI. This library is serial for non heavy units. I also believe the byte code is much different too. It would like putting two libraries into one, which makes no sense. 
It is best to keep them separate. 
I suggest you fork his and contact everyone that has also a fork or is using it and work together to fix it/normalize it. When it is stable enough,  I can add a link for heavies to redirect them to your repo and vs versa. You can look at my mqtt sketch to see how reconnects are handled for mqtt, that might help you some. Last I remember he was having issues getting esp to be an spi slave or was it master? Hence his esp and non esp bridge units. 
If you create a forked repo, let me know and I'll check in. We also have gitter chat here, which might help with other questions.

closed 2019-03-06 by SwiCago

coyote1, 2019-03-06:
Ok, thanks for your answer and explanations.

andig, 2019-12-02:
> his library was for heavies via SPI

I'm only getting started, so please excuse the question. Is this expected to work for Mitsubishi Heavy SRK 20 ZS-W? I understand the connector should be internally available, offers power and does not need any additional "wifi" or similar adapter installed?