SwiCago/HeatPump #117: MQTT topic must contain "heatpump" to work???

pippin88, 2019-03-17:
This might be an odd one...

Firstly, thank you for the library. I've been implementing it on an ESP8266.

I'm basing things off the OpenHAB integration example.

In my part of the world, we use the term aircon rather than heatpump - so I naturally changed the MQTT topics to match:
home/ac/back (short for home / aircon / back bedroom)
and it sort of worked... except I was getting random mode changes at times...
I did some digging through the code, and thought I should I be seeing the current settings published to home/ac/back (in place of home/heatpump), but I never saw the current settings obtained from the heatpump published there. The status and debug topics are fine (home/ac/status, home/ac/debug).

If I change the main topic (heatpump_topic) back to home/heatpump, I get the messages as expected.
E.g. when the ESP boots, I get {"power":"OFF","mode":"COOL","temperature":25,"fan":"2","vane":"5","wideVane":"|"}.

Further testing has shown:
I can add to the topic, e.g. home/back/heatpump works.
home/heatpump1 does not work - nothing is published to home/heatpump1

The other topics all seem to behave fine with whatever name I give them, but the heatpump_topic (= "home/heatpump";  //contains current settings) seemingly must end in "heatpump" for the code to function properly!

I believe this was causing my random mode changes, as openhab was never seeing what mode was already set, and I think picking a random one to start with.

My coding is not good enough to track this issue down better. I'm a copy others to get my projects done sort of person.

SwiCago, 2019-03-18:
Not sure about the openhab stuff, but you can have any topic you want. Just keep it same everywhere
Example:
onst char* heatpump_topic              = "home/ac/back;
const char* heatpump_set_topic          = "home/ac/back/set";
const char* heatpump_status_topic       = "home/ac/back/status";
const char* heatpump_timers_topic       = "home/ac/back/timers";

const char* heatpump_debug_topic        = "home/ac/back/debug";
const char* heatpump_debug_set_topic    = "home/ac/back/set";

As mentioned in mqtt example header, you will need to adjust PubSubClient.h  for longer topic name, due to max packet length
// Note PubSubClient.h has a MQTT_MAX_PACKET_SIZE of 128 defined, so either raise it to 256 or use short topics

closed 2019-03-18 by SwiCago

pippin88, 2019-03-18:
Thanks for the reply.

I was already using uniform topics names as you suggest.
Today, renaming the topics is working...
I can't explain what is different. Last night I spent quite a while checking and the behaviour was reproducible.

Maybe I had a hidden typo in my code? I did however start again with the examples and still had the problem.

Anyway, happy to have it working as expected!

SwiCago, 2019-03-18:
👍