SwiCago/HeatPump #118: Arduino IDE Compile fails

wellsy57, 2019-03-23:
Using Arduino IDE I have downloaded the library and setup ready to flash a Wemod D1 mini but I get:

```
C:\Users\Steve\Documents\Arduino\arduino-1.8.5\portable\sketchbook\HP_cntrl_esp8266\mitsubishi_heatpump_mqtt_esp8266_esp32\mitsubishi_heatpump_mqtt_esp8266_esp32.ino:9:22: fatal error: HeatPump.h: No such file or directory

 #include <HeatPump.h>

                      ^

compilation terminated.
```
I tried looking for a heatpump library and found HeatpumpIR by Tony Arte so I Installed that but still fails looking for HeatPump.h

Thanks in advance if you help.

SwiCago, 2019-03-24:
Why would you try a heatpump library from a totally different dev? Tony Arte Heatpump IR has nothing to do with this library. Re-read the readme, it is very clear as to where you are to put this library. You have to install this library to use the example sketches

closed 2019-03-24 by SwiCago

wellsy57, 2019-03-24:
Why would I? Maybe because I am fairly new to the world of flashing devices but have a good grasp of flashing tasmota onto sonoff devices with Arduino IDE. The support/documentation at tasmota was very useful to get that all sorted. Documentation on this project (for newbies) is scarce I think?

After sorting out where I had placed the folder with your library I encountered problems with a conflict with Arduino Json Version 5 and 6. The compiler gave an error and sent me [here](https://arduinojson.org/v6/doc/upgrade/). 
Ended up much easier to just downgrade from Json 6 back to Json 5 and that then the compile worked. You may want to give people a heads up on that.

Cheers mate!

SwiCago, 2019-03-24:
If you have any more questions, please use the gitter chat. The link is in the upper right corner of the initial page. This library is not really for "newbies" as you call it, please be careful and don't destroy your heatpump. Again, reach out in gitter, lots of experienced people willing to lend a hand, so you won't fry. 
What version of IDE are you using? I am on latest and the json works fine, but I may be using an older version. When this library was created json for arduino was a third party library, as is pubsubclient.