SwiCago/HeatPump #119: Upgrade code to support JSON library 6+

SwiCago, 2019-03-24:
Current code was written with JSON library version 5. Please use 5 for now, until I can upgrade it to 6.
Upgrading to 6 is not difficult, but requires some changes
Ex:
`const size_t bufferSize = JSON_OBJECT_SIZE(6);
DynamicJsonDocument root(bufferSize);
root["test"] = "test";
char buffer[512];
root.printTo(buffer, sizeof(buffer));`
Becomes:
`const size_t bufferSize = JSON_OBJECT_SIZE(6);
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonObject& root = jsonBuffer.createObject();
root["test"] = "test";
char buffer[512];
serializeJson(root, buffer);`

SwiCago, 2019-03-24:
I have made an experimental MQTT example. Code converted to JSON 6 library. 
https://github.com/SwiCago/HeatPump/tree/JSON6_test
@wellsy57 or anyone else willing to give it a try?

wellsy57, 2019-03-24:
I will flash one of my devices with this example and give you feedback after I have done so.

Maybe (while this is experimental) you should just advise people about the error if using Json 6 to let them know they can just downgrade to Json 5 for now?

SwiCago, 2019-03-24:
@wellsy57  already updated the readme with a note about json 5/6 and also posted in gitter

wellsy57, 2019-03-24:
Cool!

wellsy57, 2019-03-25:
Using Arduino IDE 1.88
Downloaded your test files to the correct spot
UN-Installed ArduinoJson by Benoit Blanchon version 5.13.5
Shut down IDE and restart
Installed ArduinoJson by Benoit Blanchon version 6.91
Shut down IDE and restart
Opened IDE and start to compile project....stops on many errors
After reading them found references to multiple sources and decided to ditch the Arduino/libraries/ArduinoJson Folder.

Restart the process and all good this time.
MQTT reports _1553506149: New client connected from 192.168.1.122 as heatpump-controller-2 (c1, k15, u'WTF')._

SwiCago, 2019-03-25:
@wellsy57 , do all functions and report backs work as expected?

wellsy57, 2019-03-25:
Unfortunately I am not able to connect to the AC at the moment as I’m waiting on the pigtail to connect to the CN105 connector.

SwiCago, 2019-03-25:
Ok thanks, maybe someone else will chime in before then, else we just wait

wellsy57, 2019-03-26:
progress....I managed to get a suitable pigtail locally today (after waiting a month and the wrong ones turned up...grrr)
So the wemos is nicely fitted in the AC control enclosure. Still talking to MQTT.

So...I added the climate config to my homeassistant config and then setup the custom_components/mitsubishi_mqtt/climate.py 

But on restart I get and ERROR "Unable to find component climate.mitsubishi_mqtt"

I've searched all through the forum and it seems nobody else has had this error.
Any ideas?
I have a few other custom components which all work fine so not sure why this one should fail to be found?

SwiCago, 2019-03-26:
No clue. I don't use homeassistant. In fact other wrote it and I have asked many times for someone else to take it over in a new repo as it really doesn't belong in this repo. I will probably delete it soon. Too many people having issues with it. Your best bet is to first see if you can control/read from hp via mosquitto sub/pub manually. That will at least confirm the hp library works, then join gitter and ask others for help as there are folks in there that know this component well

gysmo38, 2019-03-27:
@wellsy57 I create another example (https://github.com/gysmo38/HeatPump/tree/master/examples/mitshubishi_mqtt_ha) where you don't need to add third part componement. The example with you mqtt autodiscovery to create to device. It should be easier to use. For the plugin it stop work with last version of HA.

wellsy57, 2019-03-27:
@gysmo38 sounds good thanks for the heads up!
Which version of HA is best to use it with?
By the “third part” do you mean no custom component?
I am busy for a couple of days but will report back when I have tried this example.
Cheers for the help to all!

SwiCago, 2019-03-27:
Guys, please don't intermix code here. This issue I created, was about testing json 6 with the branch I created. After we confirm this is fine, you guys can collaborate on how you want to deal with HA. I have removed your @gysmo38  comment  from this issue. You shared your version in the git. If you want to update HA code for auto detect, then make a pull for that. Thanks

SwiCago, 2019-03-27:
@wellsy57 , let me know if the test branch I created for mqtt json6 works or not. You don't need HA to test that, just mosquitto sub/pub. Thanks

wellsy57, 2019-03-28:
@SwiCago Ok I understand you are saying that the test example can be tested with MQTT. Can you give some examples of topics to subscribe to and publish to?
I have subscribed to heatpump/status and it sends me {roomTemperature":0,"operating":false} every minute and that's the only topic that seems to be working atm?
Also subscribed to heatpump but n/a is all I get back.
I have maaged to write some automations and rules that publish MQTT but by no means anything more than a newbie with the subject.

SwiCago, 2019-03-28:
@wellsy57  i replied to your comment in gitter. I do not think you are connected to the hp. Do you have the pull up resistors?

SwiCago, 2019-03-28:
Also note pubsubclient byte limit as described in the .h of the mqtt example 
// Note PubSubClient.h has a MQTT_MAX_PACKET_SIZE of 128 defined, so either raise it to 256 or use short topics

gysmo38, 2019-03-30:
@SwiCago I found time to test. It's not work for me.
I have this output
heatpumpdev/status {"roomTemperature":18,"operating":false}
heatpumpdev/timers {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
heatpumpdev/debug {null:null}
I have same issue with my version for HA when I use DynamicJsonDocument

gysmo38, 2019-03-30:
I made some test with debu packet.
If i replace 
const size_t bufferSize = JSON_OBJECT_SIZE(1);  to     const size_t bufferSize = JSON_OBJECT_SIZE(10); 
It work.
heatpumpdev/debug {"packetRecv":"fc 62 01 30 10 06 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00 52 "}
heatpumpdev/debug {"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}

gysmo38, 2019-03-30:
With some test it work with this code
   const size_t bufferSize = JSON_OBJECT_SIZE(1);
    DynamicJsonDocument root(bufferSize);
    String pd = String(packetDirection);
    String ms = String(message);
    
    root[pd.c_str()] = ms.c_str();
    String buffer;
    serializeJson(root, buffer);

    if (!mqtt_client.publish(heatpump_debug_topic, buffer.c_str())) {
      mqtt_client.publish(heatpump_debug_topic, "failed to publish to heatpump/debug topic");
    }
  }

Others packets look like to work. Only debug packet failed

SwiCago, 2019-03-30:
@gysmo38 , thanks for testing. Will look at it when I have a chance. I think the json_object_size(X) guesses the needed chars. In JSON 5 was dynamic expanded, in JSON 6 it needs to be equal or greater to be able to adjust. In thoery we know how long each packet is, so it could be hard coded. Or just leave at x =10, since that should be plenty long lol

SwiCago, 2019-04-12:
@wellsy57 , any progress with you HPs?

wellsy57, 2019-04-12:
Good timing @SwiCago ! I posted on gitter earlier today..."Progress has been made! I received my new leads and tried the Wemos D1 devices with no success.....very disappointing! So I thought it best to retry the ESP01 devices now that I had the new leads. Flashed them successfully using Arduino IDE, made up the leads to suit, checked all wiring, fitted to the AC and success! VERY Happy! Next step is to try integrate with HA. Will report again once that is tested. Thanks to all for your help and encouragement!"
and
"All went smoothly and the unit is controllable via the Thermostat Control. I have a ‘however’ though. It seems something is amiss (even though it is working?) as I have errors in HA telling me there are problems. I feel certain it is a problem with the way the custom folder is setup. I know this the wrong place to report HA problems so I will just post a link to my thread in the HA forum https://community.home-assistant.io/t/mitsubishi-ac-with-wemos-d1-mini-pro/107007/22 if anyone is interested."

BUT: the example code used was the Master and not the Json6_test.
I have another unit to setup and also another two ESP01 modules that I will test both versions and report back...might take a few days though.

SwiCago, 2019-04-12:
Don't forget, once up and running you can use OTA.

wellsy57, 2019-04-12:
That would be useful maybe...how do I connect? I thought it would be via Tools/Port/Network but there is no network option available atm?

SwiCago, 2019-04-12:
You can OTA via arduino IDE. Set debug to "OTA" and select the OTA target under com port section. Your device(s) should be listed with their respective name and ip address

wellsy57, 2019-04-15:
@SwiCago I have flashed my ESP01 with the json6 test (using ArduinoJson version 6.10.0)  earlier and found that although the settings are as per the my other device (except OTA is enabled?) the test version device: 
1. fails to connect to the Mosquitto server
2. does not blink the blue led 
Also
If the Serial Monitor via USB port is viewed it shows:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```

While...The earlier version shows:

```
  ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
PLUS...A SERIES OF REVERSED ??????????????????????????????????????????????????? Which grows presumably as it sends packets?
```
I did not try it connected to the AC as it is not working correctly it seems.

SwiCago, 2019-04-15:
@wellsy57 ok thanks, I'll have to bench for a bit until I have time to look deeper into it and get one of my devices out. Unfortunately, OTA came after I got all mine installed, so I need to disassemble one of my units to add OTA. For now, to get you up an running, I suggest using the prior JSON, leave OTA enabled so you can update it, once I have a fix.

wellsy57, 2019-04-15:
@SwiCago That’s sounds like a plan!

unixko, 2019-05-04:
I tested with arduinoJson 6.10.1. All topics are working properly except debug_topic:

heatpump           -> working, report current setting correctly.
heatpump/debug_topic = {null:null}
heatpump/status -> working, report current temp and operating normally.
heatpump/timers -> working, tested by set timer on IR remote control.
heatpump/set      -> working, can set mode/temp/fan/vane normally.

I can easily access to esp8266 board and I have http ota so please let me know if you need to test more.

SwiCago, 2019-05-04:
@unixko , since debug is fairly large, maybe adjust the buffer for that one to be larger, maybe make it a 2 or more, since 1 is not enough. Can you try that and get back to me. Thanks

unixko, 2019-05-04:
I tested and it must be 6 or higher then debug topic works.

SwiCago, 2019-05-04:
@unixko thanks, so all others were fine? Just this one needed 6?

SwiCago, 2019-05-05:
@unixko i upped the buffer for debug. I will be merging my json branch to master soon.

SwiCago, 2019-05-05:
The merge with my json 6 is done. Closing this issue

closed 2019-05-05 by SwiCago

unixko, 2019-05-05:
I confirm that all topics are working. My test result is:
main topic must be 6+.
status topic must be 2+.
timers topic must be 5+. Same length as debug but doesn't need 6?
debug topic must be 6+.
set topic can be 2 if set one parameter at time (mode/temp/fan/vane) but spare 6 for full parameter set command is better.

Here is sample payload from JSON6:
```
(6) heatpump = {"power":"ON","mode":"COOL","temperature":24,"fan":"AUTO","vane":"AUTO","wideVane":"|"}
(2) status   = {"roomTemperature":29,"operating":true}
(5) timers   = {"mode":"BOTH","onMins":320,"onRemainMins":320,"offMins":1340,"offRemainMins":1340}
(6) debug    = {"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
(6) set      = {"temperature":22.0}
```

SwiCago, 2019-05-05:
Great, glad all worked