SwiCago/HeatPump #120: Not work with async MQTT and Wifi

dzungpv, 2019-03-26:
For some reason  I would like to use async with with wifi and mqtt. I use this lib: https://github.com/marvinroger/async-mqtt-client but HeatPump lib will enter endless loop and not respond, OTA will not work too.
I use this sample and add HP code https://github.com/marvinroger/async-mqtt-client/blob/master/examples/FullyFeatured-ESP8266/FullyFeatured-ESP8266.ino. I could not debug it because HP use hardware serial, may be i will try to use software serial for communication and debug it

SwiCago, 2019-03-26:
Hi, it was coded using https://github.com/knolleary/pubsubclient, use it instead. See of that works, if not then something else is the problem.

SwiCago, 2019-03-26:
Re-ooen issue if you need more help or join gitter chat please

closed 2019-03-26 by SwiCago