SwiCago/HeatPump #123: externalUpdate must use together with autoUpdate enabled

hawkefly, 2019-04-13:
externalUpdate must use together with autoUpdate enabled, IMO,externualUpdate should be use separatly. I think this may be a bug.
Because of these line of code . When autoUpdate is false , the expression always false.

https://github.com/SwiCago/HeatPump/blob/c48c80046c22855e42badd090a8a2ea071d1146d/src/HeatPump.cpp#L588

SwiCago, 2019-04-13:
Hi @hawkefly , this was by design.
autoUpdate tells the library it is OK to send updated wantedSettings right away, otherwise you have to send hp.update() manually.
This is used in combination with externalUpdate, where when a person uses IR remote to change settings, they want wantedSettings synchronized. 
The two variables work together and don't apply to 90% of the users of this library. Most of use do not use IR remote anymore, but rather HA openHAB or some other Automation control/web interface. 

Example: if autoUpdate == false and externalUpdate == false, then library will never update HP, even if currentSetting(HP) is out of sync with wantedSetting(esp8266) until someone sends hp.update() in code.

 if autoUpdate == true and externalUpdate == false, then the library will automatically send wantedSettings to HP and resync HP if it was changed by a person using IR. Basically a lock out. If a child or spouse changes the settings to make it warmer, instead of putting on a sweater LOL, then HP will auto revert back the way it was.

if autoUpdate == true and externalUpdate == true, then the library will automatically send wantedSettings to HP and will update wantedSettings from currentSettings, if changed by IR. Basically keeping them in sync.

They work together and was by design. As said 90% of use have no use for externalUpdate, but someone wanted it and this was what was agreed on for implementation.

hawkefly, 2019-04-13:
Thank you very much. @SwiCago 
I try to enable autoUpdate. But I have an issue, it is same as the issue  #79 .
Enable auto update will cause it send the setting package forever .
I am using the example code mitsubishi_heatpump_mqtt_esp8266_esp32. When enable autoUpdate , I already comment this line.
https://github.com/SwiCago/HeatPump/blob/c48c80046c22855e42badd090a8a2ea071d1146d/examples/mitsubishi_heatpump_mqtt_esp8266_esp32/mitsubishi_heatpump_mqtt_esp8266_esp32.ino#L235-L239
This is the screenshot
[image]

SwiCago, 2019-04-13:
Yes, it will keep sending the auto update(wantedSettings), until the HP finally responds with the currentSettings. If wanted and current != and auto update is on, it will send them over and over until they are in sync. This can take many times until the HP finally responds. It is async response, even the ack. So depending on unit it can be pretty quick or pretty slow. 
Recommendation in this case is not to use autoUpdate, if it does not do what you anticipate. The library was written to be unit agnostic, as in you can send wideVane, even of you unit does not support it. So some things may not work as you would hope. 
If I were you, I would leave autoUpdate off. Maybe see how long it takes for the unit to confirm currentSettings after update, as in see how long it takes for wanted to match current and then you can maybe make some adjustments yourself to meet your needs.

hawkefly, 2019-04-14:
Yes , I can disable auto update . But unfortunately , I cannot use externalUpdate in this case.

hawkefly, 2019-04-14:
The problem is :
When sending "fc 41 01 30 10 01" package to update the setting . And the receive the "fc 61 01 30 10 00" package , so (packetType == RCVD_PKT_UPDATE_SUCCESS) is True.
https://github.com/SwiCago/HeatPump/blob/c48c80046c22855e42badd090a8a2ea071d1146d/src/HeatPump.cpp#L125-L133

In line 127,  you sync a RQST_PKT_SETTINGS immediately.  But this fail . So, you can not see the package "fc 41 01 30 10 02" in the screenshot.  because of this,  currentSettings will never update. So it will send update package (01) over and over

SwiCago, 2019-04-14:
@hawkefly i will have to review this a bit closer. A contributor years ago came up with this, as they had a need for it. I guess you are the only one since then needing this, since then. Can you open a new issue in regards to autoUpdate. With your details here. I believe if can resolve this, then external will work as expected as well.

SwiCago, 2019-04-14:
Going to leave this issue open for now and see if from my above EDIT if this can get reaolved. Thank you @hawkefly  fornyou hawk eyes ;)

hawkefly, 2019-04-15:
^_^
May be you can set infoMode = 0 to reset the package request  loop from the beginning(0x02).

SwiCago, 2019-04-15:
Unfortunately that will not work.  Because as soon as sync is hit again, PACKET_TYPE_DEFAULT is the current packet. And so line 139 would be true again, you'd be stuck in a update loop and never break free.

Line 127 should be changed to 
`if(autoUpdate) {
while(!canSend(true)) { delay(10); }
sync(RQST_PKT_SETTINGS);
}`
I feel calling sync as soon as update happens fails all the time, even when manually called. So no point in doing it in normal case, as sync will eventually call RQST_PKT_SETTINGS on its own. But for autoUpdate we need to get this data from HP as soon as possible, to make sure what was auto sent matches what HP has. Wanted == current, other wise auto will update again and we have same issue as you state before.
Doing this code change would not effect standard usage, only when autoUpdate were used. 
What do you think?

hawkefly, 2019-04-15:
Yes,I change line 127 to you suggestion and tested for a long time. This works like a charm now. Thank you :-)

SwiCago, 2019-04-15:
Cool, thanks @hawkefly  i will update master branch now with the update. Thanks again for being patient, but not pushy.

commit 2bbc2e92 (2019-04-15): Fix #123

Fix for #123 adverse effect in sync loop when autoUpdate is enabled
closed 2019-04-15 by SwiCago