SwiCago/HeatPump #126: Handling of Unit Function Settings

luiscgalo, 2019-05-02:
Hello,

First of all, thanks for the providing very usefull information about Mitsubishi Electric AC unit protocol.
It allows development of neat automation solutions for better control of HVAC systems. ;)

I've managed to successfully integrate the source code available in this repository with PEAD-M140JA units.
Everything is working as expected and I'm able to perform remote monitoring and control of the ducted AC unit.

However, I was wondering the possibility of collecting and/or changing the named "Function Settings" of referred units via CN105 connection.
The "Function Settings" are advanced technical parameters that can be set within service menu of wired remote control such as the PAR-32MAA (refer to picture below).
[image]

Those settings allow to set useful parameters such as behaviour of indoor fan when setpoint temperature is reached, locking of wired remote control, thermistor being used, etc.

At begging I thought that these setting could only be changed via wired remote control (impossible via CN105 serial interface).

However, according to the specs of MHK1 controller, it is possible to adjust referred settings.
Since MHK1 controller is connected by CN105 serial connection, it means that we can indeed control "Function Settings" through serial interface.

On the following link there is clear mention to this capability of MHK1 controller.
https://www.greenbuildingadvisor.com/question/mitsubishi-minisplit-secret-advanced-settings

Additionally, the following PDF presents a full list of supported "Function Settings" (from page 6 to page 8)
[remote-controller-function-settings.pdf](https://github.com/SwiCago/HeatPump/files/3137313/remote-controller-function-settings.pdf)

Has anyone tried to change "Funtion Settings" via CN105 interface?
Since I don't have an MHK1 interface, I don't know which serial packets are exchanged between AC unit the MHK1 controller...

For who has an MHK1 controller, there is any chance of dumping serial packets exchanged when changing "Function Settings" of AC unit? :)

I think implementing this feature on the CN105 protocol will be a very nice improvement since it allows much more flexibility to adjust advanced operation settings of AC unit.

Thanks in advance for your help,
Luís Galo

SwiCago, 2019-05-04:
Those are control settings of the mhk1, not cn105...
For example, when temp reached turn off fan or leave running. That is easy to do with this library via automation. If you want to turn off the fan when temp is reached, then turn it off. If you want the blower fan to continue running, then switch mode to fan instead of cool. 
Same thing with special modes with ir remote, such as max cool, which is just cool plus max fan in one command. 
Other than simulating commands mhk1 or similar controllers do, I know of no such commands..but if you have time, you could connect an analyzer to the serial and see yourself what mhk1 sends to the unit, when you change the settings. I am going to bet it sends nothing, but rather that is an internal thing of mhk1 unit or similar control units.

closed 2019-05-04 by SwiCago

luiscgalo, 2019-05-04:
Hi SwiCago,
Thanks for your response.

Well, sorry to insist but I'm sure that referred settings are stored on the indoor unit memory (not on remote controller).

I've a PAR-32MAA wired remote and I see two interesting messages when accessing the "Function Settings" mode:

1. When reading settings from AC to the wired remote it takes about one minute and the display shows a text message that says "reading settings from unit...".

2. When saving "Function Settings" changes to AC unit it takes again several seconds and another message is displayed: "transmitting settings to unit...

This means that there's communication between wired remote controller and the AC unit.
I you take a look on the PDF I referred above, it says that takes several seconds to retrieve and save settings.
The reason is because those settings are stored on the AC unit.
And it make sense since some settings are AC unit dependent... (not remote controller logic)

Never a wired controller takes 40s to read/write some bytes from its internal memory 😀

I guarantee that "Field Settings" belong to the indoor units since not all functions are available for every unit.
And there are other very useful settings such as static pressure of the indoor unit, automatic recovery after power loss, wired remote lock, etc (it is not just the behaviour of the indoor fan).
Those functionalities are only available via "Functions Settings".

The problem I have is that I only have a PAR-31MAA wired remote. 
I don't have an MHK1 controller in my hand to record serial communication...

That's why I kindly request some help from people that has an MHK1 to help me dumping the serial commands exchanged when reading and saving "Functions Settings"

Thanks again for your help.
If you want I can move this topic to the "Other Packets" issue.

SwiCago, 2019-05-04:
@luiscgalo  someone in #39 has a mhk1, ask in that issue if they would be willing to do a dump of the packets.

luiscgalo, 2019-05-04:
Thanks @SwiCago 
Yes, today I've written some text about this functionality in the "Other Packets" issue.
Indeed it makes sense to maintain this thread on that issue instead of creating a new one.

I will wait to see if there is someone that can help me dumping packets of the MHK1 controller. :)

Thanks again for your help. ;)