SwiCago/HeatPump #133: Anyone knows how to find the the correct 105 connector on Aliexpress?

jascdk, 2019-05-12:
Hi all. I am going to build the system for my fathers summerhouse. I have almost every part i need, but not the cn105 connector "yet".

I have been looking at the links for the connectors and pigtails in the description. They look good, but they cost a fortune if im going to get them shipped to my location which is Denmark.

Do anyone have a link for the exact connector - maybe including the wires on Ali Express? I have searched on different JST connectors, but i cant find the right one.

Anyone who can help me?

SwiCago, 2019-05-12:
Issues are not to be used for asking questions on parts. Use the gitter chat for that!

closed 2019-05-12 by SwiCago

SwiCago, 2019-05-12:
I'll throw you a bone. But don't open issues that have nothing to do with the library. 
You are looking for  PAP-05V-S
https://m.aliexpress.com/item/32840133526.html?pid=808_0000_0101&spm=a2g0n.search-amp.list.32840133526&aff_trace_key=&aff_platform=msite&m_page_id=875amp-ZRBmwvh5FePoQ_YgpOGpPg1557677774226