SwiCago/HeatPump #134: Home Assistant: NoEntitySpecifiedError

miguelangel-nubla, 2019-06-18:
Home assistant logs show this for each hvac configured. It has been happening for a few months
```
2019-06-18 01:53:14 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 215, in async_update_ha_state
    "No entity id specified for entity {}".format(self.name))
homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity Aire de la oficina
```
```
climate:
  - platform: mitsubishi_mqtt
    name: Aire de la oficina
    command_topic: "home/office/heatpump/set"
    temperature_state_topic: "home/office/heatpump/status"
    state_topic: "home/office/heatpump"
```

Anyone else experiencing this? Any ideas?

SwiCago, 2019-06-18:
https://github.com/SwiCago/HeatPump/tree/master/integrations
Use chat
https://gitter.im/Mitsubishi-Heat-Pump

closed 2019-06-18 by SwiCago

yang3535, 2019-06-28:
I'm seeing this same error.  I've posted to gitter, but so far no response.