SwiCago/HeatPump #140: Voltage levels

tve, 2019-07-29:
I just spent a bunch of hours trying to talk to my indoor unit (SEZ-KD09) and after banging my head against the wall I finally figured out the serial voltage thing... This issue is in response to several other issues (#52, #78, ...), I believe I have finally resolved the serial line voltage question, maybe you can add it to the README?

I turns out the TX of the unit is open-collector, which means that without some form of pull-up the esp8266/esp32/serial-dongle cannot receive what the unit is sending. The fact that it's OC is nice because it resolves the "is it 5v or does 3.3v work?" question. On a 3.3v microcontroller simply pull-up to 3.3v and you have a 3.3v level signal. Note that the esp8266/esp32 have an internal pull-up which can be enabled programatically, but the arduino serial library doesn't provide built-in support for that.

The RX of the unit expects an OC driver as well, a tell-tale sign is that without anything connected to the CN-105 cable the RX pin of the unit sits at 5V, which is due to its own pull-up.

Note that with nothing connected to the CN-105 the TX from the unit sits at 0V, which is an unlikely/invalid state for an idle totem-pole serial output but makes perfect sense for an OC output.

The upshot of all this is that a 3.3V microcontroller can be connected by providing a pull-up to 3.3V on the hp->uC line and by using a shottky diode in series on the hp->uC line. The pull-up to 5V in the unit is most likely weak enough that no shottky diode is really needed, but I didn't try to actually measure the pull-up resistor value.

SwiCago, 2019-07-30:
As you stated it has already been beaten to death in #52 and #78 
The demo circuit shows pullups, if someone can't follow a simple circuit, they shouldn't be messing with a $1000+ head unit risking getting electricuited by 240V

closed 2019-07-30 by SwiCago

tve, 2019-07-30:
Jeez, I've rarely received such a condescending reply to a message. None of the issues/posts/comments I've read mentioned open collector signal lines, which explain the purpose of the pull-ups. Good luck!

SwiCago, 2019-07-30:
The makers of esp8266 already said the i/o ports are 5V tolerant, hence pull up to 5V.
My problem with your post/issue, we it was not necessary to open or create a new issue on something that has already been discussed (at least twice). No one else seems to have problems, since like forever...especially since there is a chat to ask questions as well and a search in git to see if the question has already been asked/answered...but yeah, I am the asshole for making this library and giving it away free and dealing with retarted peoples questions...Piss off.

tve, 2019-07-30:
LOL