SwiCago/HeatPump #148: wideVane not working for me (MQTT example)

jimbo72, 2019-12-05:
Hi all,

Everything is working as expected (thanks for your amazing work), except one command : wideVane

Here is a log from mosquitto, I send 
set {"wideVane":"<"}
on the set topic. Others commands works fine.

How can I help to figure out the problem ? Tell me and I will give you all logs/informations needed

2019-12-05 11:35:55.663 heatpumpc/set {"vane":"AUTO"}
2019-12-05 11:36:11.958 heatpumpc {"power":"OFF","mode":"HEAT","temperature":22,"fan":"QUIET","vane":"AUTO","wideVane":"<<"}
2019-12-05 11:36:18.823 heatpumpc/status {"roomTemperature":19,"operating":false}
2019-12-05 11:36:18.829 heatpumpc/timers {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
2019-12-05 11:36:25.216 heatpumpc/set {"vane":"SWING"}
2019-12-05 11:36:32.447 heatpumpc {"power":"OFF","mode":"HEAT","temperature":22,"fan":"QUIET","vane":"SWING","wideVane":"<<"}
2019-12-05 11:36:35.603 heatpumpc/set {"wideVane":"<"}

SwiCago, 2019-12-06:
Turn debug on. And put the debug packets here, so we can see what is sent to HP and what it responds with.

jimbo72, 2019-12-08:
Thanks for your help.

Here is a log:

2019-12-08 18:19:38.852 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 "}
2019-12-08 18:19:40.754 heatpumpa/debug {"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
2019-12-08 18:19:41.875 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 00 01 08 01 07 00 00 8c ae 28 00 00 00 e8 "}
2019-12-08 18:19:41.882 heatpumpa/status {"roomTemperature":16,"operating":false}
2019-12-08 18:19:41.888 heatpumpa/timers {"mode":"NONE","onMins":0,"onRemainMins":0,"offMins":0,"offRemainMins":0}
2019-12-08 18:19:43.766 heatpumpa/debug {"packetSent":"fc 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7a "}
2019-12-08 18:19:44.882 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 03 00 00 06 00 01 a0 a4 fe 42 00 00 00 9a 00 00 35 "}
2019-12-08 18:19:46.220 heatpumpa/set {"wideVane":"SWING"}
2019-12-08 18:19:46.241 heatpumpa/debug {"packetSent":"fc 41 01 30 10 01 1d 01 01 00 00 00 05 00 00 00 00 00 0c ac 00 a1 "}
2019-12-08 18:19:47.359 heatpumpa/debug {"packetRecv":"fc 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5e "}
2019-12-08 18:19:49.256 heatpumpa/debug {"packetSent":"fc 42 01 30 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 "}
2019-12-08 18:19:50.374 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 04 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 d9 "}
2019-12-08 18:19:52.272 heatpumpa/debug {"packetSent":"fc 42 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 78 "}
2019-12-08 18:19:53.388 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 "}
2019-12-08 18:19:55.288 heatpumpa/debug {"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}
2019-12-08 18:19:56.408 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 06 00 00 00 00 00 10 00 3d 00 00 42 00 00 00 00 c8 "}
2019-12-08 18:19:58.306 heatpumpa/debug {"packetSent":"fc 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 "}
2019-12-08 18:19:59.423 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 "}
2019-12-08 18:20:01.320 heatpumpa/debug {"packetSent":"fc 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7b "}
2019-12-08 18:20:02.442 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 09 00 05 00 00 8c ac 28 00 00 00 eb "}
2019-12-08 18:20:02.448 heatpumpa {"power":"ON","mode":"HEAT","temperature":22,"fan":"AUTO","vane":"5","wideVane":"<<"}
2019-12-08 18:20:04.339 heatpumpa/debug {"packetSent":"fc 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7a "}
2019-12-08 18:20:05.455 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 03 00 00 06 00 01 a0 a4 fe 42 00 00 00 9a 00 00 35 "}
2019-12-08 18:20:07.359 heatpumpa/debug {"packetSent":"fc 42 01 30 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 "}
2019-12-08 18:20:08.472 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 04 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 d9 "}
2019-12-08 18:20:09.343 heatpumpa/debug/set off
2019-12-08 18:20:09.428 heatpumpa/debug debug mode disabled

But I fear that the problem is even worst, I still have "<<" regardless of the real position that I change with the remote.

jimbo72, 2019-12-08:
The unit is a MSZ-AP25VG.

SwiCago, 2019-12-09:
@jimbo72 the debug packet you are interested in is
2019-12-08 18:20:02.442 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 09 00 05 00 00 8c ac 28 00 00 00 eb "}

fc **62** 01 30 10 **02** 00 00 01 01 09 00 05 00 00 **8c** ac 28 00 00 00 eb
Watch packets that are 62 for info packet and 02 for setting. Look at byte that 8c is in, change with your remote and see how it changes. If 8c is swing, then it is possible that the 8 stands for something else and we may need to bit & it with 0x0f to get just 0x0c. Right now there is no mapping for 8c, so it defaults to 01, which is << and that is what you see reported.

Watch the packet type I just suggested above, set the HP with the remote to a different wideVane setting(don't change others) and see if the byte in the position 8c is in changes and note down what setting you did and what the byte is for that setting.
Current code has following mapped
01 = << 
02 = < 
03 = | 
04 = > 
05 = >> 
08 = <>
0c = SWING
In your case they might all start with 8 instead of 0, so 81,82,83,84,85,88,8c

jimbo72, 2019-12-10:
@SwiCago you have it ! amazing !

This is strange but I don't have <>, not possible from the remote, only 6 positions, but for the rest this is what you guessed:

<<
2019-12-10 07:21:12.566 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **81** aa 28 00 00 00 f4 "}

<
2019-12-10 07:21:30.663 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **82** aa 28 00 00 00 f3 "}

|
2019-12-10 07:21:48.760 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **83** aa 28 00 00 00 f2 "}

>
2019-12-10 07:22:06.860 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **84** aa 28 00 00 00 f1 "}

>>
2019-12-10 07:22:24.957 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **85** aa 28 00 00 00 f0 "}

SWING 
2019-12-10 07:22:43.053 heatpumpa/debug {"packetRecv":"fc 62 01 30 10 02 00 00 01 01 0a 01 07 00 00 **8c** aa 28 00 00 00 e9 "}

I guess you know how to patch this ? But also to generate the correct commands ?
How to manage those different behavior ? check for is the MSB is 0 or 8 ?

jimbo72, 2019-12-10:
I tried to modify the code:

const byte WIDEVANE[7]         = {0x01, 0x02, 0x03, 0x04, 0x05, 0x08, 0x0c};

into

const byte WIDEVANE[7]         = {0x81, 0x82, 0x83, 0x84, 0x85, 0x88, 0x8c};

It works as exepected. Still 0x88 = <> doesn't work for me (not possible also from the remote)

jimbo72, 2019-12-10:
Why define constant such WIDEVANE_MAP_SIZE 7

const byte WIDEVANE[WIDEVANE_MAP_SIZE ] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x08, 0x0c};

And remove all hardcoded 7....

This is just an idea...

SwiCago, 2019-12-10:
@jimbo72 then I guess your unit doesn't support <> = 0x08 or 0x88
But I am glad you got it to work with your unit by changing the numbers. Better would be to update HeatPump.cpp and bit AND the wideVane byte with 0x0F and then do lookup.
I set it up to return correct value, just need to add set support for your unit type.

closed 2019-12-10 by SwiCago

SwiCago, 2019-12-10:
@jimbo72  please try latest. I believe it should support yours and other units now.

Masurov, 2020-02-08:
Hi @SwiCago 

I just added a comment on the [commit related to this fix](https://github.com/SwiCago/HeatPump/commit/f5755b1daa36d105278b8a1b6e0fb01225b48cad). It could cause a vertical vane command issue. 

The byte adjustment  | 0x80 is applied to the vertical vane byte setting. After using the remote, I have packet logs showing that the vertical vane byte use 0x00, 0x01, 0x02, ...

benbiais, 2020-02-08:
Hello,
I contacted Masurov some days ago. I have issue with vertical vane commands which are not responding.
Here below is a debug logs, with two Vane commands (AUTO and SWING) not responding, and one wideVane command (SWING) responding.
I have sent a logfile to the commit related to this fix
Thanks to both of you for your help.

benbiais, 2020-02-10:
Hello,
Inspired by Masurov comments of 8th feb, I have suppressed  from heatpump.cpp the widevane adjustment of 10 dec 2019, without any modification in heatpump.h (const byte WIDEVANE[7] still equal to 0x01, 0x02, etc.).
Both VANE and WIDEVAN commands are now perfectly working. Thanks to both of you for your great job.