SwiCago/HeatPump #149: strcat() appending to a constant

RyanDhuse, 2019-12-08:
https://github.com/SwiCago/HeatPump/blob/c87ff9f05c97e4de3e6438dead76798cbef80de7/examples/mitsubishi_heatpump_mqtt_esp8266_esp32/mitsubishi_heatpump_mqtt_esp8266_esp32.ino#L248

`mqtt_client.publish(heatpump_debug_topic, strcat("heatpump: wrong mqtt topic: ", topic));`

Using strcat this way is going to cause a SIGSEGV.  The first parameter can't be constant because it's used as the destination buffer.

SwiCago, 2019-12-08:
@RyanDhuse I am not the author of this example, so I do not know if it is correct or not. I have also not heard of anyone getting SIGSEGV. Have you tried to cause it with this example? If so and it does in fact cause a problem, then please suggest a fix or better yet make a pull.
Thanks

SwiCago, 2020-01-09:
Strcat removed. The else in thoery should never be called if subscribe is done right

closed 2020-01-09 by SwiCago