SwiCago/HeatPump #151: Four Units Reporting RoomTemperature in Unison?

kenberland, 2019-12-29:
I'm just starting. I have four units using the custom component. I thought everything was working but now I see all the reported `roomTemperature` readings moving together when one room is heating.

Has anyone seen anything like this?

[image]

SwiCago, 2019-12-30:
Sorry, I do not give support for custom component or any other automation software such as HAS. Please use gitter chat or #138 for help. But I can give you one thing to check before you continue. Make sure all your units have unit names clientID, topics etc... I run 6 units and do not have this issue, but then again I do not use open source automation but rather my own custom system.
Again, gitter chat or #138 
Good luck

closed 2019-12-30 by SwiCago

kenberland, 2019-12-31:
My bad. It's got nothing to do with HAS, that was just a convenient visualization. I can see it on the mqtt messages sent from the units. *E.g.,* below, the ne unit is reporting 23 C, far above the ambient while the nw unit is operating. When I turn off the nw unit and turn on the ne unit, the reported temperature drops like a rock to the true ambient.

```
$ docker run -it --rm eclipse-mosquitto mosquitto_sub -h mqtt.hero.net -t 'heatpump-ne/status' -v | ts
Dec 31 10:18:12 heatpump-ne/status {"roomTemperature":23,"operating":false}
Dec 31 10:18:37 heatpump-ne/status {"roomTemperature":23,"operating":true}
Dec 31 10:18:55 heatpump-ne/status {"roomTemperature":22,"operating":true}
Dec 31 10:19:12 heatpump-ne/status {"roomTemperature":22,"operating":true}
Dec 31 10:19:19 heatpump-ne/status {"roomTemperature":21,"operating":true}
Dec 31 10:19:43 heatpump-ne/status {"roomTemperature":20,"operating":true}
Dec 31 10:20:12 heatpump-ne/status {"roomTemperature":20,"operating":true}
Dec 31 10:20:31 heatpump-ne/status {"roomTemperature":19,"operating":true}
Dec 31 10:21:12 heatpump-ne/status {"roomTemperature":19,"operating":true}
Dec 31 10:21:19 heatpump-ne/status {"roomTemperature":18.5,"operating":true}
Dec 31 10:21:43 heatpump-ne/status {"roomTemperature":18,"operating":true}
Dec 31 10:22:12 heatpump-ne/status {"roomTemperature":18,"operating":true}
Dec 31 10:22:31 heatpump-ne/status {"roomTemperature":17.5,"operating":true}
Dec 31 10:22:55 heatpump-ne/status {"roomTemperature":17,"operating":true}
Dec 31 10:23:12 heatpump-ne/status {"roomTemperature":17,"operating":true}
```

SwiCago, 2019-12-31:
What you are seeing is not temperature reporting, due to another unit operating and this one not. But rather Idle temperature and running temperature.
When the unit is idle(not operating) the temperature is always reported higher by the heatpump. This might be because of no air flow and hot air rises. As soon as the unit is running(operating) there is airflow and so the temperature reporting starts to drop as fresh air gets pulled in. Depending on the size of the room, the temperature will drop fast. Also note, cool and heat will report different temperatures when on/off. That is do to internal offsets. We cannot change the way these units run. If you want 100% accurate temperatures, then I suggest using a remote temperature sensor and the setRemoteTemperature function of this library.
I have attic units, in the summer when running it will fairly quickly get to actual temperature, but when off will sky rocket to 30C or higher. Or in the winter will drop down to 0C. So for these units I use a remote temperature sensor. 
My wall units, on the other hand, are within 2C accurate, so I do not use a remote sensor, but rather an offset for cool mode when not in operation, so as not to cycle too often. You may have to do the same. 
You can test what I am talking about by turning both units off. See temperature of the NE unit. Switch from COOL to HEAT or FAN and compare the returned temperature, while the unit is still off(only change mode). You will see it reports differently with some internal offset for the different modes. And then report more accurately when running (on) .

kenberland, 2023-08-05:
Just in case anyone ever gets here. I use @SwiCago's solution on Home Assistant.

For remote temperature sensing, I use [AcuRite wireless sensors](https://www.amazon.com/gp/product/B01G7BE9WK) with a cheap [USB SDR hub](https://www.amazon.com/gp/product/B018PUYPCA).

I create `temperature` devices in HA:

```yaml
- platform: mqtt
  state_topic: 'rtl_433/0e5c7dd8801c/devices/Acurite-Tower/A/1623/temperature_C'
  name: "Kids BR Temp"
  unit_of_measurement: "°C"
  device_class: 'temperature'
  value_template: '{{ value }}'
```
and publish them to the Mitsubishi units:
```yaml
- id: '1689990405'
  alias: 'Publish temp to kids br heatpump'
  trigger:
    platform: state
    entity_id: sensor.kids_br_temp
  action:
    service: mqtt.publish
    data:
      topic: heatpump-se/hvac-kids/remote_temp/set
      payload: "{{trigger.to_state.state}}"
```

Once you publish to `heatpump-nw/hvac-office/remote_temp/set` it stays set. To go back to the unit's value, using the internal value, ostensibly the "ambient" temperature at the unit, publish `-1` to the topic.