SwiCago/HeatPump #152: Temperature offset

alexmc1510, 2020-01-27:
Hello,

I have 3 units already installed in MSZ-HR units (Spain). The issue I am facing is that current temperature given by the units in the webpage is far from real temperature of the room measured with an analog thermometer. Is it normal? Is there a way to check the real temperature maybe with the remote control?

For example, unit gives me 13ºC but real temperature is 18ºC.

Regards,

SwiCago, 2020-01-31:
Yes this is normal. The units are high on the wall. Warm air rises. If you want accurate temperature, then you should place a temperature sensor lower in the room and use setRemoteTemperature to override internal sensors....or in your primary program, add an offset.

closed 2020-01-31 by SwiCago

alexmc1510, 2020-01-31:
It doesn't make sense. As you say, warm air rises, my units are 2m on the wall. The analog thermostat is 1,5m on the wall. It should be the opposite situation.
Another question regarding unit behavior, if I put an offset, when the unit stop? When webpage control reaches the target temperature or when internat unit reaches the target temperature. I don't know if I have explained well...
Thanks