SwiCago/HeatPump #154: Use Arduino Stream object instead of HardwareSerial

geoffdavis, 2020-02-23:
It would be nice to be able to use the more generic `Stream` object instead of the ESP8266-specific `HardwareSerial` object (which inherits from `Stream)`. This would allow, for example:
* use with the [ESPHome software UART](https://esphome.io/custom/uart.html) implementation
* use with the [official Arduino software UART](https://www.arduino.cc/en/Reference/SoftwareSerial) implementation

My C++ is admittedly rusty, but I only see `HardwareSerial`-specific method calls in the `HeatPump::connect()` method. It might make sense to make `connect` a general case method, and introduce more a specific `connectHardwareSerial` method that did the automatic baud rate guess and device initialization.

SwiCago, 2020-02-23:
Unfortunately software serial is not compatible with the exact settings needed to make it work with the esp. About a year ago, I added a variant for someone that used customsoftwareserial, so they could use an at mega and control multiple HPs with one device
https://github.com/ledongthuc/CustomSoftwareSerial
However, it did not compile with esp. Software serial did, but would not get parity settings right for the HP...hence hardware serial. But if you can figure it out, make a pull and I'll give it a review.
Cheers

closed 2020-02-23 by SwiCago

geoffdavis, 2020-02-24:
Hmm, even without using the software serial features, handing off a stream object would be beneficial because the ESPHome UART library doesn't directly expose the HardwareSerial object that it creates. Mind if I just send you a PR for a plain Stream version anyway?

SwiCago, 2020-02-24:
Make a fork and try it first on one of your units. If it works, only then do a pull request. Don't do a pull w/o testing first. Thanks