SwiCago/HeatPump #157: Unsupported heat pump - help with debugging.

MoLow, 2020-03-29:
Hi!
I have replaced successfully 4/5 of my Mitsubishi air pumps from Mel-cloud to use this repo (used via https://github.com/gysmo38/mitsubishi2MQTT) .
The last airpump is a [PERFY-P100VMAL-E2](http://www.mitsubishielectric.com.au/PEFY-P-VMAL.html).
it includes a CN-105 port and it works well with a [MAC-567IF-E](https://www.mitsubishi-aircon.com.ua/restricted/Manuals/JG79Y198H01.pdf)

When using this library to communicate with the heat pump - it doesn't work as intended (doesn't seem to connect probably).
Is there anything I can do to debug this? i.e. baud rate, debug raw packets?

Any help would be appreciated.

SwiCago, 2020-03-30:
Download the mqtt verion of the firmware here. Enable debugging. And watch the packets via mosquito sub 
If you see connect packets over and over, then it is not getting the response. 
You can easily test if it is a wiring problem by sending on command, to see if unit turns on. If it turns on, then your RX side has a faulty wire or is not pulled up. If it does not turn on, then your TX line has a faulty wire. Check continuity. 
Do not use cheap wemos d1 clones, most everyone seems to have problems with those.

SwiCago, 2020-03-31:
Closing. Use gitter chat for more help. I believe the issue is with the wiring or lack of proper pull up

closed 2020-03-31 by SwiCago