SwiCago/HeatPump #159: Updated hp settings not reflected for few seconds.

JMan7777, 2020-04-23:
@SwiCago 
, I have a small question:

I'm syncing with the heatpump approx. 500 times per second.
```
hp.sync();
heatpumpSettings settings = hp.getSettings();
```

After I update the heatpump settings:
```
hp.setSettings(new_settings);
hp.update();    //--> returns back true, so it was successful
```

it takes around 5 seconds (aka 2500 sync cycles) until:
```
hp.getSettings();
```

returns back the updated settings. 

Since the update was reported to be successful why the getSettings() still returns old values for few seconds.

Thx a million.

Cheers,

Markus

SwiCago, 2020-04-24:
@JMan7777  sync has built in checks to prevent sending actual packets , so even if you sent 500 in one second, it would never send that many to the HP. The HP interface would't respond anyhow. It is not a fast interface. I have an original kumo cloud, that is even slower. Up to 30seconds of delay, after sending settings to get the updated settings lol.
So 5-10 is perfectly normal. 
To make your code more efficient, use call back methods instead. See the mqtt example on how that is done. Using callbacks gives you the data, the second it is available from the HP.

Remember we are talking about an HVAC system, there is no need for them to be instant. Whether they react within a second or within 30seconds, makes no difference in the world of heating and cooling. That is why manufactures don't make them instant.

If you are making a web page, you can return to your calling page the settings you sent, until you have the response from HP callback and then update those settings with the actual. That way you can give instant response to webpage and get in sync with HP a few seconds later.

SwiCago, 2020-04-24:
Going to close this, as it is not an issue with library, but more a question for the gitter chat. Please use the chat for future questions and if it turns out to be an issue or no one answers, the  open an issue or see if a closed one exists and reopen it with new comments.
Cheers

closed 2020-04-24 by SwiCago

simkard69, 2023-01-04:
I had the same issue, but when taking a look at the library's contents, I could find these informations :
**CPP file**
if((!connected) || (millis() - lastRecv > (**PACKET_SENT_INTERVAL_MS * 10**))) {
**H file**
static const int PACKET_SENT_INTERVAL_MS = 1000;

So that means "sync()" function will not refresh any values at least 10 seconds between each call.
This can probably be modded, but as @SwiCago said, communication interface is pretty slow and we need to avoid sending data over an already existent communication (if so, it might give weird situations).
Anyway, I'm pretty sure that 10 seconds delay is meant to be on the safe side everytime values need to be updated.

The same goes probably when it comes to sending values as I found these lines related to this action :
**CPP file**
bool HeatPump::canSend(bool isInfo) {
  return (millis() - (isInfo ? PACKET_INFO_INTERVAL_MS : PACKET_SENT_INTERVAL_MS)) > lastSend;
}
**H file**
static const int PACKET_INFO_INTERVAL_MS = 2000;